zwave-js/zwave-js #3703: Reporting on Fibaro FGS211 does not work

AlCalzone, 2021-11-13:
Hi there. I have the same problem, though with a FGS211 (Fibaro 1x3kW relay switch). I have the latest HA, latest Z-Wave JS to MQTT. This bug has been present for several months now. It had first appeared over 5-6 months ago, then disappeared for a short period, and returned to never disappear again. I have 8 of these relays, but only two of them are used using the physical switch inputs on the device - I use them to control garage door and the switch is used to trigger it manually. The relays are configured to turn off automatically after one second - they only signal "change" request to a motor controller (not a zwave device).

If the relay is switched on via HA, it works correctly. When it turns off automatically after one second it is also correctly shown in HA (zwavejs). But when controlled using the connected switch it doesn't seem to propagate back to HA (nor zwavejs). The statistics on the device does show messages received from it.

Attached you can find a "silly" level log, the device in question is Node 006. What can be seen in the log is:
1. I use the physical switch 3 times (to achieve "open gate", "stop gate", "close gate") - each press turns the relay on and it turns it self off automatically after 1 second.
2. Node re-interview
3. Repeat of 1 (three on/auto-off actions)

Thank you for looking into this.
[zwavejs_2021-11-13.log](https://github.com/zwave-js/zwavejs2mqtt/files/7532589/zwavejs_2021-11-13.log)

_Originally posted by @jandado in https://github.com/zwave-js/zwavejs2mqtt/issues/1921#issuecomment-968121052_

AlCalzone, 2021-11-13:
I think this has to do with Fibaro's weird choice to reuse the double switch firmware on their single switches. Z-Wave JS cannot detect that the endpoints are useless this way. 

@jandado Can you test [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/fab202ff650b2d22b60c5b180b9d1a58fc6bb782/packages/config/config/devices/0x010f/fgs211_2.1.json) as it is described [here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files)?
You'll have to re-interview the device and the value IDs you use are likely going to change (from endpoint 1 to 0).

jandado, 2021-11-13:
Thank you @AlCalzone. Sorry it took a while - I ran into an issue with the import of param #16 ("$import": "~/templates/master_template.json#state_after_power_failure_off_prev"), but was able to resolve that by manually importing required block.

There are now three binary switches (6-37-0, 6-37-1, 6-37-2), but it is working. Using the physical switch turns on the now available 6-37-0, and it turns itself of automatically. Turning on the 6-37-1 from HA / zwavejs works as previously, 6-37-2 does nothing - as previously.

Perhaps there should only be one binary switch (actual load on the relay) and two "sensors" (there can be two switches connected to the device). Is that achievable? But even if it stays this way it is much better than not having the state change registered :)

Would you need any more logs captured?

AlCalzone, 2021-11-14:
I would expect the behavior you describe. After the change, 6-37-1 should be ignored, but we currently don't have a compat flag to hide everything but the root endpoint 0.

Do you have a second switch attached by any chance so you can capture what's sent in that situation (if anything)? I wonder if we need to preserve endpoint 2.

jandado, 2021-11-14:
I don't have a second switch attached to it, but I think it is worth trying it out to settle the behavior once and for all.

The problematic part of having endpoint 0 and 1 as it is now is that they don't work in sync, yet work on the same single (relay) switch. So currently turning 0 on keeps 1 showing as off, turning 1 on does nothing. Turning 1 off turns the relay off, though 0 stays showing as on - and vice versa. This behavior can not even be worked around in HA using a group of these two switches.
Re-interviewing the node gets them back in sync (though the correct state of both is not propagated back to HA - integration reload is required). So, ideally there should really be only one switch.
I will let you think about this :) In the mean time I will find a spare 30 minutes to try out attaching a second switch to the device and report back as soon as I get the log.

AlCalzone, 2021-11-14:
I think you need to ignore endpoint 1 for now. Question is whether exposing 2 is of any use.

jandado, 2021-11-14:
Reporting back with second switch connected, @AlCalzone. Find attached a silly level log containing:
1. S2 turned on, back off (manually) - no impact on relay switch,
2. re-interview node just to make sure the device state / config is in the log,
3. repeat of step 1,
4. S1 turned on to show the difference - the relay turned on and automatically back off after one second - repeated three times to open/stop/close the garage door (still irrelevant for the purpose of this test, just to explain the repetition)
[zwavejs_2021-11-14.log](https://github.com/zwave-js/node-zwave-js/files/7533998/zwavejs_2021-11-14.log)

I hope this helps.

AlCalzone, 2021-11-14:
Thanks, I do see all of that in the log. Looks like keeping endpoint 1 and 2 then makes more sense, but we need to map the root reports to endpoint 1 so they reach the application.

AlCalzone, 2021-11-14:
Can you test [this version of the file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/cf58a2b3c1dd42bbd6abe72cfd0da5714ef89ecd/packages/config/config/devices/0x010f/fgs211_2.1.json) to be sure?

The reports should now come in on endpoint 1 - endpoint 0 should disappear again.

jandado, 2021-11-14:
I can confirm the behavior is as you describe, @AlCalzone - and I am more than happy with it :)

closed 2021-11-14 by AlCalzone

Ascariota, 2021-11-17:
Worling with Current version: 0.1.48

PR #3704 (2021-11-14): fix(config): map reports from root to endpoint 1 on Fibaro Single Switches
fixes: #3703