zwave-js/zwave-js #3709: Destination Controller Node ID reported as 0

marcus-j-davies, 2021-11-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Having rebuilt a network on 8.7.5
I observed that the `controllerNodeId`  is reported as 0.

Inclusion logs suggest no foul play, so not sure if this is a local reporting issue?

```
2021-11-14T15:51:47.303Z CNTRLR » [Node 009] configuring wakeup destination node
2021-11-14T15:51:47.308Z SERIAL » 0x0119001309129f036700ac75eb061119dc4e066ebd62235125154e            (27 bytes)
2021-11-14T15:51:47.308Z DRIVER » [Node 009] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      21
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 103
                                    └─[WakeUpCCIntervalSet]
                                        wake-up interval:   86400 seconds
                                        controller node id: 1
2021-11-14T15:51:47.311Z SERIAL « [ACK]                                                                   (0x06)
2021-11-14T15:51:47.318Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-14T15:51:47.319Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T15:51:47.320Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-14T15:51:47.426Z SERIAL « 0x011800131500000b01d77f7f7f7f00000302000000030100003f              (26 bytes)
2021-11-14T15:51:47.427Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T15:51:47.428Z DRIVER « [REQ] [SendData]
                                    callback id:     21
                                    transmit status: OK
2021-11-14T15:51:47.431Z CNTRLR   [Node 009] wakeup destination node changed!
```

I should mention, that WAKE UPs are still being received 

### Device information

All devices

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

8.7.5

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have always seen Node 1 for destination, but since 8.7.5 I observe Node 0

### Attach Driver Logfile

[zwave-js.log](https://github.com/zwave-js/node-zwave-js/files/7534111/zwave-js.log)

marcus-j-davies, 2021-11-14:
I don't think the DB is being updated, once its set.
Direct from the DB.
`{"k":"{\"nodeId\":5,\"commandClass\":132,\"endpoint\":0,\"property\":\"controllerNodeId\"}","v":0}`

AlCalzone, 2021-11-14:
Can you query the controller node ID once more now that it was set?
https://zwave-js.github.io/node-zwave-js/#/api/CCs/WakeUp?id=getinterval

marcus-j-davies, 2021-11-14:
[image]

Local value now reports correct Node.

AlCalzone, 2021-11-14:
Ok then this is just a matter of not saving the value.

marcus-j-davies, 2021-11-14:
That was my initial guess.

When it was reported by another User - I thought it was isolated to their node, but then I experienced it (but we still got the Wake Ups).

Seeing it was 0 in the jsonl, I assumed it just hadn't been committed to the DB.

PR #3736 (2021-11-16): fix(cc): store controllerNodeId after setting during interview
fixes: #3709
closed 2021-11-16 by AlCalzone