zwave-js/zwave-js #3711: Serial API did not respond on Zwave-js Server with Vision Gen5 Stick

AWSb24, 2021-11-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Serial API did not respond
What do you observe?
No

What did you expect to happen?
Run the regularly

Steps to reproduce the behavior:
I just installed Node-js 16 and npm on a new Raspberry Pi. I installed Zwave-js and Zwave-js Server with npm -g. I had to install some missing packages (@types/minimist, @types/triple-beam, ts-node, ws). I made a config.js with securitykeys and ran "ts-node src/bin/server.ts /dev/ttyACM0 --config config.js". 

### Device information

Manufacturer: Vision Security
Model name: Z-Wave USB Stick Gen5
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [X] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.7.5
`zwavejs2mqtt` branch: not installed

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave_20211114.log](https://github.com/zwave-js/node-zwave-js/files/7534771/zwave_20211114.log)

AlCalzone, 2021-11-15:
The driver restarts (aka. soft reset) the Z-Wave controller on startup to prevent it from being in any weird state it shouldn't be in.
For USB sticks this can cause them to restart and reconnect under a different port.
You can (and should) avoid that by using `serial/by-id` paths as outlined [here](https://github.com/AlCalzone/ioBroker.zwave2/blob/master/docs/en/troubleshooting.md#the-adapter-does-not-start-after-updating-to-version-23-or-higher). The link is for an application that uses zwave-js, but the steps should be the same for you.

AWSb24, 2021-11-15:
Thanks for the feedback.
I replaced `/dev/ttyACM0` by `/dev/serial/by-id/usb-0658_0200-if00` but the error message is still the same.

This is what I get if I execute `ls -l /dev/serial/by-id`
`lrwxrwxrwx 1 root root 13 15. Nov 10:15 usb-0658_0200-if00 -> ../../ttyACM0`

I also checked the id with `lsusb`.

I also reconnect the stick and rebooted the Pi

marcus-j-davies, 2021-11-15:
The controller you're using, might have issues with soft reset, try disabling it.

```javascript
{
  .....,
  .....,
  enableSoftReset: false
}
```

AWSb24, 2021-11-15:
Thanks Marcus.

After adding `enableSoftReset: false` to my config

```
module.exports = {
  enableSoftReset: false,
  logConfig: {
      level: "silly"
  },
  securityKeys: {
  }
}
```

The driver crashes:
```
13:47:58.245 DRIVER   Failed to initialize the driver: Error: Failed to lock DB file "/usr/lib/node_
                      modules/@zwave-js/server/node_modules/zwave-js/cache/ebde84f0.values.jsonl"!
                          at JsonlDB.open (/usr/lib/node_modules/@zwave-js/server/node_modules/@alca
                      lzone/jsonl-db/src/lib/db.ts:229:10)
                          at async Driver.initValueDBs (/usr/lib/node_modules/@zwave-js/server/node_
                      modules/zwave-js/src/lib/driver/Driver.ts:909:3)
                          at async Driver.initializeControllerAndNodes (/usr/lib/node_modules/@zwave
                      -js/server/node_modules/zwave-js/src/lib/driver/Driver.ts:997:4)
                          at async Immediate.<anonymous> (/usr/lib/node_modules/@zwave-js/server/nod
                      e_modules/zwave-js/src/lib/driver/Driver.ts:817:5)
Error in driver ZWaveError: Failed to initialize the driver: Error: Failed to lock DB file "/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/cache/ebde84f0.values.jsonl"!
    at JsonlDB.open (/usr/lib/node_modules/@zwave-js/server/node_modules/@alcalzone/jsonl-db/src/lib/db.ts:229:10)
    at async Driver.initValueDBs (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/driver/Driver.ts:909:3)
    at async Driver.initializeControllerAndNodes (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/driver/Driver.ts:997:4)
    at async Immediate.<anonymous> (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/driver/Driver.ts:817:5) (ZW0100)
    at Immediate.<anonymous> (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/driver/Driver.ts:834:6) {
  code: 100,
  context: undefined,
  transactionSource: undefined
}
Shutting down
```

marcus-j-davies, 2021-11-15:
The driver uses lock files to stop the DB from getting corrupted, normally this is an indication of the process being terminated unexpectedly  and  started again within a few second.

Could also mean, another process is running using these files, I.e 2 instances of the server.

AWSb24, 2021-11-15:
I wonder how this can happen because it is a completely fresh setup.

How can I check whether a second instance is running?

Btw: If I remove `enableSoftReset: false` the error disappears

marcus-j-davies, 2021-11-15:
you're seeing the error - because it getting further 😄.

Wipe the cache (make sure the server is not running)
delete everything in (including `cache` folder)
`/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/cache`

AWSb24, 2021-11-15:
Thanks you very much, Marcus.

Now the issue is fixed.

I have to run it as root with `enableSoftReset: false` in the config.

Running it as root without `enableSoftReset: false` fails.

AlCalzone, 2021-11-15:
@AWSb24 could you post a driver log of the startup with the `serial/by-id` path, with and without soft reset disabled?
I want to make sure if we need to blacklist your stick aswell.

FYI, I just noticed this:
> I installed Zwave-js and Zwave-js Server with npm -g

You **don't** want to install stuff globally, except maybe CLI utilities. This is probably also why you need to run as root.

Just create an empty directory and `npm install` the necessary dependencies there **without** the `-g` flag.

AWSb24, 2021-11-15:
Without enableSoftReset=false it breaks after 3 attempts

[zwave_2021-11-15-enableSoftReset_false.log](https://github.com/zwave-js/node-zwave-js/files/7540692/zwave_2021-11-15-enableSoftReset_false.log)

With enableSoftReset=false it seems to run smooth. I also added a Node 46.

[zwave_2021-11-15-enableSoftReset_true.log](https://github.com/zwave-js/node-zwave-js/files/7540699/zwave_2021-11-15-enableSoftReset_true.log)

AlCalzone, 2021-11-15:
Huh, that looks like a premature exit. The driver should be waiting 2, 5, 10, then 15 seconds and each time test if the Serial API is back.
Let me fix that.

AlCalzone, 2021-11-15:
@AWSb24 can you install this version and test again with soft reset enabled?
```
npm install zwave-js@8.7.6-0-pr-3716-96ae5af
```

closed 2021-11-15 by AlCalzone

AWSb24, 2021-11-15:
@AlCalzone Sorry but it seems the error is not fixed. This is the log with 8.7.6:
[zwave_2021-11.15-8.7.6.log](https://github.com/zwave-js/node-zwave-js/files/7542069/zwave_2021-11.15-8.7.6.log)

AlCalzone, 2021-11-15:
The error is fixed, but it doesn't help you :) The stick needs to be blacklisted for soft reset.

AWSb24, 2021-11-19:
Danke!

PR #3716 (2021-11-15): fix: don't unref backoff timeout when polling Serial API for startup
fixes: #3711