zwave-js/zwave-js #3712: Had significant interview issues + healing network issues

muddro1, 2021-11-15:
Currently have 5 nodes in my network. Consistently have issues with my zooz devices. Interviewing takes insanely long, and constantly times out, reports device as dead and then would come back. Also, having issues with healing the network.
[zwavejs_2021-11-14.log](https://github.com/zwave-js/node-zwave-js/files/7535133/zwavejs_2021-11-14.log)

`2021-11-14T23:16:43.528Z CNTRLR   [Node 005] [Configuration] 22: metadata updated
2021-11-14T23:16:43.528Z CNTRLR   [Node 005] [~] [Configuration] isParamInformationFromConfig: false  [internal]
                                  => true
2021-11-14T23:16:43.529Z CNTRLR » [Node 005] finding first configuration parameter...
2021-11-14T23:16:43.530Z SERIAL » 0x011000a9010504700e00002500000000071a                              (18 bytes)
2021-11-14T23:16:43.530Z DRIVER » [Node 005] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      7
                                  └─[ConfigurationCCPropertiesGet]
                                      parameter #: 0
2021-11-14T23:16:43.536Z SERIAL « [ACK]                                                                   (0x06)
2021-11-14T23:16:43.536Z SERIAL « 0x010401a90152                                                       (6 bytes)
2021-11-14T23:16:43.536Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T23:16:43.536Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2021-11-14T23:16:43.668Z SERIAL « 0x011d00a90700000c01a77d7f7f7f02010303000000010100007f7f7f7f7f98    (31 bytes)
2021-11-14T23:16:43.668Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T23:16:43.669Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:     7
                                    transmit status: OK
2021-11-14T23:16:43.803Z SERIAL « 0x011400a80001050b700f00000000010000000000a89a                      (22 bytes)
2021-11-14T23:16:43.805Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T23:16:43.805Z DRIVER « [Node 005] [REQ] [BridgeApplicationCommand]
                                  └─[ConfigurationCCPropertiesReport]
                                      parameter #:         0
                                      next param #:        1
                                      value size:          0
                                      value format:        SignedInteger
                                      alters capabilities: false
                                      readonly:            false
                                      advanced:            false
                                      bulk support:        true
2021-11-14T23:16:43.806Z CNTRLR » [Node 005] querying parameter #1 information...
2021-11-14T23:16:43.807Z SERIAL » 0x011000a9010504700e000125000000000814                              (18 bytes)
2021-11-14T23:16:43.807Z DRIVER » [Node 005] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      8
                                  └─[ConfigurationCCPropertiesGet]
                                      parameter #: 1
2021-11-14T23:16:43.812Z SERIAL « [ACK]                                                                   (0x06)
2021-11-14T23:16:43.813Z SERIAL « 0x010401a90152                                                       (6 bytes)
2021-11-14T23:16:43.813Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T23:16:43.813Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2021-11-14T23:16:44.057Z SERIAL « 0x011d00a90800001701a87d7f7f7f02010303000000010100007f7f7f7f7f83    (31 bytes)
2021-11-14T23:16:44.057Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T23:16:44.057Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:     8
                                    transmit status: OK
2021-11-14T23:16:45.450Z SERIAL « 0x011400a80001050b700f00010900020000020000a992                      (22 bytes)
2021-11-14T23:16:45.450Z SERIAL » [ACK]                                                                   (0x06)
2021-11-14T23:16:45.451Z DRIVER « [Node 005] [REQ] [BridgeApplicationCommand]
                                  └─[ConfigurationCCPropertiesReport]
                                      parameter #:         1
                                      next param #:        2
                                      value size:          1
                                      value format:        UnsignedInteger
                                      min value:           0
                                      max value:           2
                                      default value:       0
                                      alters capabilities: false
                                      readonly:            false
                                      advanced:            false
[... 737 lines cut ...]
2021-11-15T00:48:02.890Z SERIAL « [ACK]                                                                   (0x06)
2021-11-15T00:48:02.890Z SERIAL « 0x0104014601bd                                                       (6 bytes)
2021-11-15T00:48:02.890Z SERIAL » [ACK]                                                                   (0x06)
2021-11-15T00:48:02.891Z DRIVER « [RES] [AssignReturnRoute]
                                    has started: true
2021-11-15T00:48:03.046Z SERIAL « 0x01210046180000000000000000000000000000000000000000000000000000000 (35 bytes)
                                  00080
2021-11-15T00:48:03.046Z SERIAL » [ACK]                                                                   (0x06)
2021-11-15T00:48:03.047Z DRIVER « [REQ] [AssignReturnRoute]
                                    callback id:     24
                                    transmit status: OK
2021-11-15T00:48:03.049Z CNTRLR   [Node 004] healed successfully
2021-11-15T00:48:03.051Z CNTRLR » [Node 004] requesting node neighbors...
2021-11-15T00:48:03.052Z SERIAL » 0x01070080040000007c                                                 (9 bytes)
2021-11-15T00:48:03.052Z DRIVER » [REQ] [GetRoutingInfo]
                                    remove non-repeaters: false
                                    remove bad links:     false
2021-11-15T00:48:03.059Z SERIAL « [ACK]                                                                   (0x06)
2021-11-15T00:48:03.060Z SERIAL « 0x01200180050000000000000000000000000000000000000000000000000000000 (34 bytes)
                                  05b
2021-11-15T00:48:03.060Z SERIAL » [ACK]                                                                   (0x06)
2021-11-15T00:48:03.060Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 1, 3
2021-11-15T00:48:03.062Z CNTRLR « [Node 004] node neighbors received: 1, 3
2021-11-15T00:48:03.062Z CNTRLR   Healing sleeping nodes in parallel. Wake them up to heal.`

AlCalzone, 2021-11-15:
Both problems indicate that Node 5 has actual connectivity issues. Let's rule out the common causes first.

Z-Wave sticks in particular are prone to interference, especially by USB(3) ports. We recommend putting the stick in a suitable location:

-   on an **USB extension cord** (this works wonders!)
-   away from other USB ports
-   away from metallic surfaces
-   and especially not in the back of a server rack

muddro1, 2021-11-15:
The USB is on an extension (a three footer). My MOBO is old and has no USB(3) ports none-the-less). I have tried to position the usb in a way to be off the case  in many different positions, but hasn't really made much of a difference one way or the other.

For a bit more info:

- Node 5 is about 40 feet away, almost line of sight
- Node 5 doesn't have other nodes particularly close to it
- Node 2&3 are GE Jasco Fan switches that have no issues, and 3 is also probably 40 ft as the crow flies, just with some walls in the way, and there is never really an issue with that connection.

GrizzlyAK, 2021-11-16:
Just as a point of reference, I have 49 various Zooz devices, both powered and battery, all over my house, some in-wall, some on-wall, some 100' away through several walls and a couple of floors down, and in a crawlspace with boilers, pipes, electrical cords, water heaters, and freezers. Controller is Aeotec Z-Stick 5+ plugged directly into a USB3 port on my RPi 4 on the very top floor. I have no problems with any of them. I have all my battery devices set to wakeup every 2 hours, so healing the network takes a reasonable time. Only one uses encryption. Which Zooz device is Node 5?

AlCalzone, 2021-11-16:
> Controller is Aeotec Z-Stick 5+ plugged directly into a USB3 port [...]. I have no problems

Consider yourself lucky - or maybe it works just well enough that you don't notice. We've had multiple cases where just putting an extension cord between the stick and the USB port made a world of difference.

muddro1, 2021-11-16:
> Another option could be interference where node 5 is located, can you try putting it in a different location for a test? Or do you have another mains-powered node you could put about halfway between the controller and node 5?

That seems to be difficult to do. I have adjusted the positioning of my server to be perpendiculat to the was it was so I can move the usb off to the side up to the legnth of the usb extension rather than above the server. Seems to be working slightly better. Have a 6 footer extension coming from amazon today and will report back if that fixes it. In any case, just shocked there could be so much interference on these USBs to make them practically unusable. Well maybe thats not fair, two nodes are GE Jascos, and they work just fine. My Zooz Zen 77s seem to be very weak signalled.

blhoward2, 2021-11-16:
USB can cause so much interference that a device a few feet away won't include. I second trying an extension cable.

AlCalzone, 2021-11-16:
> In any case, just shocked there could be so much interference on these USBs to make them practically unusable

The USB ports themselves can create radio noise. See https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html, page 8. The power is the highest around 800-900 MHz, which is just where Z-Wave operates.

GrizzlyAK, 2021-11-16:
>My Zooz Zen 77s seem to be very weak signalled.
Have you tried using the range test function on the switch to see what it might tell you? Also, please be sure to let the folks at Zooz support know. They are fantastic to work with and have been extremely helpful to me and quick to respond. It could be a faulty device. They will swap it out.

muddro1, 2021-11-17:
> The USB ports themselves can create radio noise. See https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html, page 8. The power is the [...]

My MoBo doesnt have usb 3 (its frm like 2014), but even so, I bought the 6.5 footer and still far from ideal. I am in communication with the zooz folks, so have to wait and see what they say. These zooz switches have just been screwy. Also seen some new things in the logs:

`2021-11-16 21:07:58.372 ERROR ZWAVE: Error while writing 30 on 7-38-0-targetValue: Failed to send the command after 3 attempts (Status Fail) (ZW0202)
2021-11-16 21:07:58.372 ERROR ZWAVE: Unable to write 30 on 7-38-0-targetValue
2021-11-16 21:07:58.372 INFO ZWAVE: Success zwave api call writeValue undefined
2021-11-16 21:07:59.424 ERROR ZWAVE: Error while writing 30 on 7-38-0-targetValue: Failed to send the command after 3 attempts (Status Fail) (ZW0202)
2021-11-16 21:07:59.425 ERROR ZWAVE: Unable to write 30 on 7-38-0-targetValue
2021-11-16 21:07:59.425 INFO ZWAVE: Success zwave api call writeValue undefined
2021-11-16 21:08:10.859 ERROR ZWAVE-SERVER: Failed to send the command after 3 attempts (Status Fail) (ZW0202)
ZWaveError: Failed to send the command after 3 attempts (Status Fail) (ZW0202)
at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3274:23)
at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3419:28)
at Driver.sendSupervisedCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3480:27)
at Driver.trySendCommandSupervised (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3512:16)
at MultilevelSwitchCCAPI.set (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/MultilevelSwitchCC.ts:173:47)
at Proxy.MultilevelSwitchCCAPI.<computed> (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/MultilevelSwitchCC.ts:323:33)
at ZWaveNode.setValue (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:827:14)
at Function.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:44)
at Object.node (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:39:91)
at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:95:99)`

Not sure whats happening there.

muddro1, 2021-11-17:
Also noticing a difference in operation in 2 different switches:

`2021-11-16 21:29:00.024 INFO ZWAVE: Node 7: value updated: 38-0-currentValue 99 => 59
2021-11-16 21:29:01.183 INFO ZWAVE: Node 7: value updated: 38-0-targetValue 59 => 59
2021-11-16 21:29:01.184 INFO ZWAVE: Node 7: value updated: 38-0-duration 0s => 5s
2021-11-16 21:29:01.185 INFO ZWAVE: Node 7: value updated: 38-0-currentValue 59 => 59
2021-11-16 21:29:33.085 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 99 => 50
2021-11-16 21:31:19.646 INFO ZWAVE: Node 7: value updated: 38-0-currentValue 59 => 30
2021-11-16 21:31:20.481 INFO ZWAVE: Node 7: value updated: 38-0-targetValue 30 => 30
2021-11-16 21:31:20.482 INFO ZWAVE: Node 7: value updated: 38-0-duration 5s => 5s
2021-11-16 21:31:20.482 INFO ZWAVE: Node 7: value updated: 38-0-currentValue 30 => 30`

For some reason Node 7 has a transition, which I cannot get rid of. When I try to change the parameter in zwave I get:

`2021-11-16 21:33:59.289 INFO ZWAVE: Calling api writeValue with args: [
{ nodeId: 7, commandClass: 38, endpoint: 0, property: 'duration' },
{ value: 0, unit: 'seconds' },
{ transitionDuration: '0' },
[length]: 3
]
2021-11-16 21:33:59.289 INFO ZWAVE: Writing { value: 0, unit: 'seconds' } to 7-38-0-duration
2021-11-16 21:33:59.290 ERROR ZWAVE: Unable to write { value: 0, unit: 'seconds' } on 7-38-0-duration
2021-11-16 21:33:59.290 INFO ZWAVE: Success zwave api call writeValue undefined`

GrizzlyAK, 2021-11-17:
How are you trying to change the transition? I have several ZEN22 and a ZEN30 dimmers, and I use the Configuration CC (112) to set the ramp rate, in the case of ZEN77, it is parameter 9 for On/Off, or P21 for physical dimming speed.  I've never used the transition duration in the Multilevel Switch CC (38) for anything. I don't even know if it works. Have you tried setting Parameters 9 and/or 21?

muddro1, 2021-11-17:
> How are you trying to change the transition? I have several ZEN22 and a ZEN30 dimmers, and I use the Configuration CC (112) to set the ramp rate, in the case of ZEN77, it is parameter 9 for On/Off, [...]

You are right. I had never changed anyof the parameters before, but noticed that that one was 5 seconds and that was what i was seeing in debug. The strange thing is for the same switch at another light it doesn't do that. Even when I changed param 9 it still had that 5 second delay thing in the logs.

GrizzlyAK, 2021-11-17:
Are you using the zwavejs2mqtt control panel to manage your network? If so, just make sure you press the Arrow to the right of EACH parameter value you change for each one. I make that mistake often, entering several values, before realizing that I had not told it to GO. I've found that it works consistently with all my Zooz devices for setting parameters, and immediately for powered devices. Do you have the link to the web page for the Advanced parameters for that device? They aren't in the manual (most times they are).

AlCalzone, 2021-11-17:
> Failed to send the command after 3 attempts (Status Fail)

Status Fail means that the controller is unable to send because it doesn't get the necessary time on the radio.
Either one of your nodes is spamming like crazy or there's something else jamming the radio frequency.

muddro1, 2021-11-17:
Yeah, my server must be really cluttering the signal. I put the 6.5 and the 3 footer together, and things work much better, albeit very ugly. hopefully a piece of furniture I am waiting on will dampen the signal from my server as well, or I may attempt usb over lan. I'll close this out for now, but I appreciate all your suggestions and help. And as far as the delay thing, they fixed that in a firmware update which I was able to install once I had the longer usb cable. Thanks again for being patient with me.

closed 2021-11-17 by muddro1