zwave-js/zwave-js #3742: Rejected promise

godp1301, 2021-11-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
Unhandled promise is getting rejected when adding a node in my network

**What do you observe?**
Zwavejs2mqtt docker container is in a restart loop because of the error

**What did you expect to happen?**
The zwave2js driver should handle this exception properly

**Steps to reproduce the behavior:**
Configure zwavejs2mqtt to use the zwave controller HUSBZB-1 -> OK
Include a new node in the zwave network using ZWave To MQTT control panel -> FAIL

### Device information

Manufacturer: Leviton
Model name:  DZ6HD  FW 1.17 and 1.20
Node ID in your network: 2

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwavejs2mqtt: 5.12.0
zwave-js: 8.7.6
home id: undefined
home hex: undefined

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Used to work in the bundled version in zwavejs2mqtt version 5.0.4

### Attach Driver Logfile

```
2021-11-16 19:45:30.938 INFO APP: Version: 5.12.0
2021-11-16 19:45:30.942 INFO APP: Application path:/usr/src/app

 ______                       _     ___                  _   _
|___  /                      (_)   |__ \                | | | |
   / /_      ____ ___   _____ _ ___   ) |_ __ ___   __ _| |_| |_
  / /\ \ /\ / / _` \ \ / / _ \ / __| / /| '_ ` _ \ / _` | __| __|
 / /__\ V  V / (_| |\ V /  __/ \__ \/ /_| | | | | | (_| | |_| |_
/_____|\_/\_/ \__,_| \_/ \___| |___/____|_| |_| |_|\__, |\__|\__|
                            _/ |                      | |
                           |__/                       |_|

2021-11-16 19:45:30.977 WARN STORE: scenes.json not found
2021-11-16 19:45:31.000 INFO APP: Listening on port 8091 host 0.0.0.0 protocol HTTP
2021-11-16 19:45:31.047 INFO MQTT: MQTT is disabled
2021-11-16 19:45:31.063 INFO ZWAVE: Connecting to /dev/ttyUSB0
2021-11-16 19:45:34.140 DEBUG SOCKET: New connection yLf7uenraBK-Zu3OAAAB
2021-11-16 19:45:38.874 INFO GATEWAY: Driver is READY
2021-11-16 19:45:38.877 INFO ZWAVE: Zwave driver is ready
2021-11-16 19:45:38.878 INFO ZWAVE: Controller status: Driver ready
2021-11-16 19:45:38.918 DEBUG ZWAVE: Binding to node 1 events
2021-11-16 19:45:38.921 DEBUG ZWAVE: Node 1 has been added to nodes array
2021-11-16 19:45:38.923 DEBUG ZWAVE: Binding to node 2 events
2021-11-16 19:45:38.924 DEBUG ZWAVE: Node 2 has been added to nodes array
2021-11-16 19:45:38.927 INFO ZWAVE: Scanning network with homeid: 0xf56f8569
2021-11-16 19:45:38.931 INFO ZWAVE: Node 1 is alive
2021-11-16 19:45:38.936 DEBUG ZWAVE: Node 1: mapping absence of CC 128 (Battery) to node property 'isBatteryPowered
2021-11-16 19:45:38.937 INFO ZWAVE: Node 1 ready: Sigma Designs (Former Zensys) - HUSBZB-1 (QuickStick Combo)
2021-11-16 19:45:39.005 INFO ZWAVE-SERVER: ZwaveJS server listening on port 3000
2021-11-16 19:45:39.027 INFO ZWAVE: Node 2 is alive
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
TypeError: _b.getAssociationConfigForEndpoint is not a function
    at ZWaveNode.configureLifelineAssociations (/usr/src/app/node_modules/zwave-js/src/lib/node/Endpoint.ts:545:24)
    at AssociationCC.interview (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/AssociationCC.ts:393:18)
    at interviewEndpoint (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1565:5)
    at ZWaveNode.interviewCCs (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1750:19)
    at ZWaveNode.interview (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1294:8)
    at Driver.interviewNode (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1156:10)
    at /usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1103:7`
```

AlCalzone, 2021-11-17:
I've seen this error a handful of times, very sporadically, but it seems to be "impossible", so we will need to edit some files in the container to get to the bottom of it.

Open a shell in the container and edit the following file:
```
/usr/src/app/node_modules/zwave-js/build/lib/node/Endpoint.js
```

Lines 410ff should look like this:
```js
        for (const group of lifelineGroups) {
            const groupSupportsMultiChannelAssociation = group <= mcGroupCount;
            const assocConfig = (_b = node.deviceConfig) === null || _b === void 0 ? void 0 : _b.getAssociationConfigForEndpoint(this.index, group);
```

Directly above the line with `const assocConfig` please insert the following:
```
            this.driver.controllerLog.logNode(node.id, "device config: " + JSON.stringify(node.deviceConfig));
```

Then save and exit.

Now please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.

godp1301, 2021-11-18:
Good day,

I was unfortunately unable to reproduce the issue after raising this bug. I kept on investigating yesterday and I deleted the container's configuration volume and store directory. Since then, I am unable to get this error, node are probed properly and everything is working fine now.

I will reopen the issue if even I encounter it again.
Thanks,
Phil

closed 2021-11-18 by godp1301