zwave-js/zwave-js #3743: Nodes stuck in protocolinfo/nodeinfo state

mdegroux, 2021-11-17:
Hello,

In a fresh install of zwavejs2mqtt 5.12.0 on docker, with a zwave controller having an existing network

My previous software was runnin openzwave 1.4 and I had custom network key
In zwavejs2mqtt, I have copied the network key in S0 legacy and generated new S2 keys

In the control panel,:
- all electric powered nodes are in nodeinfostate 
- all battery powered nodes (node 18 - fibaro FGMS-001 and node 43 - Secure SRT321) are in protocolinfo state and the manufacturer is unknwon

I have woke up manually my devices but it does nothing

I have attached zwave-js and zwavejs2mqtt logs
[zwavejs_2021-11-17 (1).log](https://github.com/zwave-js/zwavejs2mqtt/files/7550775/zwavejs_2021-11-17.1.log)
[zwavejs2mqtt_2021-11-17 (1).log](https://github.com/zwave-js/zwavejs2mqtt/files/7550776/zwavejs2mqtt_2021-11-17.1.log)

robertsLando, 2021-11-17:
Cannot say much about my side, it looks like some commands are dopped but I don't see any errors on first sight on zwavejs side. Let's see @AlCalzone opinion about logs

AlCalzone, 2021-11-17:
The interview of all nodes is still going on, so you'll need to have a bit of patience. It takes long because several of your nodes reply with incorrectly formatted commands. Those are dropped, leading to timeouts of 10s each (total 390s).

Nodes 23, 47, 49, 50, 54, 58, 67:
- Support Association CC, but reply with a truncated command for the group count (byte X is missing):
[image]
- Don't include the marked byte in the command list response: 
[image]
- Respond with association group names where the length value is higher than the name field is long, e.g. 12 for "BasicSetKey" (11 chars):
[image]

I guess those are all the same devices?

mdegroux, 2021-11-17:
Yes, they are all qubino ZMNHJD1

mdegroux, 2021-11-17:
I don't know if it can helps, I ha ve made tests with zwave2mqtt (not zwave**js**2mqtt)
I have set a command timeout of 5s and the network is ready in few minutes

I have restarted zwavejs2mqtt and I'll wait without touching the configuration

AlCalzone, 2021-11-17:
AFAIK, OpenZWave relies heavily on the information in their XML config files while Z-Wave JS tries to get as much information as possible from the devices themselves. OZW likely didn't query these commands at all.

Might be worth contacting the Qubino support about this, although this won't help you immediately since AFAIK they only update their devices offline.

mdegroux, 2021-11-17:
After a long wait (more than an hour) all nodes are in complete state

mdegroux, 2021-11-17:
A last question,
When migrating from OZW to zwave-js
in OZW, we have only a network key
in zwave-js we have one S0 and three S2

Is it a good practice to copy the OZW key in S0 and generate three random S2?

robertsLando, 2021-11-17:
> Is it a good practice to copy the OZW key in S0 and generate three random S2?

Yes

> in OZW, we have only a network key

That's because OZW didn't ever supported S2 security

closed 2021-11-17 by mdegroux