zwave-js/zwave-js #3745: Missing device configuration: Zooz Zen15 firmware 1.06 parameter 27

alvinchen1, 2021-11-17:
Hello,

Amazing work here, I'm way too new to this but wanted to bring up a request to modify a parameter to the following configuration:

https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x027a/zen15.json

According to the manufacturer:  https://www.support.getzooz.com/kb/article/616-zen15-power-switch-ver-1-06-advanced-settings/

If the firmware version is 1.06 or greater, parameter 27 gets a few additional/tweaked settings.

Remaining the same as what exists in the database:
0 = Always On

Changed/New:
1 = Changes from 'Momentarily (at status change)' to 'LED displays power consumption color when on, none when off' (per the LED Notification section in the attached User Manual) 
2 = (New) 'LED displays power consumption color for 5 seconds regardless if on or off' (again per the LED Notification section in the attached User Manual)
3 = (New) 'LED indicator is OFF'

I was unsure if it made sense to create a new zen15.json file with this firmware version (as done with the zen16.json files in https://github.com/zwave-js/node-zwave-js/tree/master/packages/config/config/devices/0x027a), or if these changes could be incorporated into the existing zen15.json using 

"$if": "firmwareVersion >= 1.06",

And how that might affect <1.06 and the description change for setting 1 in this parameter.

I also was unsure how to apply the style guide for the description of the changed or new parameters to be consistent, I certainly appreciate the fact that you focus a lot of effort on user consistency.  Overall, I was just too unsure as to how to make the appropriate change, but would really love to see the additional parameters (especially since I prefer to have the LED indicators off).    Hoping this provides enough information so someone more skilled and experienced than me could incorporate these changes.  Thanks and keep up the great work!

AlCalzone, 2021-11-17:
Yes, we can include the change using conditionals (`"$if": "firmwareVersion >= 1.6",`). Note that there are no leading zeroes in Z-Wave firmware versions, even if manufacturers print it like that.
The best way to do that change IMO is to copy parameter 27, and make both conditional - one with firmware < 1.6 and an updated one with firmware >= 1.6.

Regarding consistency, I'll refer to @blhoward2

alvinchen1, 2021-11-20:
Thanks, @AlCalzone, threw a little sponsorship your way as a thanks for working this issue.

GrizzlyAK, 2021-11-21:
@AlCalzone if you want me to work this, I can contact Zooz and get the info for the various FW versions and, depending on the magnitude of changes, suggest that I either use conditionals in a single file, or break them up.

blhoward2, 2021-11-21:
We're deprecating separate files so they'd have to be conditional. Let's standby. I owe them a list of devices we want to prioritize for the info.

GrizzlyAK, 2021-11-21:
Cool. I'll work it. 

> We're deprecating separate files

I like the sound of that

> standby on bugging them, though. I owe them a list

Sounds good. I am in contact with them quite a bit for various reasons, but will coordinate with you before requesting any information specifically related to updating device files so as not to duplicate their efforts.

blhoward2, 2021-11-21:
I'm happy to include you in the convo about the device files and let you be the point on that. I just don't want that...duplicating. Let's just join forces on it. Email me: blhoward2@gmail.com.

closed 2021-11-21 by AlCalzone

AlCalzone, 2021-11-21:
@alvinchen1 the changes have been made and merged. If you're using zwavejs2mqtt, you can use the [updated file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/0x027a/zen15.json) as [described here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files).

alvinchen1, 2021-11-21:
Awesome!  Thanks so much for that. Keep up the great work!

PR #3768 (2021-11-21): fix(config): update Zooz ZEN15 Config to FW 1.6+
This PR updates Parameter #27 to include additional options for later FW versions. Note that the changes to this parameter were introduced in FW v1.3, not v1.6 as suggested in the Issue (per the device's [changelog](https://www.support.getzooz.com/kb/article/309-zen15%C2%A0power-switch-change-log/)).

For reference, the manufacturer's parameter guide is [here](https://www.support.getzooz.com/kb/article/616-zen15-power-switch-ver-1-06-advanced-settings/).

Also fixed an incorrect FW version used in an existing conditional for Parameters #33 and # [... cut]