zwave-js/zwave-js #3752: HS1CA-Z Shenzhen Heiman Technology Co., Ltd. / CO Detector

sanderlv, 2021-11-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Missing device paramters in zwavejs. It was working in Vera before and I wanted to migrate

What do you observe?
I do not see a sensor for CO, only battery.

What did you expect to happen?
an entity sensor.battery of equal to appear

### Device information

Manufacturer: Shenzhen Heiman Technology Co., Ltd. / CO Detector
Model name: HS1CA-Z
Node ID in your network: 56

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

ZwaveJS version: 0.1.48 

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

In my older Vera controller

### Attach Driver Logfile

Sorry, I do not have logging as I added 2 devices in a hard position. the second (ID57) succeeded. But I have trouble with this one:
https://devices.zwave-js.io/?jumpTo=0x0260:0x8005:0x1000:0.0#

[image]

AlCalzone, 2021-11-19:
I really can't help here without logs. Since I don't use HA, I suggest you open this issue in https://github.com/home-assistant/core/ and come back if it has been identified as a driver issue.

sanderlv, 2021-11-19:
How and where can  I provide you with logs and from what? I am willing to help here.

AlCalzone, 2021-11-19:
I need a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) from a re-interview of the device, loglevel `debug` or `silly`, attached here as a file.

I don't know how you do that in HA though. Make sure to wake up the device after starting the re-interview or there won't be anything useful in the logs.

closed 2021-11-30 by zwave-js-assistant[bot]