zwave-js/zwave-js #3754: Events not generated for Heatit Z-wave Thermostat (TF016) basic reports

razzymoose, 2021-11-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
[Heatit Z-Wave TF016 Thermostat (FW version 1.8)](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x019b/tf016_1.8-1.91.json) sends Basic reports with relay state. The issue is that these do not result in an event. The event is needed in home assistant to be able to perform actions based on the relay state in the Basic report. The deprecated Z-wave component in Home Assistant reported events based on these reports and it would be good to achieve feature parity with zwave-js being promoted as the replacement.

What do you observe?
I see the zwave_js debug log show the following messages which indicates that these reports are received. 

2021-11-18T21:23:30.995Z CNTRLR   [Node 003] [~] [Basic] currentValue: 0 => 255                     [Endpoint 0]
2021-11-18T21:25:08.011Z CNTRLR   [Node 003] [~] [Basic] currentValue: 255 => 99                    [Endpoint 0]
2021-11-18T21:25:50.030Z CNTRLR   [Node 003] [~] [Basic] currentValue: 99 => 0                      [Endpoint 0]

Listening for zwave_js_notification or zwave_js_value_notification in Home Assistant does not show any events listed for these reports. I do see other unrelated z-wave events reported, so the event listere seems to be configured correctly.

What did you expect to happen?
The expected behavior is for a zwave_js_notification/zwave_js_value_notification to be generated when receiving a Basic report for this device. 

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Heatit Controls AB
Model name: Heatit Z-Wave TF016
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
HomeAssistant zwave_js version: 0.1.48, which includes:
    Bump Z-Wave JS to 8.7.6
    Bump Z-Wave JS Server to 1.11.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have not seen it work with zwave-js, but worked in [deprecated Z-wave component in home assistant](https://www.home-assistant.io/integrations/zwave/)

### Attach Driver Logfile

[Full log not available for download using Home Assistant running in HassOS](https://community.home-assistant.io/t/where-are-the-full-zwave-js-logs/293226)

kpine, 2021-11-18:
> Full log not available for download using Home Assistant running in HassOS

That post is 8 months old and no longer correct. In HA go to Configuration -> Integrations -> Select CONFIGURE for Z-Wave JS Integration -> Go to the Logs tab, select "Debug" log level and interact with the device, then click the download button to download the logs as a file. The only kinds of logs that aren't available are when the addon starts up.

AlCalzone, 2021-11-19:
The config file most likely needs the `disableBasicMapping` compat flag. @razzymoose I guess you've added the controller to association group 2 to get the Basic Set commands?

razzymoose, 2021-11-19:
@kpine Thanks. Added a comment about this back to the HA thread for the next guy. Added a log to the comment above now. Also filled in the controller device. 

@AlCalzone. I have not added the controller to the association group 2 of the thermostat. The thermostats were included with the old deprecated z-wave home assistant component (no associations done then either) and I haven't changed the network since migrating to zwave-js. Is it necessary for the controller to be added to association group 2 for the controller to receive the basic set commands? It is showing them in the debug log already. If it is needed, is there a way I can do this with zwave-js? I can only find a way to set associations using zwavejs2mqtt control panel. Should I switch to this to set the association and then move back to zwave-js?

AlCalzone, 2021-11-19:
I didn't mean that you have to add the controller to group 2, but it would be good to verify if it is already in there. Its possible that OZW set up this association along with the group 1 lifeline. You can check that in zwavejs2mqtt.

AlCalzone, 2021-11-19:
@zwave-js-bot add compat flag to 0x019b:0x0001:0x0001:1.8
```json
// The device reports its relay state via the Basic CC
disableBasicMapping: true
```

AlCalzone, 2021-11-19:
@razzymoose can you test [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/79a959331f623fe357a65595a9518bd47799d55e/packages/config/config/devices/0x019b/tf016_1.8-1.91.json) please as described [here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files)?

blhoward2, 2021-11-22:
He can't test it in the zwavejs addon. He'll have to switch to the zjs2mqtt addon temporarily (or permanently).

razzymoose, 2021-11-23:
@AlCalzone I tested the file with the disableBasicMapping flag set. That did not have an effect. Looking through some other similar issue, I also tested using the treatBasicSetAsEvent compat flag instead. This worked and I can now see the events coming through as zwave_js_value_notification events in home assistant. Not sure what these flags actually do, but from the sound of it, the latter sounds more like what we are looking for. 

BTW, with both flags I verified the debug log of the thermostats in the zwavejs2mqtt control panel which shows that the config file with the correct flags were found correctly, although the debug log did not show any message like the one below as the [guide](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files) said to look for.

`10:32:09.322 CNTRLR   [Node 001] User-provided device config loaded`

AlCalzone, 2021-11-23:
> I also tested using the treatBasicSetAsEvent compat flag instead. This worked and I can now see the events coming through as zwave_js_value_notification events in home assistant. Not sure what these flags actually do, but from the sound of it, the latter sounds more like what we are looking for.

I'm not sure. Events are stateless, but you're looking for something that mirrors the relay state.

Can you try the original file with `disableBasicMapping` and re-interview the device?

blhoward2, 2021-11-23:
In addition to reinterviewing,what happens in the log? I'm not sure that HA sets up entities for Basic CC entities so the problem may be further up the chain.

You may need to use the events and a template sensor in HA to store the last event information so it's sort of like a state.

razzymoose, 2021-11-23:
@AlCalzone Here is a log where I set disableBasicMapping again and interviewed node 3

[zwavejs2mqtt_2021-11-23.log](https://github.com/zwave-js/node-zwave-js/files/7590947/zwavejs2mqtt_2021-11-23.log)

@blhoward2 In home assistant have a set of automations that set/clear an input_boolean based on these events and their value. From that I calculate power consumption as the load of the thermostat is known. So for that usecase the stateless events work fine.

AlCalzone, 2021-11-23:
@razzymoose the other logfile (zwavejs_...) please :)

razzymoose, 2021-11-23:
OK, so now I see that even though the disableBasicMapping flag didn't generate events in HA, I see it generated a Thermostat Basic number entity for each thermostat that directly reflects the Basic value. I didn't notice that before. Was this the intended fix? If so that should work even better as it bypasses the need for me to keep this state. I was too hung up on the events...

AlCalzone, 2021-11-23:
>  I see it generated a Thermostat Basic number entity for each thermostat that directly reflects the Basic value. I didn't notice that before.

I'm not sure if I understand it correctly, but if that is the Basic CC `currentValue`, then yes - it is the intended fix.

razzymoose, 2021-11-23:
> 
Yes, the Basic CC currentValue. So that fix looks all good to me then. Not what the title of the bug asks for directly, but I guess a better fix. Thanks!

blhoward2, 2021-11-23:
As I recall, that basic entity won't idle and should keep the last value for you. You can use a template sensor to create a new sensor that matches the number with whatever you like, e.g. "heating" or whatever.

PR #3763 (2021-11-24): fix(config): disable Basic CC mapping for HeatIt TF016
fixes: #3754

**TODO:**
- [x]	Change PR title to be more specific