zwave-js/zwave-js #3755: Support for Remotec BW8510

AWSb24, 2021-11-19:
**Is your feature request related to a problem? Please describe.**
Currently, the Battery status is visible in HA only.

**Describe the solution you'd like**
I use the BW8510 to control the periphery of my 3D printer. It worked with some limitations on the old Z-Wave integration (before Open Z-Wave). It would be great to have it in Zwave-JS.

AlCalzone, 2021-11-19:
I can't do anything with this info. Please choose an appropriate issue template and fill it out completely.

closed 2021-11-19 by AlCalzone

AWSb24, 2021-11-19:
Sorry, I was not aware of which information you are looking for.
The device has 8 buttons and each button has 3 functions.
1. Key press
2. Key held down
3. Key press 2x

This is the dump file:
[zrc-90.txt](https://github.com/zwave-js/node-zwave-js/files/7573055/zrc-90.txt)

This is a link to the old Home Assistant integration. There are some details about the command class.
https://www.home-assistant.io/integrations/zwave/#remotec-zrc-90-scene-master

If you give me a hint how to modify the json file I can try it on my own.

Thanks!

AlCalzone, 2021-11-19:
You have to listen for Central Scene events. I don't know how to do that in HA, you need to ask in the HA forums/discord or whereever.