zwave-js/zwave-js #3776: Re-interview discards node info too soon

kpine, 2021-11-22:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Re-interviewing a node then restarting Z-Wave JS before the interview starts.

What do you observe?

When performing the above steps, all of the node information is lost after Z-Wave JS restarts. For a battery device especially, this causes loss of all functionality, since all of the value IDs are missing. You either need to wake up the node manually, or wait it out.

What did you expect to happen?

The node info to be retained until the interview actually starts. Also either the re-interview to be resumed on startup, or just cancelled without loss of node info.

Steps to reproduce the behavior:
  1. Re-interview a device. A battery device reproduces this easiest.
  2. Notice the interview status is now ProtocolInfo (I think, or one of the very early stages). At this point, the node info is still valid and available.
  3. Restart Z-Wave JS
  4. Notice all of the information about the node is lost, include values.

### Device information

N/A

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [X] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: sha-11099eb
`zwavejs2mqtt` branch: 8.7.7

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2021-11-21.log](https://github.com/zwave-js/node-zwave-js/files/7578398/zwavejs_2021-11-21.log)

AlCalzone, 2021-11-22:
> Also either the re-interview to be resumed on startup

This should happen already, you have to wake the device though.

kpine, 2021-11-22:
> This should happen already, you have to wake the device though.

Well, yeah, because the node info was deleted. 😅 I meant it in the context of a possible solution. If instead you don't delete the node info, perhaps remember that a re-interview was queued up, and resume it after the restart. I guess that idea (resuming queued operations after a restart) is really a feature request on its own.

> Actually it is not. The driver deletes all info as soon as the refreshInfo method is called.
> zwavejs2mqtt doesn't update though.

Ok, I see that now. Both HA and zwavejs2mqtt have no clue that anything happened to the node info, but they don't receive any events anymore.

What started this was that zwavejs2mqtt is showing "Unknown" for many of my battery nodes' "beaming" field. I was trying to re-interview them to clear that up. So I started re-interviewing them one-by-one, and was just letting them interview at wake up, not realizing they wouldn't function anymore. Did an OS update (docker restart) and saw the lost node...

To tangent a little bit, do you know anything about the "supportsBeaming" property? It's always true. Is this some useless bit like the routing one, where it doesn't mean what you would think it does? There's no way my battery-powered smoke detectors support beaming.

[image]

PR #3974 (2021-12-28): feat: delay resetting node info for sleeping nodes until wakeup
fixes: #3776
closed 2021-12-28 by AlCalzone