zwave-js/zwave-js #3777: [bug] Driver 8.7.6 Serial port errored: The socket closed unexpectedly! (ZW0100)

Aesculapius, 2021-11-22:
### Version

**Checklist:**

- [ ] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 5.12.0
Z-Wave JS version: 8.7.6

### Describe the bug

While I do run Homeassistant, this doesn't have anything to do with that communication. I'm running zwavejs2mqtt on a different server, in a Proxmox hosted VM using docker. I updated zwavejs2mqtt to the latest version by running docker-compose pull, and docker-compose up -d. With the new version, the connection to my Aeotec Zwave Stick v5 is lost. This stick is stuck in a separate Raspberry Pi, which has always provided perfect connection to the zwavejs2mqtt docker by using ser2net. 

The Zwavejs2mqtt config is referring to the stick with this address: tcp://10.0.15.11:3334. This has always worked perfectly.

Now with the new version, the following errors occur: 

2021-11-21T21:11:16.660Z DRIVER   starting driver...
2021-11-21T21:11:16.663Z DRIVER   opening serial port tcp://10.0.15.11:3334
2021-11-21T21:11:16.854Z DRIVER   serial port opened
2021-11-21T21:11:16.854Z SERIAL ▒ [NAK]                                                                   (0x15)
2021-11-21T21:11:16.861Z DRIVER   loading configuration...
2021-11-21T21:11:16.862Z DRIVER   Serial port errored: The socket closed unexpectedly! (ZW0100)
2021-11-21T21:11:16.863Z DRIVER   destroying driver instance...

In the GUI:
`
Driver: Failed to initialize the driver: ZWaveError: The driver is not ready or has been destroyed (ZW0103) at Driver.ensureReady (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1933:10) at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3200:8) at ZWaveController.identify (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:668:33) at Driver.initializeControllerAndNodes (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:947:26) at Immediate.<anonymous> (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:817:16) (ZW0`100)
`

All my nodes are removed from the GUI, it looks like the complete config is destroyed. Ofcourse Homeassistant can't connect but that's understandable. 

### To Reproduce

Steps to reproduce the behavior:

1. Upgrade to latest ZwaveJS2Mqtt, including latest zwavejs
2. Use TCP connection served by ser2net on another machine in the network
3. try to start your docker and load the nodes
4. See error

### Expected behavior

Would expect just to see the version numbers updated and all nodes visible

### Additional context

When I downgraded docker to a previous image, all start working again. I edited my docker-compose.yml to a specific image id (had to look that up running docker image ls). Doing that for the time being now, but expect to be able to upgrade to latest version... 
Everything works on version 8.4.1 with zwavejs2mqtt 5.8.0.

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/7577137/zwavejs2mqtt-store.zip)

AlCalzone, 2021-11-22:
The posted log doesn't look like it even gets to the soft reset. @Aesculapius do you do anything special on the Pi with the USB stick or just running `ser2net` directly? What's your `ser2net` config?

Aesculapius, 2021-11-22:
Nope, nothing special. Ser2Net config is as follows in /etc/ser2net.conf:

```
#Z-Wave USB Stick
3334:raw:0:/dev/ttyACM0:115200 8DATABITS NONE 1STOPBIT
```
[image]

closed 2021-11-22 by Aesculapius