zwave-js/zwave-js #3781: Losing my mind. Locked out. Conexis L1 lock with Yale Z-wave Module 2 constantly dead

user897943, 2021-11-22:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
tapping lock or unlock after adding device

What do you observe?
remove node from network... restart host system... add node... it adds as s0_legacy only. Appears in HA. Tap lock or unlock, it does the command. Walk away. A couple of minutes later: node is dead. Lock battery 100%

What did you expect to happen?
the node must not go dead. I was using with openzwave and it was fine. Stopped using zwave for a while, came back today and it's no longer working as it was before.

Steps to reproduce the behavior:
  1. Restart raspi4 - using the aeotec stick 5 (version compatible with pi4)
  2. Click on  option to include node, select s0 since s2 fails
  3. node is added... go to HA, click option to either lock or unlock, it works first time, then node goes dead

I also removed the integration from HA, deleted the container, removed the config files from the bind mount for the container, recreated from scratch, and made sure to reset the stick using a pin on the reset hole until it stopped flashing

### Device information

Manufacturer:  Aeotec Z-stick 5 (rpi4 compatible). Yale Zwave Module 2.
Node ID in your network: stick =  node 1. yale module = node 2

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 5.12.0
zwave-js: 8.7.6
home id: 3251038448
home hex: 0xc1c6e8f0

container created like this:
```
docker create --name zwavejs2mqtt --restart always -it -p 8091:8091 -p 3000:3000 --device=/dev/serial/by-id/usb-0658_0200-if00:/dev/zwave -v /home/pi/DockerBinds/zwavejs2mqtt:/usr/src/app/store zwavejs/zwavejs2mqtt:latest && docker start zwavejs2mqtt
```

### Did you change anything?

no

### If yes, what did you change?

I installed the Yale Access module (which allows the Conexis L1 lock to be used with the August app (or Yale Access as it's known in Europe). After that, I used a module provided by Yale to restore the firmware to the original that came with the lock, however this particular module was provided before Yale Access was available so I don't know if the lock firmware is corrupted or if the issue is with ZwaveJS.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

pretty sure it was working several months ago. I also know it worked fine with openzwave but that's now deprecated.

### Attach Driver Logfile

[zwavejs_2021-11-22.log](https://github.com/zwave-js/node-zwave-js/files/7584859/zwavejs_2021-11-22.log)
[zwavejs2mqtt_2021-11-22.log](https://github.com/zwave-js/node-zwave-js/files/7584860/zwavejs2mqtt_2021-11-22.log)

I will send an email with the keys.

user897943, 2021-11-22:
Email sent.

user897943, 2021-11-22:
I deleted zwavejs2mqtt container AND the integration on home assistant, then added OpenZwave container "openzwave/ozwdaemon:latest", configured with the network key environment variable that I had saved from the past, cleaned the bind mount and rebooted my host (pi4), then added the node using OpenZwave admin GUI on Windows, asked if I wanted to add with encryption, I responded yes, it added the node to OpenZwave flawlessly and it's working perfectly so the problem is DEFINITELY with ZwaveJS2Mqtt. It's a lock, I need it working 100% of the time. It's a shame that HA is deprecating OpenZwave because despite being stale it's also stable. This project is not stable. Hopefully you can resolve with the logs, then I would be able to try again.

user897943, 2021-11-23:
Lock has been rock solid overnight on openzwave, so the issue is definitely on ZwaveJS.

sotatech, 2021-11-23:
Last weekend I wiped my 8 years old OZW network and started afresh with Z-WaveJS running under Docker on Ubuntu. The process was slow but apart from an issue with my Fibaro FGRGBW 441 light controllers, everything went amazingly smoothly.
In particular, the Yale Conexis L1 lock is now working much better and so far has caused no issues. This is how I set it up:

I removed it from the old network, then ran Manage Nodes - Inclusion - selected Default and ticked Force Security. On the lock I then pressed the button three times and after a brief wait, I got a prompt asking for the first five digits on the lock Z-Wave module. After entering this, the lock paired up and I was good to go.

For comparison, on the Z-WaveJS control panel the lock is showing Security, Beaming and Z-Wave+ all green and Z-Wave+ is showing 1. Firmware version is 33.16 and everything else is green.

I had to attempt the inclusion several times, but I think this was due to my not pressing the button correctly. Others have reported that you need to bring the controller near the lock to ensure success, but I didn't experience this. However, it might be worth trying this in case you are experienceing interference issues.

I hope this helps.

user897943, 2021-11-23:
Thank you. I have done as you said. It made no difference. I also tried
with the first option and was asked to put the underlined first characters
on the sticker they glued to the module.

It doesn't work. It's working flawlessly with openzwave. Every time I
ensured to clear my config files and re-install the container, I also reset
the controller/stick.

Openzwave is controlling super quick and I get immediate status updates of
lock/unlock.

sotatech, 2021-11-23:
The only other thing I can think of is that you have an older version of the Z-Wave module in the lock. It's unlikley as it has a bug which drains the battery in a few weeks. It happened to me when I bought the lock, but Yale replaced the module free of charge.

If you haven't already, have a look here: [https://community.home-assistant.io/t/yale-z-wave-module-2-and-usb-z-wave-stick/64073/303](url) It has a lot of info on the lock.

user897943, 2021-11-23:
I have the Zwave 2 module, it was running on the lock for 2 years without a hitch, with openzwave... so definitely not that!? I contacted Yale this morning, they are sending me a replacement module, we'll see, but I highly doubt that it's the module if openzwave is making it work. Thanks.

AlCalzone, 2021-11-23:
@user897943 Your lock is included with Security S0, which means that before sending each command, a temporary key must be exchanged, tripling the amount of commands necessary for each communication attempt.
This fails multiple times because the lock either does not get that request or its acknowledgement doesn't reach the controller. OZW cannot really do anything differently here, so I'll assume that you just didn't notice the commands fail.

I'd do two things:
**First** make sure you have no connectivity issues.  Z-Wave sticks in particular are prone to interference by USB ports, especially by USB3 ports. We recommend putting the stick in a suitable location:

-   on an **USB extension cord** (this works wonders!)
-   away from other USB ports
-   away from metallic surfaces
-   and especially not in the back of a server rack

**Second**, once you've done that, re-include your lock with Security S2. This is automatically chosen when using the *default* inclusion strategy. S2 requires much less overhead than S0.

user897943, 2021-11-23:
> @user897943 Your lock is included with Security S0, which means that before sending each command, a temporary key must be exchanged, tripling the amount of commands necessary for each communication [...]

Ok, I'll respond one by one but first:

1. OpenZwave works. It paired first thing last night, and hasn't dropped once. Node hasn't shown dead once. " IT JUST WORKS! "
2. ZwaveJS - Using S2 does not work. It doesn't even finish pairing. Keeps stuck on NodeInfo stage. S0_Legacy is the only mode that allows it to even become available. And most important " IT DOES NOT WORK! "

The stick is on the same spot, using the same cable, connected to the same raspi. It works with OpenZwave, it does not work with ZwaveJS, so it's not an issue with me, or the stick, or the raspi, or my module. It's an issue with ZwaveJS. You'll find it very hard to convince me otherwise when one controller works without trouble while the other doesn't.

Telling me about overheads goes over my head. If you're telling me that OpenZwave has overhead because it's using S0 but it works, while ZwaveJS with S2 has less overhead but DOESN'T WORK, who cares about the overhead (I'm saying in the nicest way possible). All I care about (and any other user) is that it pairs without pain, and that it continues to work flawlessly.

Please don't pass the burden to the user when they can prove a use case that is currently operating on another implementation. This is an issue with ZwaveJS plain and simple.

AlCalzone, 2021-11-23:
> Using S2 does not work. It doesn't even finish pairing. Keeps stuck on NodeInfo stage.

Please post a driver log where I can see that.

> Again, it works perfectly with OpenZwave.

Mind posting a log from OZW where I can see the difference?

The log you posted above is full with this:
```
2021-11-22T22:58:42.293Z DRIVER « [REQ] [SendData]
                                    callback id:     130
                                    transmit status: NoAck
```
which is an information **from the stick** that it did not receive a response from the node. It doesn't matter which software is talking to the stick. If the stick doesn't get a response, the stick doesn't get a response.

> You'll find it very hard to convince me otherwise when one controller works without trouble while the other doesn't.

I can just work off the information in the log and the log tells me there's trouble communicating with the device. 

---

Oh and please stay respectful. I do this for free, I'm happy to help, but if it feels like that isn't appreciated, I can spend my time more productively.

sotatech, 2021-11-23:
The lock certainly does work with Z-WaveJs, as my experience shows. In fact, it works much better than on the old OZW integration. If the lock and the software are the same in both cases, then the only variable is your environment - hardware (I use a NUC not a Pi); distance from device to controller; interference or else a hardware fault.
Start with the logs and work from there.

user897943, 2021-11-23:
> Please post a driver log where I can see that.
>
>
> Mind posting a log from OZW where I can see the difference?
>
> The log you posted above is full with this:
>
> ```
> 2021-11-22T22:58:42.293Z DRIVER « [REQ] [...]

First I appreciate your help, don't think I take it for granted. I already have a working solution through openzwave so I'm here trying to other the project as well as other users of this lock who might have the same issue. I'm also giving my time to this project by reporting and providing bugs as well as discussing the environment. Lets put all ego aside and work on facts only. I felt a bit patronized to be told that it's an issue caused by everything but the software, I mean come on, please. We already know that openzwave works fine with it, nothing has been changed but the software, so obviously it's the software, not me, or the controller, or the end device.

As for the logs from ozw, please find attached.

Please show me instructions on how to get "driver log", I can't figure this out on my own.

Thank you
[ozw_log3.txt](https://github.com/zwave-js/node-zwave-js/files/7588550/ozw_log3.txt)
.

AlCalzone, 2021-11-23:
> Please show me instructions on how to get "driver log", I can't figure this out on my own.

You already posted one earlier, its the file `zwavejs_<date>.log`

AlCalzone, 2021-11-23:
Your OZW log shows the same behavior I'm seeing in the log from Z-Wave JS. It's a bit harder to read than our logs, but what happens is that when you try to unlock the door, OZW requests a nonce (the one-time key for encryption). After it does not get a response twice, it gives up:
```
2021-11-23 08:36:04.563303529  [20211123 8:36:04.563 UTC] [ozw.library] [info]: Info - Node: 2 Value::Set - COMMAND_CLASS_DOOR_LOCK - Locked - 0 - 1 - False
2021-11-23 08:36:04.563564543  [20211123 8:36:04.563 UTC] [ozw.library] [info]: Info - Node: 2 ValueID_Index_DoorLock::Lock::Set - Requesting lock to be Unlocked
2021-11-23 08:36:04.563751984  [20211123 8:36:04.563 UTC] [ozw.library] [debug]: Detail - Node: 2 Setting Encryption Flag on Message For Command Class COMMAND_CLASS_DOOR_LOCK
2021-11-23 08:36:04.563947277  [20211123 8:36:04.563 UTC] [ozw.library] [debug]: Detail - Node: 2 Queuing (Send) DoorLockCmd_Set (Node=2): 0x01, 0x0a, 0x00, 0x13, 0x02, 0x03, 0x62, 0x01, 0x00, 0x25, 0x6a, 0xcb
2021-11-23 08:36:04.564154921  [20211123 8:36:04.564 UTC] [ozw.library] [debug]: Detail - Node: 2 Setting Encryption Flag on Message For Command Class COMMAND_CLASS_DOOR_LOCK
2021-11-23 08:36:04.564372028  [20211123 8:36:04.564 UTC] [ozw.library] [debug]: Detail - Node: 2 Queuing (Send) DoorLockCmd_Get (Node=2): 0x01, 0x09, 0x00, 0x13, 0x02, 0x02, 0x62, 0x02, 0x25, 0x6b, 0xcb
2021-11-23 08:36:04.564682893  [20211123 8:36:04.564 UTC] [ozw.mqtt.commands] [info]: Processed Message for  "setvalue" :  "{\"ValueIDKey\": 39354384, \"Value\": false}"
2021-11-23 08:36:04.565818965  [20211123 8:36:04.565 UTC] [ozw.library] [info]: Info - Node: 2 Processing (Send) Nonce Request message (Callback ID=0x6a, Expected Reply=0x04)
2021-11-23 08:36:04.565833687  [20211123 8:36:04.565 UTC] [ozw.library] [info]: Info - Node: 2 Sending (Send) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(DoorLockCmd_Set) - 0x01, 0x09, 0x00, 0x13, 0x02, 0x02, 0x98, 0x40, 0x05, 0x02:
2021-11-23 08:36:05.566863254  [20211123 8:36:05.565 UTC] [ozw.library] [critical]: Error - Node: 2 ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "DoorLockCmd_Set (Node=2): 0x01, 0x0a, 0x00, 0x13, 0x02, 0x03, 0x62, 0x01, 0x00, 0x25, 0x6a, 0xcb"
2021-11-23 08:36:05.566878217  [20211123 8:36:05.566 UTC] [ozw.library] [debug]: Detail - Node: 2 Removing current message
2021-11-23 08:36:05.566885661  [20211123 8:36:05.566 UTC] [ozw.library] [debug]: Detail - Node: 2 Notification: Notification - TimeOut
2021-11-23 08:36:05.566892365  [20211123 8:36:05.566 UTC] [ozw.library] [info]: Info - Node: 2 Processing (Send) Nonce Request message (Callback ID=0x6b, Expected Reply=0x04)
2021-11-23 08:36:05.566988604  [20211123 8:36:05.566 UTC] [ozw.notifications] [debug]: Notification pvt_ozwNotification NotificationTypes::Notification_Code_MsgTimeout Thread:  0xb3c91010
2021-11-23 08:36:05.567385504  [20211123 8:36:05.567 UTC] [ozw.mqtt.publisher] [debug]: Publishing Event ozwNotification
2021-11-23 08:36:05.567680536  [20211123 8:36:05.567 UTC] [ozw.library] [info]: Info - Node: 2 Sending (Send) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(DoorLockCmd_Get) - 0x01, 0x09, 0x00, 0x13, 0x02, 0x02, 0x98, 0x40, 0x05, 0x02:
2021-11-23 08:36:06.569272519  [20211123 8:36:06.568 UTC] [ozw.library] [critical]: Error - Node: 2 ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "DoorLockCmd_Get (Node=2): 0x01, 0x09, 0x00, 0x13, 0x02, 0x02, 0x62, 0x02, 0x25, 0x6b, 0xcb"
2021-11-23 08:36:06.569289037  [20211123 8:36:06.568 UTC] [ozw.library] [debug]: Detail - Node: 2 Removing current message
2021-11-23 08:36:06.569296833  [20211123 8:36:06.568 UTC] [ozw.library] [debug]: Detail - Node: 2 Notification: Notification - TimeOut
2021-11-23 08:36:06.569496237  [20211123 8:36:06.569 UTC] [ozw.notifications] [debug]: Notification pvt_ozwNotification NotificationTypes::Notification_Code_MsgTimeout Thread:  0xb3c91010
```

I'm also seeing this 5 times:
```
2021-11-23 08:37:08.776417093  [20211123 8:37:08.776 UTC] [ozw.library] [info]: Info - Node: 2 WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
2021-11-23 08:37:08.776574572  [20211123 8:37:08.776 UTC] [ozw.library] [warning]: Warning - Node: 2 WARNING: Device is not a sleeping node.
```
which is exactly the same situation as the Z-Wave JS logs indicate.

TL;DR: This confirms my original suspicion. You do have a connectivity issue and you didn't notice it with OZW because it doesn't surface these issues as well as Z-Wave JS does.

user897943, 2021-11-23:
But if I have a connectivity issue, why is my lock working perfectly with their controller and not yours? NOTHING changes, just the controller. The only different is, when it works I am using openzwave, when it doesn't work I'm using ZwaveJS. This is what baffles me, why are we focusing on the logs and ignoring the REAL WORLD? I understand you "can only go by what you have" but don't you find it absolutely weird that it works perfectly on the other and not on this one? You don't feel like it raises any questions and you don't want to find out? On openzwave the inclusion happens within seconds of me pressing the R button on the lock three times within 1.5 seconds. It appears on the OpenZwave Admin app even with the picture of the lock so it fully recognizes.

On ZwaveJS it's taking more than a minute, and then I see this:

[image]

I try to re-interview, then the same.

You're ignoring the empirical data here when it's a huge question, WHY is one working and the other is not working even when you get the same errors? Obviously the one that works is better than the one that doesn't. It's doesn't matter if you buy a Ferrari if the engine doesn't run when the Volkswagen is taking you where you want to go.

ZwaveJS:
[image]
[image]
[image]
[zwavejs_2021-11-23.log](https://github.com/zwave-js/node-zwave-js/files/7589020/zwavejs_2021-11-23.log)
[zwavejs2mqtt_2021-11-23.log](https://github.com/zwave-js/node-zwave-js/files/7589021/zwavejs2mqtt_2021-11-23.log)

OpenZwave:
[image]
[image]
[image]
[image]
had to remove node then re-add but appears normally
[image]
[image]
[image]
And you can see above it unlocks.

AlCalzone, 2021-11-23:
> But if I have a connectivity issue, why is my lock working perfectly with their controller and not yours?

Ok explain me something: How can the lock work in OZW if it cannot even send the command to unlock because it doesn't have the necessary encryption code? Do you only look at the control panel or do you check if the lock physically responded to the command?
The snippet from your log I posted above clearly shows that the command was **NOT** sent.

Your screenshots (under which you wrote "had to remove node then re-add") even show that there are problems in OZW:
[image]

> On ZwaveJS it's taking more than a minute, and then I see this:

I'm not sure at which point the OZW tool displays the node information, but it seems like it shows that information as soon as possible. Zwavejs2mqtt only does so at the end of the entire interview, which can take a while. Also, OZW relies heavily on their config files while Z-Wave JS tries to query as much info as possible directly from the device, excerbating your problem.

From your new log:
```
2021-11-23T14:42:18.226Z DRIVER » [Node 002] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      5
                                  └─[SecurityCCSchemeGet]
2021-11-23T14:42:18.239Z SERIAL « [ACK]                                                                   (0x06)
2021-11-23T14:42:18.242Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-23T14:42:18.244Z SERIAL » [ACK]                                                                   (0x06)
2021-11-23T14:42:18.246Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-23T14:42:20.511Z SERIAL « 0x01180013050000e300bf7f7f7f7f0101060000000042020000eb              (26 bytes)
2021-11-23T14:42:20.513Z SERIAL » [ACK]                                                                   (0x06)
2021-11-23T14:42:20.514Z DRIVER « [REQ] [SendData]
                                    callback id:     5
                                    transmit status: OK
2021-11-23T14:42:28.232Z CNTRLR   [Node 002] Security S0 bootstrapping failed, the node was not granted the S0 s
                                  ecurity class: a secure inclusion timer has elapsed.
```

Or in plain english, this is the start of the secure inclusion process:
- Z-Wave JS asks the lock for its security schemes
- the command was sent
- the command was received by the node
- ...
- no response with 10 seconds, Z-Wave JS aborts the secure inclusion

This means your lock won't communicate securely and is thus not usable until you exclude it and include it again.

AlCalzone, 2021-11-23:
Again - and I'm not trying to be patronizing: you have a connectivity issue. Arguing and denying won't help.
When you're ready to accept that and trust my experience with this, let me know.

user897943, 2021-11-23:
> Ok explain me something: How can the lock work in OZW if it cannot even send the command to unlock because it doesn't have the necessary encryption code? Do you only look at the control panel or do [...]

The lock physically works! I press unlock, it unlocks (beeps, shows green LED, pulls the deadbolt into the door). And if I press lock, it flashed red, beeps, and pushes the deadbolt out. It does what you would expect. So if you're surprised, that's good! Because it shouldn't be happening but IT IS.

I accept that you think it's a connectivity issue from the logs, but I ask again, why is the lock working perfectly with openzwave? You want me to shoot a video? I totally can. I will show inclusion through to operation. Despite all the errors, it works. On the screenshots above when it was showing as Node 5, it was not working. So I did the exclusion and then restarted my openzwave container and made the inclusion, within seconds it was operating on HA, and still is.

AlCalzone, 2021-11-23:
How often did you try? The thing with connectivity issues is that they are often intermittent. It works for a while, then it stops working for a few seconds, then it works again, etc...

> On the screenshots above when it was showing as Node 5, it was not working. 

This should be enough to tell you that it isn't working reliably. Operate it from OZW until a new MsgTimeout appears in the log, then go and check.

user897943, 2021-11-23:
I made a video, I'm uploading. BRB

user897943, 2021-11-23:
> How often did you try? The thing with connectivity issues is that they are often intermittent. It works for a while, then it stops working for a few seconds, then it works again, etc...
> 
> > On the screenshots above when it was showing as Node 5, it was not working.
> 
> This should be enough to tell you that it isn't working reliably. Operate it from OZW until a new MsgTimeout appears in the log, then go and check if the last command did anything.

Let me satisfy your inner nerd, say it works 99% of the time. That's better than not working 100% of the time. Every time I issued a command it has worked fine.

user897943, 2021-11-23:
Video: openzwave working perfectly... https://youtu.be/rhd6QyCGa1Y

user897943, 2021-11-23:
I slightly moved the position of the stick and now it added as S2, rather than S0_Legacy as it was doing before. I will monitor the situation and report back. If I was wrong, then I stand corrected and thank you for your patience and perseverance. What is hard to understand is why OpenZwave is able to operate under the exact same circumstance while ZwaveJS fails. Maybe the timeouts are different? Maybe ZwaveJS is more sensitive to timeouts/more aggressive?

AlCalzone, 2021-11-23:
> What is hard to understand is why OpenZwave is able to operate under the exact same circumstance while ZwaveJS fails

Let me try to paint a picture using the car allegory you used before:

You own a car (the lock). The make doesn't matter, it's a car. It drives.
You also employ a driver (the Z-Wave stick).
Unfortunately the driver only speaks chinese, so you hire two personal assistants that speak chinese (OZW and Z-Wave JS).

**First, OZW:**
You tell the assistant (OZW) to tell the driver to accelerate. The driver pushes down the pedal, but the car doesn't move.
Driver: "The car doesn't react. Also the check engine light is on"
Assistant: "Try again"
Driver: "I did, the car doesn't react. Also the check engine light is still on"
Assitant writes in his notebook: "Something didn't work" 🤷🏻‍♂️

**Second, Z-Wave JS:**
You tell the assistant (OZW) to tell the driver to accelerate. The driver pushes down the pedal, but the car doesn't move.
Driver: "The car doesn't react. Also the check engine light is on"
Assistant: "Try again"
Driver: "I did, the car doesn't react. Also the check engine light is still on"
Assitant to you: "The car doesn't react. Also the check engine light is still on. I think the car is dead".

user897943, 2021-11-24:
> > What is hard to understand is why OpenZwave is able to operate under the exact same circumstance while ZwaveJS fails
> 
> Let me try to paint a picture using the car allegory you used before:
> 
> You own a car (the lock). The make doesn't matter, it's a car. It drives. You also employ a driver (the Z-Wave stick). Unfortunately the driver only speaks chinese, so you hire two personal assistants that can speak chinese (OZW and Z-Wave JS).
> 
> **First, OZW:** You tell the assistant (OZW) to tell the driver to accelerate. The driver pushes down the pedal, but the car doesn't move. Driver: "The car doesn't react. Also the check engine light is on" Assistant: "Try again" Driver: "I did, the car doesn't react. Also the check engine light is still on" Assitant writes in his notebook: "Something didn't work" 🤷🏻‍♂️
> 
> **Second, Z-Wave JS:** You tell the assistant (OZW) to tell the driver to accelerate. The driver pushes down the pedal, but the car doesn't move. Driver: "The car doesn't react. Also the check engine light is on" Assistant: "Try again" Driver: "I did, the car doesn't react. Also the check engine light is still on" Assitant to you: "The car doesn't react. Also the check engine light is still on. I think the car is dead".

Thanks but reading the above, it sounds like you think the lock does not work (doesn't move when pedal is pressed) with openzwave when actually, the lock works despite the logs saying that the command wasn't received... So unless you believe in ghosts, there is something that openzwave is doing differently. Maybe like I suggested above, the timeouts are more aggressive in ZwaveJS? Or maybe it doesn't retry as many times? The lock absolutely functions when I give the command to lock and unlock using openzwave.

I am very happy that the lock is working on ZwaveJS now after moving the stick an inch, but I don't think this issue should be closed just yet. There is something that allows openzwave to control the device where zwavejs fails under the same scenario and I think if you want to make a better product than the one that was abandoned by their developer, you need to at least match its reliability.

I'm not criticising by any means! I'm saying openzwave was able to control the lock (despite the logs suggesting otherwise) and zwavejs was not able, so clearly there is something that we can learn from openzwave here. The most important thing is that the device does what the user expects and with openzwave it was doing it, i.e it locked when I told it to lock, and it unlocked when I told it to unlock, I never had to check logs because it was simply doing what I expected. So feel free to close the issue but I think it's a missed opportunity to understand why the abandoned implementation made it work while your implementation didn't.

AlCalzone, 2021-11-24:
> Thanks but reading the above, it sounds like you think the lock does not work

That was for lack of a better allegory. I'm not saying the lock doesn't work. I'm saying the communication between the stick and the lock is not reliable. Communication in Z-Wave is more like a bidirectional game of [Chinese whispers](https://en.wikipedia.org/wiki/Chinese_whispers): You talk to the software, the software talks to the stick, the stick tries to talk to the nodes, the nodes try to talk to the stick, the stick talks back to the software and the software does or does not talk back to you.

> There is something that allows openzwave to control the device where zwavejs fails under the same scenario

Both fail because of the unreliable connection. It is just more visible in Z-Wave JS because of two reasons:
1. It updates its node status to "dead" when continuous commands fail and it makes this very obvious, unless OZW (and the softwares built on it) which tend to swallow these kinds of problems silently
1. It sends many, many more commands to the lock while interviewing it, therefore it has many more chances to fail. Unlike OZW, Z-Wave JS does not rely only on its configuration files to know what a device does, but instead requests every single bit of information it can gather from the device. If the connection is 95% reliable (I'm guessing lower), then of the roughly (just an estimation) 300 messages that are exchanged during the interview, **15** (!) are not going to reach their target. OZW probably sends like 10-20 messages, meaning you'll have at most 1 failure to be expected during the interview (and it is over faster).

> I'm saying openzwave was able to control the lock (despite the logs suggesting otherwise)

I still don't agree. Why should OZW write in the log that it didn't send a command when it actually did? That would be a nightmare to diagnose when all one has to go by (including the OZW maintainer) are logs. Maybe it worked most of the time (including the times you checked), but it didn't always work.

> I think it's a missed opportunity to understand why the abandoned implementation made it work while your implementation didn't.

Maybe you don't believe me, but there's nothing more to understand. The underlying issue was of a physical nature, one the software doesn't have any influence on. Both softwares are limited by the exact same methods to communicate with a device (by telling the stick to send a command and waiting for the result), so there simply cannot be any actual difference.

> I am very happy that the lock is working on ZwaveJS now after moving the stick an inch

Glad that it functions properly now. By the way, I'm working on methods to measure the connection quality between devices, which should make it easier to diagnose these kinds of things in the future.

closed 2021-11-24 by AlCalzone

user897943, 2021-11-24:
@AlCalzone 

"Both fail because of the unreliable connection. It is just more visible in Z-Wave JS because of two reasons:"

Actually they don't both fail. Only ZwaveJS fails... I judge failure by "does the lock work when I issue a command". You are judging failure by logs. We are measuring success and failure differently. When you start judging success by what actually happens in the physical world rather than the logs then you will see from my perspective. From a user's perspective, I don't care about the logs. All I care about is: I press a switch, does the light turn on? If yes, then it's working. Same principle. I send a command to unlock the door, does the lock unlock? Yes. Then it's working.

I'm not sure where the message is confusing. When I use openzwave the lock actually DOES work. It DOES NOT fail. It doesn't matter what the logs say, forget about the logs please. I am saying when I send a command to unlock it DOES UNLOCK. When I send a command to lock it DOES LOCK on openzwave. It doesn't matter what the logs say, physically the lock is doing what I am asking it to do. You are focusing on the logs, FORGET THE LOGS please. I am focusing on what actually happens and I can see with my eyes. I press Lock, the door lock locks. I press unlock, the door lock unlocks. Even when the logs show a communication failure. It still works. It does not fail.

The issue here is that you are judging failure or success by logs, where I am judging failure or success by what the lock actually does. What the lock is doing with openzwave does not correspond to what the logs are saying. I made a video, how could you not see it!?!?!!? You SAW the video with your own eyes, I was using openzwave and when I press lock, the lock reacts instantly. I press unlock, the lock reacts instantly. Forget the logs please, put your developer brain aside for a bit and look at what is actually happening.

There is something in openzwave that allows the lock to function. If you don't want to find out then that's fine, we can walk away from the discussion and leave it. I think if they are able to make it be more compatible than yours, surely you would be interested to know how they are doing it!? I am willing to test more but I can see you closed this, thanks for your help anyway.

AlCalzone, 2021-11-24:
I really shouldn't be having this conversation, but on the off-chance that there's something to be improved... I'll have to focus on the logs though, so please bear with me.

> Even when the logs show a communication failure. 

Both softwares retry when a command fails. I can't clearly see it in your video, but did you manage to create a situation where OZW retried a single command multiple times and the lock still reacted? To be clear, I'm looking for this:
1. You press lock or unlock in the OZW tool
2. The OZW tool now creates **multiple** entries "MsgTimeout"
3. The lock reacts

To figure out how often OZW retries, wrap the stick in aluminum foil or something before 1. and count the "MsgTimeout"s that appear once you lock/unlock. This test should fail, meaning the lock should not react.

Then recreate the situation without the foil, but with the same number of MsgTimeout, e.g. by moving the stick back to its original location, and retry the test.

user897943, 2021-11-24:
> I really shouldn't be having this conversation, but on the off-chance that there's something to be improved... I'll have to focus on the logs though, so please bear with me.
> 
> > Even when the logs show a communication failure.
> 
> Both softwares retry when a command fails, so it is possible that OZW just tries 3 times and it works after the 2nd or 3rd attempt.
> 
> I can't clearly see it in your video, but did you manage to create a situation where OZW retried a single command multiple times and the lock still reacted? To be clear, I'm looking for this:
> 
> 1. You press lock or unlock in the OZW tool
> 2. The OZW tool now creates **multiple** entries "MsgTimeout"
> 3. The lock reacts
> 
> To figure out how often OZW retries, wrap the stick in aluminum foil or something before 1. and count the "MsgTimeout"s that appear once you lock/unlock. This test should fail, meaning the lock should not react.
> 
> Then recreate the situation without the foil, but with the same number of MsgTimeout, e.g. by moving the stick back to its original location, and retry the test.

I totally understand why you said that you shouldn't be, and I appreciate that you're trying to see if something can be improved. You are awesome for having that attitude.

I am using Home Assistant, so I use the ozw integration in home assistant and that is where I issue lock and unlock, not on the ozw software. HA integration talks to ozw docker container and it in turn talks to the stick which talks to the module.

I can answer the question - yes, that is the case. It had multiple entries but eventually it reacted. But I remember there were three or four times when it didn't react at all, so I restarted the container and then it worked.

I will do the tin foil thing for you but it might be next week, I am going abroad later tonight so I won't have time to do it today.

AlCalzone, 2021-11-24:
> I am going abroad later tonight so I won't have time to do it today.

Alright, let me know.

> But I remember there were three or four times when it didn't react at all

That might just be one of the situations I meant. I might be on the wrong path here, but I'm thinking this is what happened:
1. Connection was bad
2. You triggered a command, OZW tried a few times but gave up
3. After the container restart the connection is better (coincidence)
4. it works again