zwave-js/zwave-js #3782: Minoston MP21ZP device incorrectly identified as MP21Z

ilyavolodin, 2021-11-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Inlusion of MP21ZP device.

What do you observe?

Minoston MP21ZP plug is incorrectly identified as MP21Z. Plug with "P" ending has power metering feature, which is MP21Z doesn't have. Since the device is incorrectly identified, power metering is missing.

What did you expect to happen?

I expect the device to be correctly identified and contain power metering feature.

### Device information

Manufacturer: Minoston
Model name: MP21ZP
Node ID in your network: 51

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.30.0
`node-zwave-js` branch: 
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2021-11-22.log](https://github.com/zwave-js/node-zwave-js/files/7585270/zwavejs_2021-11-22.log)

AlCalzone, 2021-11-23:
> Since the device is incorrectly identified, power metering is missing.

It doesn't work that way. Features are queried from the device itself, the label you're referring to is just cosmetical. This is what your device claims to support:
```
2021-11-23T00:53:58.900Z CNTRLR « [Node 051] received secure commands (S2_Unauthenticated)
                                  supported CCs:
                                  · Version
                                  · Binary Switch
                                  · Association
                                  · Multi Channel Association
                                  · Association Group Information
                                  · Manufacturer Specific
                                  · Device Reset Locally
                                  · Powerlevel
                                  · Configuration
                                  · Firmware Update Meta Data
```
As you can see, there's neither `Meter` nor `Multilevel Sensor` in the list, so no power metering capabilities.

ilyavolodin, 2021-11-23:
Ah, thanks for pointing that out. Apparently Amazon shipped me a wrong product. Even though I ordered an outlet with power metering, I can see that the one they sent me is actually PM21Z without that feature. Sorry about the noise.

closed 2021-11-23 by ilyavolodin