zwave-js/zwave-js #3787: Fibaro FGRGBW Exhibiting Strange Behaviour

sotatech, 2021-11-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Changing the values on any of the device channels.
What do you observe?
Toggling or changing some of the device entities causes changes in others, or they go unavailable.
What did you expect to happen?
To be able to control each channel individually and to see a colour wheel when the LEDs are on.

### Device information

Manufacturer: Fibaro
Model name: FGRGBW 441 F/W 22.22
Node ID in your network: 27 & 28

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:8.7.6
`zwavejs2mqtt` branch:5.12.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Did a factory reset on the device and then Replaced it. I have two of these devices, both show the same problem. I am using the WS server to talk to Home Assistant.
This is a clean install of Z-WaveJS with 38 devices that were excluded from the OZW integration and then included on Z-WaveJS. This is the only device that is having problems, so overall a very impressive result :).

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes, it worked with the old OZW integration.

Another uses is having the same problem and reported it here: [https://community.home-assistant.io/t/zwave-js-weird-behaviour-with-fibaro-fgrgbw-help/358324](url) which gives a clearer idea of what is happening.

### Attach Driver Logfile

[zwavejs_2021-11-23.zip](https://github.com/zwave-js/node-zwave-js/files/7589375/zwavejs_2021-11-23.zip)

AlCalzone, 2021-11-24:
I'm not sure what the actual problem is here. AFAIK the Fibaro RGBW (at least the first generation) has 6 channels controlling the brightness:
0 & 1: One of those toggles all channels, the other ... no idea - probably the same.
2-5: R, G, B, W each as a single channel.

In the linked video I only see the compound channel being toggled, which (as I would assume) correctly changes all channels.

sotatech, 2021-11-24:
Yes, this is a first generation device and I've been following the issues other people had with the colour wheel not appearing. This seems to have been resolved for MQTT users but I'm using the WS server.
I suspect most of what I am seeing is down to this problem, but what is not shown on the HA post is that all of the entities become unavailable for a few seconds when a change is made. I managed to capture it and you can see what happens below. After about 3 - 5 seconds they become available again. Also, each device has two Controller entities, one of which is permanently marked as unavailable, which is what let me to suspect the the devices were not being configured correctly.
I know this is a very old device and I will replace it if necessary with something newer, but I'm trying to avoid adding more stuff to my Home Automation "sin bin"!

[image]

AlCalzone, 2021-11-24:
It sounds like this is more of an issue with HomeAssistant (the UI) than with node-zwave-js (the thing talking to the Z-Wave stick).
I suggest you open your issue in https://github.com/home-assistant/core

sotatech, 2021-11-24:
Thanks @AlCalzone I found a similar issue here: [https://github.com/home-assistant/core/issues/53114](url) I will link this issue to it.

kpine, 2021-11-25:
@AlCalzone the dump file in the linked issue ([53144](https://github.com/home-assistant/core/issues/53114)) shows that the Blue `currentColor` value ID has no `value` field, but Red and Green do. How would that end up happening?

AlCalzone, 2021-11-25:
Each color component is queried separately during the interview. Maybe there was a timeout for blue. Refreshing the Color Switch CC values might help.

kpine, 2021-11-25:
It queries for Blue, but gets Green back??? (Linked issue has full log)

```
2021-11-25T16:43:24.871Z CNTRLR » [Node 028] querying current color state (Blue)
2021-11-25T16:43:24.874Z SERIAL » 0x010a00131c033303042506ee                                          (12 bytes)
2021-11-25T16:43:24.875Z DRIVER » [Node 028] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      6
                                  └─[ColorSwitchCCGet]
                                      color component: Blue
2021-11-25T16:43:24.877Z SERIAL « [ACK]                                                                   (0x06)
2021-11-25T16:43:24.883Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-25T16:43:24.884Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T16:43:24.884Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-25T16:43:24.905Z SERIAL « 0x011800130600000300bf7f7f7f7f01010300000000020100004e              (26 bytes)
2021-11-25T16:43:24.905Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T16:43:24.906Z DRIVER « [REQ] [SendData]
                                    callback id:     6
                                    transmit status: OK
2021-11-25T16:43:24.936Z SERIAL « 0x010c0004001c0433040300bf0064                                      (14 bytes)
2021-11-25T16:43:24.937Z CNTRLR   [Node 028] [~] [Color Switch] currentColor[3]: 0 => 0             [Endpoint 0]
2021-11-25T16:43:24.938Z CNTRLR   [Node 028] [~] [Color Switch] currentColor: {"green":0} => {"gree [Endpoint 0]
                                  n":0}
2021-11-25T16:43:24.939Z CNTRLR   [Node 028] [~] [Color Switch] hexColor: "000000" => "000000"      [Endpoint 0]
2021-11-25T16:43:24.940Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T16:43:24.941Z DRIVER « [Node 028] [REQ] [ApplicationCommand]
                                  └─[ColorSwitchCCReport]
                                      color component: Green
                                      current value:   0
```

kpine, 2021-11-25:
According to OZW, devices with firmware versions prior to 25.25 have a "color index bug". It looks like the color report is invalid.

```c++
/* Fibaro is messed up. It always returns 0x03 as the Coloridx in a report message, but the values are correct for what the request was.
 * I read a forum post about it being a bug. If this is a case, its going to make this class messy as hell :(
 *
 * http://forum.z-wave.me/viewtopic.php?f=3422&t=20042
 *
 * Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x34, 0x04, 0x33, 0x04, 0x03, 0xff, 0x0a
 *                                                                 ^^^^
 *                                                                 Always 0x03
 *                                                                       ^^^^
 *                                                                       But this appears to be the right value for the channel we requested
 *
 */
/* request the next color index if we are bugged */

```

Looking at the logs again, this appears to be true for these devices. Red is also reported as green (0x3).
```
2021-11-25T08:11:02.042Z CNTRLR » [Node 027] querying current color state (Red)
2021-11-25T08:11:02.046Z SERIAL » 0x010a00131b0333030225aa43                                          (12 bytes)
2021-11-25T08:11:02.046Z DRIVER » [Node 027] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      170
                                  └─[ColorSwitchCCGet]
                                      color component: Red
2021-11-25T08:11:02.048Z SERIAL « [ACK]                                                                   (0x06)
2021-11-25T08:11:02.055Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-25T08:11:02.056Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T08:11:02.056Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-25T08:11:02.076Z SERIAL « 0x01180013aa00000300b97f7f7f7f0101030000000002010000e4              (26 bytes)
2021-11-25T08:11:02.076Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T08:11:02.077Z DRIVER « [REQ] [SendData]
                                    callback id:     170
                                    transmit status: OK
2021-11-25T08:11:02.108Z SERIAL « 0x010c0004001b0433040300b50069                                      (14 bytes)
2021-11-25T08:11:02.109Z CNTRLR   [Node 027] [~] [Color Switch] currentColor[3]: 0 => 0             [Endpoint 0]
2021-11-25T08:11:02.110Z CNTRLR   [Node 027] [~] [Color Switch] currentColor: {"green":0} => {"gree [Endpoint 0]
                                  n":0}
2021-11-25T08:11:02.111Z CNTRLR   [Node 027] [~] [Color Switch] hexColor: "000000" => "000000"      [Endpoint 0]
2021-11-25T08:11:02.112Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T08:11:02.113Z DRIVER « [Node 027] [REQ] [ApplicationCommand]
                                  └─[ColorSwitchCCReport]
                                      color component: Green
                                      current value:   0
```

OZW has [a workaround for this issue](https://github.com/OpenZWave/open-zwave/commit/f174eb6027475b567d8b2b91f40f8aa921704fdd) as a compat flag. I don't think it sounds like something you want to do in node-zwave-js?

I would suggest upgrading the firmware for the devices to 25.25 or later. Both of the devices in this log have old firmware.

AlCalzone, 2021-11-26:
> I don't think it sounds like something you want to do in node-zwave-js? 

Nope.

> I would suggest upgrading the firmware for the devices to 25.25 or later. Both of the devices in this log have old firmware.

The problem is that Fibaro AFAIK does not offer firmware updates except through their HomeCenter (which one would have to buy). I'd contact the Fibaro support.

sotatech, 2021-11-26:
Understood, I know this is a very old device with some non-standard "features", so I'll try the Fibaro forum and report back.

sotatech, 2021-11-26:
I contacted Fibaro support and they confirmed that firmware files are not available for download. My choices were either use a Home Center (n/a), contact a Fibaro installer (none available in my country), or sent the devices to them in Poland.

It will cost about €35 in shipping costs, but there is no charge for the service and they pay return shipping. I have decided to go ahead with this option, just get a final resolution to the issue.

It will likley take a couple of weeks for the round trip, so I will let you know what happens when they arrive.

closed 2021-11-30 by zwave-js-assistant[bot]

sotatech, 2021-12-21:
I received my FGRGBW 441 devices back from Fibaro and they had updated the firmware from 22.22 to 27.27. They turned around the update pretty quickly but with shipping delays it still took about three weeks. The €35 cost of shipping made it just about worthwhile, but it's crazy in this day and age that a manufacturer cannot supply firmware updates for local installation. This would definitely put me off buying any more Fibaro devices.
Anyway, the good news is that the devices worked immediately after installation so it was a firmware bug. Thanks for all your assistance and here's the proof:

[image]

AlCalzone, 2021-12-21:
> The €35 cost of shipping made it just about worthwhile, but it's crazy in this day and age that a manufacturer cannot supply firmware updates for local installation. This would definitely put me off buying any more Fibaro devices.

100% agree here - this could just have been an email instead.

clowgg, 2023-12-05:
Can you choose and set the RGBW programs on the FGRGBW 441 from HA?