zwave-js/zwave-js #3788: No more action from zwavejs2mqtt to my iot devices

titidnh, 2021-11-24:
Dear,

Since xxx days (not really knows how many :(), I'm not able to switch on/off my iot devices. I received the informations from the device to home assistant via zwavejs but I can not turn on/off my iot devices directly from the dashboard of zwavejs2mqtt.

Any idea which action I need to take ?

2021-11-23T20:09:38.880Z CNTRLR [Node 010] [~] [Binary Switch] currentValue: true => false [Endpoint 0]
2021-11-23 21:09:38.880 INFO ZWAVE: Node 10: value updated: 37-0-currentValue true => false
2021-11-23T20:09:39.885Z SERIAL » 0x010900130a022502250ae5 (11 bytes)
2021-11-23T20:09:39.886Z DRIVER » [Node 010] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 10
└─[BinarySwitchCCGet]
2021-11-23T20:09:39.887Z SERIAL « [ACK] (0x06)
2021-11-23T20:09:39.894Z SERIAL « 0x0104011301e8 (6 bytes)
2021-11-23T20:09:39.894Z SERIAL » [ACK] (0x06)
2021-11-23T20:09:39.895Z DRIVER « [RES] [SendData]
was sent: true
2021-11-23T20:09:39.936Z SERIAL « 0x010700130a000004e5 (9 bytes)
2021-11-23T20:09:39.937Z SERIAL » [ACK] (0x06)
2021-11-23T20:09:39.938Z DRIVER « [REQ] [SendData]
callback id: 10
transmit status: OK
2021-11-23T20:09:49.943Z CNTRLR [Node 010] Timed out while waiting for a response from the node (ZW0201)

AlCalzone, 2021-11-24:
> Invalid auth code

Did you change the security keys in your configuration?

titidnh, 2021-11-24:
I removed the node and readd it ... it works fine. I will close discussion. 

Thanks to all

closed 2021-11-24 by titidnh