zwave-js/zwave-js #3790: Thermostat issue after update to 0.1.49

Danny2100, 2021-11-24:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? The recent zwave update to 0.1.49

What do you observe? The thermostat is not being read properly anymore by home assistant since the update. I noticed the temperature is not updating. The thermostat says it's 62 home assistant says 63. When I adjust the temperature on home assistant. The temperature is not the same as what it is on the thermostat.

What did you expect to happen? For the thermostat to update properly when anything changes or if a change is made by home assistant for home assistant to read the correct temperature.

My thermostat is a Honeywell t6 pro Z-Wave.

Steps to reproduce the behavior:
  1. Go to '...' 
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer:  Honeywell
Model name:  T6 Pro 
Node ID in your network: 43

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.49
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I updated it from 0.1.48

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It worked prior to the update.

### Attach Driver Logfile

Having trouble uploading it. I will email it.

Danny2100, 2021-11-25:
I restored the backup of 0.1.48 and everything is working again. It's something with 0.1.49. 

I'm on an odroid n2 running home assistant os and using a nortek stick. Completely up to date on everything. Supervisor, core and os.

AlCalzone, 2021-11-25:
The file you sent is not a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), so it doesn't tell me much. You should also be able to make one within HomeAssistant, put it on loglevel `debug` or `silly`, re-interview the problematic device and attach it here as a file through drag&drop.

Danny2100, 2021-11-25:
[debug 0.1.48.txt](https://github.com/zwave-js/node-zwave-js/files/7603521/debug.0.1.48.txt)
[Debug on 0.1.49.txt](https://github.com/zwave-js/node-zwave-js/files/7603523/Debug.on.0.1.49.txt)

I took logs before and after the update and left notes.

AlCalzone, 2021-11-25:
Ok this is caused by https://github.com/zwave-js/node-zwave-js/commit/6df1190c3347d03b16a18281a3e72ac798ad119a

Danny2100, 2021-11-25:
Sounds good I figured it was an update issue. I appreciate everything you do for zwave js.

AlCalzone, 2021-11-25:
fixed in 8.7.8 / 8.8.1

closed 2021-11-25 by AlCalzone