zwave-js/zwave-js #3793: No total consumption update on GreenWave sockets

Aohzan, 2021-11-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What do you observe?
Energy sensor (kWh total consumption) don't update, if I reset it with the service, it stays at 0.0 kWh

### Device information

Manufacturer: GreenWave
Model name: 2x GWPN1 and 1x NP240 / NP242
Node ID in your network: 8, 9 and 16

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.49:

    Bump Z-Wave JS to 8.7.7
    Bump Z-Wave JS Server to 1.12.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes, but I don't know when it was borken

### Attach Driver Logfile

Subscribed to Z-Wave JS Log Messages…
Log Level changed to: Debug
2021-11-25T17:43:47.921Z SERIAL « 0x0114000400080e320221340000024c00010000025bda                      (22 bytes)
2021-11-25T17:43:47.924Z CNTRLR   [Node 008] [~] [Meter] value[66049]: 65.4 => 58.8                 [Endpoint 0]
2021-11-25T17:43:47.927Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T17:43:47.931Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       W
                                      rate type:   Consumed
                                      value:       58.8
                                      time delta:  1 seconds
[... 190 lines cut ...]
                                  │ transmit options: 0x25
                                  │ callback id:      249
                                  └─[MeterCCGet]
                                      rate type: Consumed
                                      scale:     kWh
2021-11-25T17:46:43.303Z SERIAL « [ACK]                                                                   (0x06)
2021-11-25T17:46:43.311Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-25T17:46:43.314Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T17:46:43.320Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-25T17:46:43.332Z SERIAL « 0x01050013f90010                                                     (7 bytes)
2021-11-25T17:46:43.339Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T17:46:43.344Z DRIVER « [REQ] [SendData]
                                    callback id:     249
                                    transmit status: OK
2021-11-25T17:46:43.356Z SERIAL « 0x0114000400080e32022184000040850001000040847c                      (22 bytes)
2021-11-25T17:46:43.363Z CNTRLR   [Node 008] [~] [Meter] value[65537]: 0.0004 => 1.6517             [Endpoint 0]
2021-11-25T17:46:43.374Z SERIAL » [ACK]                                                                   (0x06)
2021-11-25T17:46:43.381Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       kWh
                                      rate type:   Consumed
                                      value:       1.6517
                                      time delta:  1 seconds

Aohzan, 2021-11-25:
[zwave_js_dump.json.txt](https://github.com/zwave-js/node-zwave-js/files/7604703/zwave_js_dump.json.txt)

AlCalzone, 2021-11-26:
If the device reports W on its own but never kWh, you'll probably need to set up polling. This is described [here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/polling) for zwavejs2mqtt - not sure how you can do it in HomeAssistant though.

Aohzan, 2021-11-26:
oh right, I have to call the service `zwave_js.refresh_value`, it works today. Thanks

closed 2021-11-26 by Aohzan