zwave-js/zwave-js #3796: Create symlink to latest logfile

kpine, 2021-11-26:
**Is your feature request related to a problem? Please describe.**

It's useful to be able to tail the driver log file and watch it for awhile, or grep for certain terms and leave it running overnight, in order to do troubleshooting. The problem is that the log file becomes unavailable when the rotation occurs. Then you need to start over with the new file name.

**Describe the solution you'd like**

Create a symlink that always points to the latest logfile. This would make it easier to analyze the latest log messages. Programs that watch the file via the symlink should be able to automatically start watching the new file.

My understanding is that the log file rotation is based on [winston-daily-rotate-file](https://github.com/winstonjs/winston-daily-rotate-file), which has these optional configuration options:

> - **createSymlink:** Create a tailable symlink to the current active log file. (default: false)
> - **symlinkName**: The name of the tailable symlink. (default: 'current.log')

So would it be as simple as turning those options on? (Not sure how this workings in Windows)

**Describe alternatives you've considered**

None

**Additional context**

N/A

AlCalzone, 2021-11-26:
Yes I think it is as easy as that.

kpine, 2022-01-04:
The linked PR doesn't seem to be working. In both my z2m and custom docker enviroment, the log file name is an absolute path. I added debug logging and it attempts to create a symlink with an absolute path to the same directory as the file:

```
****** creating symlink from zwavejs_2022-01-04.log to /usr/src/app/store//usr/src/app/store/zwavejs_current.log
```

The symlink name should therefore not include the path, only the symlink name. I replaced the symlink name in the constructor with:

```
symlinkName: path.basename(this.logConfig.filename).replace(`_%DATE%`, "_current"),
```

and it's working fine now.

AlCalzone, 2022-01-05:
🤔 for some reason, the winston transport uses string concatenation to determine the path instead of `path.join()`

AlCalzone, 2022-01-23:
fixed by https://github.com/zwave-js/node-zwave-js/pull/4039

PR #3977 (2021-12-30): feat: create symlink to current logfile
fixes:  #3796
closed 2022-01-23 by AlCalzone