zwave-js/zwave-js #3800: Configuration parameter invalid for ZW096

JvdBosch, 2021-11-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Setting the value for parameter 83 causes an error. This seems to have a relation with issue: https://github.com/zwave-js/node-zwave-js/issues/3036

Trying to run the same YAML as in issue 3036, leads to an error.
```
service: zwave_js.bulk_set_partial_config_parameters
target:
  entity_id: light.nachtlamp
data:
  parameter: '83'
  value:
    'Night Light: Red Color': 200
    'Night Light: Green Color': 100
    'Night Light: Blue Color': 50
```

What do you observe?
When changing any of the three night light paremeters in Home assistant, it fails and shows in the GUI:
**Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons**

No error is present in the Home Assistant Log however.
The Zwave log shows an error (see below). When trying to set 255,255,255 it does want to set -256, which is correct for this value:

Trying to set it manually with the above YAML:
Error in Home Assistant log:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 381, in async_bulk_set_partial_config_parameters
    cmd_status = await async_bulk_set_partial_config_parameters(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 148, in async_bulk_set_partial_config_parameters
    raise SetValueFailed(
zwave_js_server.exceptions.SetValueFailed: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
```
Error in Zwave log is similar to below.

What did you expect to happen?
The night light should change color.

Steps to reproduce the behavior:
  1. Go to the device.
  2. Click on Configure device.
  3. Scroll down to parameter 83 and try to change either one of the colors.
  4. See error

### Device information

Manufacturer: AEON Labs
Model name: ZW096 (I have a ZW099 which exhibits the same issue)
Node ID in your network: 11

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Version 0.1.49 (Z-Wave JS 8.7.7, Z-Wave JS Server 1.12.0)

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

With OZW this worked fine.

### Attach Driver Logfile

```
Error in driver ZWaveError: The value -256 is invalid for configuration parameter 83 (size = 4, format = UnsignedInteger)! (ZW0322)
    at throwInvalidValueError (/usr/src/node_modules/zwave-js/src/lib/commandclass/ConfigurationCC.ts:2441:8)
    at Proxy.ConfigurationCCAPI.<computed> (/usr/src/node_modules/zwave-js/src/lib/commandclass/ConfigurationCC.ts:399:4)
    at ZWaveNode.setValue (/usr/src/node_modules/zwave-js/src/lib/node/Node.ts:827:14)
    at Function.handle (/usr/src/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:44)
    at Object.node (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:39:91)
    at Client.receiveMessage (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:95:99)
    at WebSocket.<anonymous> (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:48:45)
    at WebSocket.emit (events.js:400:28)
    at WebSocket.emit (domain.js:475:12)
    at Receiver.receiverOnMessage (/usr/src/node_modules/ws/lib/websocket.js:1093:20) {
  code: 322,
  context: undefined,
  transactionSource: undefined
}
```

AlCalzone, 2021-11-29:
I've checked with the HomeAssistant devs. This is a bug on their side. HA should be using the [`setBulk`](https://zwave-js.github.io/node-zwave-js/#/api/CCs/Configuration?id=setbulk) method of the Configuration CC API instead of computing the param value itself.

kpine, 2021-11-29:
@JvdBosch Can you turn on debug logging in HA and see what value it's trying to send to node-zwave-js?

You can temporarily turn on debug logs with a [service call](https://www.home-assistant.io/integrations/logger/#service-set_level).

```yaml
service: logger.set_level
data:
  homeassistant.components.zwave_js: debug
  zwave_js_server: debug
```

The zwave-js-server messages will then be logged to the home-assistant.log file.

JvdBosch, 2021-11-29:
The Zwave log is the same as above.

This is the HA log:
```
2021-11-29 10:24:17 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140488655814816] Received {'type': 'zwave_js/set_config_parameter', 'entry_id': '1e15b0bb7325e18ffec82c1d5136bd2e', 'node_id': 11, 'property': 83, 'value': 255, 'property_key': 65280, 'id': 75}
2021-11-29 10:24:17 DEBUG (MainThread) [zwave_js_server] Publishing message:
{'command': 'node.set_value',
'messageId': '6a0622f51372409db02ad46149eef518',
'nodeId': 11,
'value': 255,
'valueId': {'ccVersion': 1,
'commandClass': 112,
'commandClassName': 'Configuration',
'endpoint': 0,
'metadata': {'allowManualEntry': True,
'default': 221,
'format': 1,
'isFromConfig': True,
'label': 'Night Light: Blue Color',
'max': 255,
'min': 0,
'readable': True,
'type': 'number',
'valueSize': 4,
'writeable': True},
'property': 83,
'propertyKey': 65280,
'propertyName': 'Night Light: Blue Color',
'value': 0}}
2021-11-29 10:24:17 DEBUG (MainThread) [zwave_js_server] Received message:
WSMessage(type=<WSMsgType.TEXT: 1>, data='{"type":"result","success":true,"messageId":"6a0622f51372409db02ad46149eef518","result":{"success":false}}', extra='')
2021-11-29 10:24:17 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140488655814816] Sending {"id": 75, "type": "result", "success": false, "error": {"code": "unknown_error", "message": "Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons"}}
```

AlCalzone, 2021-11-29:
This looks like only blue is being set. Try setting all values to something else first.

JvdBosch, 2021-11-29:
That is correct, I tried to set the blue to 255 (was 0).
[image]

I can only set one value at a time AFAIK. Setting all at the same time I tried with:
```
service: zwave_js.set_config_parameter
data:
  parameter: 83
  value: -256
target:
  entity_id: light.nachtlamp
```

But that fails as mentioned above.

AlCalzone, 2021-11-29:
We need to see what happens when you try to set the combined value (e.g. 200/100/50 or 255/255/255) like in the original report.

JvdBosch, 2021-11-29:
In the old OZW implementation I set the config values myself with zwave.set_config_parameter. I did use -256 in that case, as it equals to 0xFFFFFF, so the value is as I expect.

kpine, 2021-11-29:
The service call you originally posted, which you said is not working ("Trying to run the same YAML as in issue 3036, leads to an error."), will set all three values at the same time. Did you actually use this service? What is the log from running it, that's what we have been asking for.

```yaml
service: zwave_js.bulk_set_partial_config_parameters
target:
  entity_id: light.nachtlamp
data:
  parameter: '83'
  value:
    'Night Light: Red Color': 200
    'Night Light: Green Color': 100
    'Night Light: Blue Color': 50
```

If you aren't using the `zwave_js.bulk_set_partial_config_parameters` service call, then that changes things.

JvdBosch, 2021-11-29:
Thanks, now I understand what you mean. Yes, I tried with the above bulk command. It failed as well. I tried it again with the debug on to get the info you need. This is the result:

```
2021-11-29 20:02:51 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140054836233840] Received {'type': 'execute_script', 'sequence': [{'service': 'zwave_js.bulk_set_partial_config_parameters', 'target': {'entity_id': 'light.nachtlamp'}, 'data': {'parameter': '83', 'value': {'Night Light: Red Color': 200, 'Night Light: Green Color': 100, 'Night Light: Blue Color': 50}}}], 'id': 33}
2021-11-29 20:02:51 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Running websocket_api script
2021-11-29 20:02:51 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Executing step call service
2021-11-29 20:02:51 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zwave_js, service=bulk_set_partial_config_parameters, service_data=parameter=83, value=Night Light: Red Color=200, Night Light: Green Color=100, Night Light: Blue Color=50, entity_id=['light.nachtlamp']>
2021-11-29 20:02:51 DEBUG (MainThread) [zwave_js_server] Publishing message:
{'command': 'node.set_value',
 'messageId': '3efec220b6bc429f92f87c12314bc558',
 'nodeId': 11,
 'value': 3362009600,
 'valueId': {'commandClass': 112, 'property': 83}}

2021-11-29 20:02:51 DEBUG (MainThread) [zwave_js_server] Received message:
WSMessage(type=<WSMsgType.TEXT: 1>, data='{"type":"result","success":true,"messageId":"3efec220b6bc429f92f87c12314bc558","result":{"success":false}}', extra='')

2021-11-29 20:02:51 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 381, in async_bulk_set_partial_config_parameters
    cmd_status = await async_bulk_set_partial_config_parameters(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 148, in async_bulk_set_partial_config_parameters
    raise SetValueFailed(
zwave_js_server.exceptions.SetValueFailed: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
2021-11-29 20:02:51 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140054836233840] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 26, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 523, in handle_execute_script
    await script_obj.async_run(msg.get("variables"), context=context)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1243, in async_run
    await asyncio.shield(run.async_run())
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 353, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 381, in async_bulk_set_partial_config_parameters
    cmd_status = await async_bulk_set_partial_config_parameters(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 148, in async_bulk_set_partial_config_parameters
    raise SetValueFailed(
zwave_js_server.exceptions.SetValueFailed: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
```

kpine, 2021-11-29:
> Thanks, now I understand what you mean. Yes, I tried with the above bulk command. It failed as well. I tried it again with the debug on to get the info you need.

Thank you. And is the error message from the driver the same thing, is it saying -256 is invalid?

JvdBosch, 2021-11-29:
It says the following. It seems to want to set it to 3362009600, which is correct I think (for 200,100,50).
```
Error in driver ZWaveError: The value 3362009600 is not valid for configuration parameters! (ZW0322)
    at Proxy.ConfigurationCCAPI.<computed> (/usr/src/node_modules/zwave-js/src/lib/commandclass/ConfigurationCC.ts:389:11)
    at ZWaveNode.setValue (/usr/src/node_modules/zwave-js/src/lib/node/Node.ts:827:14)
    at Function.handle (/usr/src/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:44)
    at Object.node (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:39:91)
    at Client.receiveMessage (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:95:99)
    at WebSocket.<anonymous> (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:48:45)
    at WebSocket.emit (events.js:400:28)
    at WebSocket.emit (domain.js:475:12)
    at Receiver.receiverOnMessage (/usr/src/node_modules/ws/lib/websocket.js:1093:20)
    at Receiver.emit (events.js:400:28) {
  code: 322,
  context: undefined,
  transactionSource: undefined
}
```

kpine, 2021-11-29:
I think the value is correct. It seems like node-zwave-js having trouble figuring out the correct value format, so it defaults to SignedInteger?

JvdBosch, 2021-11-29:
When trying:
```
service: zwave_js.bulk_set_partial_config_parameters
target:
  entity_id: light.nachtlamp
data:
  parameter: '83'
  value:
    'Night Light: Red Color': 255
    'Night Light: Green Color': 255
    'Night Light: Blue Color': 255
```

These are the logs. No -256, but 4294967040.
```
2021-11-29 20:11:02 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140054836233840] Received {'type': 'execute_script', 'sequence': [{'service': 'zwave_js.bulk_set_partial_config_parameters', 'target': {'entity_id': 'light.nachtlamp'}, 'data': {'parameter': '83', 'value': {'Night Light: Red Color': 255, 'Night Light: Green Color': 255, 'Night Light: Blue Color': 255}}}], 'id': 107}
2021-11-29 20:11:02 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Running websocket_api script
2021-11-29 20:11:02 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Executing step call service
2021-11-29 20:11:02 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zwave_js, service=bulk_set_partial_config_parameters, service_data=parameter=83, value=Night Light: Red Color=255, Night Light: Green Color=255, Night Light: Blue Color=255, entity_id=['light.nachtlamp']>
2021-11-29 20:11:02 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 381, in async_bulk_set_partial_config_parameters
    cmd_status = await async_bulk_set_partial_config_parameters(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 148, in async_bulk_set_partial_config_parameters
    raise SetValueFailed(
zwave_js_server.exceptions.SetValueFailed: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
2021-11-29 20:11:02 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140054836233840] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 26, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 523, in handle_execute_script
    await script_obj.async_run(msg.get("variables"), context=context)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1243, in async_run
    await asyncio.shield(run.async_run())
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 353, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 381, in async_bulk_set_partial_config_parameters
    cmd_status = await async_bulk_set_partial_config_parameters(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 148, in async_bulk_set_partial_config_parameters
    raise SetValueFailed(
zwave_js_server.exceptions.SetValueFailed: Unable to set value, refer to https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue for possible reasons
```

```
Error in driver ZWaveError: The value 4294967040 is not valid for configuration parameters! (ZW0322)
    at Proxy.ConfigurationCCAPI.<computed> (/usr/src/node_modules/zwave-js/src/lib/commandclass/ConfigurationCC.ts:389:11)
    at ZWaveNode.setValue (/usr/src/node_modules/zwave-js/src/lib/node/Node.ts:827:14)
    at Function.handle (/usr/src/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:44)
    at Object.node (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:39:91)
    at Client.receiveMessage (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:95:99)
    at WebSocket.<anonymous> (/usr/src/node_modules/@zwave-js/server/dist/lib/server.js:48:45)
    at WebSocket.emit (events.js:400:28)
    at WebSocket.emit (domain.js:475:12)
    at Receiver.receiverOnMessage (/usr/src/node_modules/ws/lib/websocket.js:1093:20)
    at Receiver.emit (events.js:400:28) {
  code: 322,
  context: undefined,
  transactionSource: undefined
}
```

JvdBosch, 2021-11-29:
Testing in Python:
`int.from_bytes(bytes.fromhex("ffffff00"), byteorder = 'big')`
equals 4294967040.

So that adds up... I'm curious that -256 worked in my old ozw case.
`int.from_bytes(bytes.fromhex("ffffff00"), byteorder = 'big', signed = True)`
equals to -256. So maybe the mistake is that we are converting to unsigned integer?

kpine, 2021-11-29:
You can't compare to OZW, this is completely different software.

kpine, 2021-11-29:
Have you tried the bulk parameter service call with -256?

JvdBosch, 2021-11-29:
Setting -256 with the bulk call seems to work, no errors. However the light does not change. Maybe the value size?
```
2021-11-29T19:38:54.962Z DRIVER » [Node 011] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      176
                                  └─[ConfigurationCCSet]
                                      parameter #:      83
                                      reset to default: false
                                      value size:       2
                                      value format:     SignedInteger
                                      value:            -256
2021-11-29T19:38:54.968Z SERIAL « [ACK]                                                                   (0x06)
2021-11-29T19:38:54.973Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-29T19:38:54.974Z SERIAL » [ACK]                                                                   (0x06)
2021-11-29T19:38:54.975Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-29T19:38:54.992Z SERIAL « 0x01070013b000000259                                                 (9 bytes)
2021-11-29T19:38:54.993Z SERIAL » [ACK]                                                                   (0x06)
2021-11-29T19:38:54.994Z DRIVER « [REQ] [SendData]
                                    callback id:     176
                                    transmit status: OK
2021-11-29T19:38:55.000Z CNTRLR   [Node 011] [~] [Configuration] 83: 0 => -256
2021-11-29T19:38:55.001Z CNTRLR   [Node 011] Scheduled poll canceled because value was updated
```

kpine, 2021-11-29:
Yeah, it needs to be a 4-byte value, so that won't work.

JvdBosch, 2021-11-29:
OK, HA also does not see the change:
[image]

So we need to change the value size? Can I do this to test?

kpine, 2021-11-29:
No, that's not the solution. The values are [already correctly set to 4-bytes](https://github.com/zwave-js/node-zwave-js/blob/24688ec65af88eda79cfc2dd363f2613c98a1e83/packages/config/config/devices/0x0086/templates/aeotec_template.json#L1861-L1877). Let's wait to hear from Al about the Unsigned vs Signed integeter issue.

At least one solution is for HA to move from its custom implementation of partials and use the setBulk API provided by the driver. That is not as trivial as changing a config file.

JvdBosch, 2021-11-29:
OK, let me know if I can do anything. I'm handy with Python and willing to help. 
Thanks for your support!

AlCalzone, 2021-11-29:
The error message

> is not valid for configuration parameters

only appears once in my codebase and that is when the value size and value format aren't known and fall back to the default settings which are incompatible with the unsigned values that should be used. 

Let me check at which point it goes wrong.

kpine, 2021-11-29:
@JvdBosch Have you tried a re-interview? In the past these values were defined as 3-byte values. Maybe you have old entries in the cache? https://github.com/zwave-js/node-zwave-js/commit/b56f069b323fe11420578ed75c3c3491abd2982d#diff-941e38fb7ba4469f3746d1d92ce0d314b5af33427632a4800ded6c7ba44de522

AlCalzone, 2021-11-29:
Ok the problem is that our config file only defines partial parameters (param 83+bitmasks), but the `setValue` call is done for a non-partial (83) that does not exist in the config files - and by extension for a non-existant value ID, which should not happen. This causes the lookup for the param definition to fail and use default values which aren't compatible with a 4-byte unsigned value.

Since fixing this will require an update anyways, I think the correct way forward would be to change the HA service call to use the `setBulk` method instead, followed by a `getBulk` to validate the changes.

kpine, 2021-11-29:
What about the failure with the single partial parameter and setValue? https://github.com/zwave-js/node-zwave-js/issues/3800#issuecomment-981446979

AlCalzone, 2021-11-29:
I can't reproduce that in my test scenario, maybe a re-interview is required here

kpine, 2021-11-29:
If you create the fake parameter w/o the property key using this bulk service, will that affect setting the actual partial values?

AlCalzone, 2021-11-29:
The `setBulk` method bypasses the setValue logic entirely. When you refresh the settings afterwards with `getBulk`, the partials will be updated.

kpine, 2021-11-29:
Sorry, I meant the HA bulk service call. When you call the HA service it generates that value ID that never existed, w/o any type information (I'm calling this the "fake" value). Does the existence of this fake value affect subsequent uses of the real partial parameters, such that zwave-js can no longer determine the value format and size? A re-interview should fix that, if so.

AlCalzone, 2021-11-29:
AFAIK it shouldn't

JvdBosch, 2021-11-30:
Re-interviewing the device did change something. I am now able to set the values in the GUI, although the light does not respond in any case I've tried, both in the GUI as in the bulk_set_partial_config_parameters. And when I set the Blue to something higher than 127 in the GUI, it fails. 255,255,127 works, 255,255,128 fails:
[image]

And in the zwave driver at 255,255,127:
```
2021-11-30T12:30:09.293Z DRIVER » [Node 011] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      32
                                  └─[ConfigurationCCSet]
                                      parameter #:      83
                                      reset to default: false
                                      value size:       4
                                      value format:     UnsignedInteger
                                      value:            2147483392
```

There are some more issues. See below. It appears the GUI sends different values to the driver than bulk_set_partial_config_parameters.

Setting to 255,255,100 in GUI works althought the light does not change:
```
2021-11-30T12:23:07.286Z DRIVER » [Node 011] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      22
                                  └─[ConfigurationCCSet]
                                      parameter #:      83
                                      reset to default: false
                                      value size:       4
                                      value format:     UnsignedInteger
                                      value:            1694498560
```

Setting to 255,255,100 in bulk_set_partial_config_parameters fails with:
```
Error in driver ZWaveError: The value 4294927360 is not valid for configuration parameters! (ZW0322)
...
}
```

Setting to 255,255,255 in GUI:
```
Error in driver ZWaveError: The value -256 is invalid for configuration parameter 83 (size = 4, format = UnsignedInteger)! (ZW0322)
...
}
```

Setting to 255,255,255 in bulk_set_partial_config_parameters :
```
Error in driver ZWaveError: The value 4294967040 is not valid for configuration parameters! (ZW0322)
...
}
```

kpine, 2021-11-30:
> There are some more issues. See below. It appears the GUI sends different values to the driver than bulk_set_partial_config_parameters.

That's expected. They have entirely different functionality.

> And when I set the Blue to something higher than 127 in the GUI, it fails. 255,255,127 works, 255,255,128 fails

Would need to see driver logs for that.

> Setting to 255,255,100 in GUI works althought the light does not change:

Sounds like either a problem with the device, or how the config param is being set. Have you already set param 81 to "Night light mode"? That is required to use the RGB settings.

> Setting to 255,255,255 in GUI:

I suspect once you're used the bulk parameter service call, it messes up the internal state of the driver. I think you would need to re-interview and then not use that service again.

AlCalzone, 2021-11-30:
> 255,255,127 works, 255,255,128 fails

These parameters go into the bitmask in the opposite order. 255,255,127 is `0x7fffff00` (2147483392); 255,255,128 is `0x80ffff00` (-2130706688 when interpreted as signed).

I still don't think you'll get anywhere until HA uses the `setBulk` method internally.

kpine, 2021-11-30:
The GUI sets individual partial values. It doesn't combine them.

AlCalzone, 2021-11-30:
Hmm... shit. Maybe I need to use an existing parameter where values actually get merged.

AlCalzone, 2021-11-30:
Ok I can reproduce it by doctoring the config file for my ZW100.

PR #3812 (2021-11-30): fix: make sure encodePartial returns an unsigned int
JS is weird sometimes...

fixes: #3800
closed 2021-11-30 by AlCalzone