zwave-js/zwave-js #3802: Missing device configuration: Aeotec MotionSensor 7 ZWA024-C

mrm47s, 2021-11-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Aeotec ZWA024-C

### Manufacturer ID

0x0371

### Product Type

0x0002

### Product ID

0x0018

### Firmware Version

1.1

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4061

### Do you have a manual?

[ES - MultiSensor 7(1).pdf](https://github.com/zwave-js/node-zwave-js/files/7617749/ES.-.MultiSensor.7.1.pdf)

AlCalzone, 2021-11-29:
@zwave-js-bot import config 4061 from zwa

henrikrox, 2021-12-21:
@AlCalzone hello my friend, do you have everything you need regarding ZWA024/multisensor 7? I bought 3 today and using them as powered, they can be added to zwavejs, but it has a general nodexx status. And i cant edit the configuration. Thank you, and have a nice christmas :)

AlCalzone, 2021-12-21:
Its being worked on: https://github.com/zwave-js/node-zwave-js/pull/3855

In the meantime you should be able to change the config parameters in zwavejs2mqtt by entering the param no., value size and value in the respective field.

henrikrox, 2021-12-21:
> Its being worked on: #3855
> 
> In the meantime you should be able to change the config parameters in zwavejs2mqtt by entering the param no., value size and value in the respective field.

nice, i tried using zwavejs2mqtt once along side zwavejs but i was confused on how they worked together, guess i have to try again.  thank you for your fast repsonse.

PR #3855 (2021-12-30): feat(config): add many new devices from Z-Wave Alliance DB
Closes #3841
Closes #3834
Closes #3802
Closes #3804
Closes #3621
Closes #3609
closed 2021-12-30 by AlCalzone