zwave-js/zwave-js #3803: [feat] ZME_UZB1 to change RF region

chorus12, 2021-11-29:
I am using ZME_UZB1 stick which is capable of changing the RF region (and it works fine for close frequencies like EU and Russia). 
When calling a code to change the RF region 
`const region = await driver.controller.getRFRegion();`
get response:
`Error while calling api driverFunction: Your hardware does not support getting the RF region! (ZW0105) 
`
Please advise if that can be fixed via configuration. Thanks.

AlCalzone, 2021-11-29:
This is working as intended. The error message clearly states that changing the region is not supported.

chorus12, 2021-12-01:
Techsheet - https://z-wave.me/products/uzb/ with info on freq change, and the scirpt - https://z-wave.me/download/changezwf.sh
Does it make sence?

AlCalzone, 2021-12-01:
I see. That stick uses proprietary commands to access the functionality. We don't support these yet, but I could document how to issue them anyways, similar to controlling the LED on Aeotecs Gen5 stick.

AlCalzone, 2025-05-09:
Relevant: https://github.com/zwave-js/zwave-js/issues/1245#issuecomment-2866317786