zwave-js/zwave-js #3811: Communication with node stalls after `ZWaveError: Security CC requires a nonce to be sent`

AlCalzone, 2021-11-30:


steve28, 2021-12-01:
I am seeing this as well - started happening after I updated to 8.8.2, more specifically zwavejs2mqtt 6.0.2

AlCalzone, 2021-12-01:
Can you share one of your logs so I can see if the issue is the same?

mwav3, 2021-12-01:
I also experienced this twice.  Received this error and everything completely locked up and stopped responding in the zwave network until I restarted (running zwavejs2mqtt in docker).  My logs were set to error only at the time so my only entry is the following:

```
2021-11-29T03:01:29.799Z DRIVER   Unexpected error during SerialAPI command: ZWaveError: Security CC requires a 
                                  nonce to be sent! (ZW1400)
                                      at throwNoNonce (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/S
                                  ecurityCC.ts:603:10)
                                      at SecurityCCCommandEncapsulation.serialize (/usr/src/app/node_modules/zwa
                                  ve-js/src/lib/commandclass/SecurityCC.ts:615:23)
                                      at SendDataRequest.serialize (/usr/src/app/node_modules/zwave-js/src/lib/c
                                  ontroller/SendDataMessages.ts:107:37)
                                      at getSerialAPICommandMachineConfig (/usr/src/app/node_modules/zwave-js/sr
                                  c/lib/driver/SerialAPICommandMachine.ts:143:18)
                                      at createSerialAPICommandMachine (/usr/src/app/node_modules/zwave-js/src/l
                                  ib/driver/SerialAPICommandMachine.ts:385:3)
                                      at executeSerialAPICommand (/usr/src/app/node_modules/zwave-js/src/lib/dri
                                  ver/CommandQueueMachine.ts:276:43)
                                      at Interpreter.exec (/usr/src/app/node_modules/xstate/lib/interpreter.js:8
                                  45:61)
                                      at Interpreter.execute (/usr/src/app/node_modules/xstate/lib/interpreter.j
                                  s:204:14)
                                      at Interpreter.update (/usr/src/app/node_modules/xstate/lib/interpreter.js
                                  :230:12)
                                      at /usr/src/app/node_modules/xstate/lib/interpreter.js:110:15
```

AlCalzone, 2021-12-01:
Can you try and make a log which captures the communication before that failure (when it happens again)? I wonder if it is the same thing I've seen.

mwav3, 2021-12-01:
Sure, whatever I can do to help. Just set logs to "silly" and will post them if it locks up again.

steve28, 2021-12-01:
> Can you share one of your logs so I can see if the issue is the same?

Attached the two times it happened to me.  I appended the line number where it occurred to the filename.
[zwavejs_2021-11-29_line_1103.log](https://github.com/zwave-js/node-zwave-js/files/7635083/zwavejs_2021-11-29_line_1103.log)
[zwavejs_2021-11-30_line_561.log](https://github.com/zwave-js/node-zwave-js/files/7635084/zwavejs_2021-11-30_line_561.log)

AlCalzone, 2021-12-01:
@steve28 the loglevel must be set to debug or I won't see anything relevant.

AlCalzone, 2021-12-01:
@Nixon506E You seem to have a huuuuuge issue with routing/connectivity.

One culprit Node 77 which tries really hard to send its reports to the controller, for example look at the range `2021-11-28T21:19:18.352Z` to `2021-11-28T21:19:24.122Z`. That's just 6 seconds but like 50 messages from the node, even falling back to broadcasts as a last measure. The outgoing tommand to Node 77 which was just sent prior takes a whopping 25 seconds to reach the target:
```
2021-11-28T21:19:27.523Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            82
                                    transmit status:        OK, took 25390 ms
                                    repeater node IDs:      118, 92
                                    routing attempts:       28
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -68 dBm
                                    ACK RSSI on repeaters:  -62 dBm, -69 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
                                    route failed here:      61 -> 77
```

Meanwhile Node 169 is waiting for a response to a supervision request, but Node 77 keeps spamming so long that other nodes start requesting responses and get impatient, resending their commands:
```
2021-11-28T21:19:38.890Z DRIVER « [Node 169] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -83 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command
...

2021-11-28T21:19:59.260Z DRIVER « [Node 172] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -69 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 72
                                    └─[SupervisionCCGet]
                                      │ session id:      33
                                      │ request updates: false
                                      └─[MultilevelSensorCCReport]
                                          type:  Air temperature
                                          scale: Fahrenheit
                                          value: 64.2
2021-11-28T21:19:59.608Z DRIVER « [Node 172] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -69 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command
2021-11-28T21:19:59.608Z SERIAL » [ACK]                                                                   (0x06)
2021-11-28T21:19:59.966Z SERIAL « 0x012100a80001ac189f034800ac2302d112308e017b0dd1540812b79f1fe019910 (35 bytes)
                                  0bbeb
2021-11-28T21:19:59.967Z DRIVER   Dropping message with invalid payload
2021-11-28T21:19:59.967Z DRIVER « [Node 172] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -69 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command
```

etc. This makes your network so busy that the outgoing request from earlier (which needs to be completed before the driver can even start responding to the others) fails after almost **52 seconds and 23 attempts**:
```
2021-11-28T21:20:20.198Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            83
                                    transmit status:        NoAck, took 51790 ms
                                    routing attempts:       23
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    TX channel no.:         1
                                    route failed here:      120 -> 118
                                    TX power:               0 dBm
                                    measured noise floor:   0 dBm
```
after which the next command to node 173 in direct range is done within 10 milliseconds. This amount of routing is excessive, but I'm not sure where you can reduce it. zwave-js doesn't have the command to change it implemented yet.

---

Note to self - the issue is somewhere in this section where the SPAN state gets reset while the message is in flight, gets retried and the retry fails to serialize:
```
2021-11-28T21:20:20.276Z DRIVER » [Node 172] [REQ] [SendDataBridge]
                                  │ source node id:   1
[... 160 lines cut ...]
                                    route failed here:      176 -> 144
                                    TX power:               0 dBm
                                    measured noise floor:   0 dBm
2021-11-28T21:20:49.739Z DRIVER   Unexpected error during SerialAPI command: ZWaveError: Security S2 CC requires
                                   the receiver's nonce to be sent! (ZW1401)
                                      at Security2CCMessageEncapsulation.serialize (/usr/src/app/node_modules/zw
                                  ave-js/src/lib/commandclass/Security2CC.ts:913:10)
                                      at SendDataBridgeRequest.serialize (/usr/src/app/node_modules/zwave-js/src
                                  /lib/controller/SendDataBridgeMessages.ts:125:37)
[... cut ...]
                                      at Interpreter.execute (/usr/src/app/node_modules/xstate/lib/interpreter.j
                                  s:204:14)
                                      at Interpreter.update (/usr/src/app/node_modules/xstate/lib/interpreter.js
                                  :230:12)
                                      at /usr/src/app/node_modules/xstate/lib/interpreter.js:110:15
                                      at Scheduler.process (/usr/src/app/node_modules/xstate/lib/scheduler.js:69
                                  :7)
```

AlCalzone, 2021-12-01:
@Nixon506E You should your network. Not all at once, that is probably going to end in much larger chaos, but individual nodes.

I'd start by searching for `route failed here` and heal the nodes that are mentioned there (on both ends of the broken link). The next step would be `transmit status:        NoAck`, but this is a bit more tricky.
From these entries you can search upwards for `callback id:` and check if the number behind that is the same as the NoAck message. This should give you some more nodes.

I'm going to add some more tools for this insight in the future so you don't need logs, but they aren't there yet.

Nixon506E, 2021-12-01:
I knew that node was potentially on the outer limit of the network and it is housed in an old construction wall that is limiting signal to it.

I will try to heal the problem nodes as you mentioned and check for successful transmissions.

Would it be beneficial to add more repeater nodes around this one? Any other reasons this node might be causing problems like this? Should it get replaced?

AlCalzone, 2021-12-01:
> Would it be beneficial to add more repeater nodes around this one? Any other reasons this node might be causing problems like this? Should it get replaced?

It might help to add 1-2 repeaters between the node and others. From the log it seems like the normal messages reach the controller just fine (although they might be delayed too) but the acknowledgements (which are very short and easily get scrambled) don't reach node 77 reliably and the existing routes aren't consistent.

AlCalzone, 2021-12-01:
@Nixon506E https://github.com/zwave-js/node-zwave-js/issues/3810 might be related. Your network seems pretty large too, and I guess you're using a 700-series stick?

Nixon506E, 2021-12-01:
@AlCalzone I am, the zooz model. I did see that issue, definitely could be part of it, any idea when the rolled back version might come out?

mwav3, 2021-12-01:
Got the issue to happen again.  I ran the "goodnight" routine that shuts off multiple devices and locks 3 doors.  I have about 22 devices.  I get the error and then everything in the network jams up and becomes unresponsive, and restarting zwavejs2mqtt is the only way to recover it.  Here's the logs, let me know if you need anything else.

[zwavejs_2021-12-01.log](https://github.com/zwave-js/node-zwave-js/files/7637379/zwavejs_2021-12-01.log)

Nixon506E, 2021-12-02:
Issues on my side still. I put a few smart plugs around the problem node and healed nodes individually until they were all consistently returning successful heal responses. It seems to have cleared up the route failed messages for the problem node but a different node that is now the farthest from other nodes is the one showing the most failed routes, and usually going to it.

Not sure if the new node is the issue or if it is still something else but I again had the controller not able to transmit during the day today and had to reboot. I will try to get some more smart plugs for around the new problem node and see if I can at least stabilize the network. Feels like if the issue is with having a large network then I may be going in the wrong direction...

AlCalzone, 2021-12-03:
@Nixon506E if you can make another log showing this behavior, I can take a look if your issue is still network related.

Nixon506E, 2021-12-03:
@AlCalzone I added another node near the new trouble node just now and upgraded to 8.8.3. I'll post new logs if I am still seeing the issue after all of that. Thanks!

AlCalzone, 2021-12-03:
Note to self: The cause of failure seems to be slightly different for S0 encapsulation. Here, the issue is that a retried security-encapsulated message gets serialized multiple times, but after the first time, the nonce has been consumed.

AlCalzone, 2021-12-06:
Should be fixed on master

closed 2021-12-06 by AlCalzone

Colorado4Wheeler, 2021-12-22:
I can confirm that this exact issue is still occurring.  I've been debugging this all week trying to determine if it's software that I needed to open a ticket on or not.  My issue is exactly the same as @mwav3 where I "goodnight" my house and part of that script is to lock the doors and one of the doors throws this error and locks up the entire system.  Z-Wave JS still "works" because I can see motion sensors triggering in HA and in the logs, but no wired device will respond.  The fix is to restart the Z-Wave JS to MQTT add-on and everything returns to normal again.

```
zwavejs2mqtt: 6.1.1
zwave-js: 8.9.0-beta.1
home id: 3823806645
home hex: 0xe3eaa8b5
```

```
2021-12-22T07:00:31.091Z DRIVER   Unexpected error during SerialAPI command: ZWaveError: Security CC requires a 
                                  nonce to be sent! Reason: nonce 0xd8 not found (ZW1400)
                                      at throwNoNonce (/opt/node_modules/zwave-js/src/lib/commandclass/SecurityC
                                  C.ts:83:8)
                                      at SecurityCCCommandEncapsulation.get receiverNonce [as receiverNonce] (/o
                                  pt/node_modules/zwave-js/src/lib/commandclass/SecurityCC.ts:596:4)
                                      at SecurityCCCommandEncapsulation.serialize (/opt/node_modules/zwave-js/sr
                                  c/lib/commandclass/SecurityCC.ts:644:30)
                                      at SendDataRequest.serialize (/opt/node_modules/zwave-js/src/lib/controlle
                                  r/SendDataMessages.ts:107:37)
                                      at getSerialAPICommandMachineConfig (/opt/node_modules/zwave-js/src/lib/dr
                                  iver/SerialAPICommandMachine.ts:143:18)
                                      at createSerialAPICommandMachine (/opt/node_modules/zwave-js/src/lib/drive
                                  r/SerialAPICommandMachine.ts:385:3)
                                      at executeSerialAPICommand (/opt/node_modules/zwave-js/src/lib/driver/Comm
                                  andQueueMachine.ts:276:43)
                                      at Interpreter.exec (/opt/node_modules/xstate/lib/interpreter.js:845:61)
                                      at Interpreter.execute (/opt/node_modules/xstate/lib/interpreter.js:204:14
                                  )
                                      at Interpreter.update (/opt/node_modules/xstate/lib/interpreter.js:230:12)
```

AlCalzone, 2021-12-22:
> I can confirm that this exact issue is still occurring

Yeah because it is only fixed in beta.3

Colorado4Wheeler, 2021-12-22:
Got it, thank you!  I was trying to pinpoint when the fix released and couldn't, I appreciate the response!