zwave-js/zwave-js #3828: All nodes are loosing connection almost immediately and reporting disconnected, but DEBUG shows communication somehow works

AlfaBravoX, 2021-12-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? Just pair nodes and let them be for a while (30 mins)

What do you observe? E5 error at node, Error in comunication - TRV is out of range from Smart home HUB. Please more HUB closer or add repeater into a system.

What did you expect to happen? TRV should not show E5 code and remain connected

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Danfoss
Model name: Living Connect Z Thermostat
Node ID in your network: 5,7,9

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 6.0.2
`zwavejs2mqtt` branch: 8.8.2

Driver Version: 8.8.2
Server Version: 1.12.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes deprecated HA z-wave integration worked well

### Attach Driver Logfile

Subscribed to Z-Wave JS Log Messages…
[... 934 lines cut ...]
2021-12-01T08:19:55.311Z SERIAL » [ACK]                                                                   (0x06)
2021-12-01T08:19:55.313Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 1
2021-12-01T08:19:55.329Z CNTRLR « [Node 007] node neighbors received: 1
2021-12-01T08:19:55.330Z CNTRLR » [Node 009] requesting node neighbors...
2021-12-01T08:19:55.346Z SERIAL » 0x010700800900000071                                                 (9 bytes)
2021-12-01T08:19:55.348Z DRIVER » [REQ] [GetRoutingInfo]
                                    remove non-repeaters: false
                                    remove bad links:     false
2021-12-01T08:19:55.359Z SERIAL « [ACK]                                                                   (0x06)
2021-12-01T08:19:55.368Z SERIAL « 0x01200180010000000000000000000000000000000000000000000000000000000 (34 bytes)
                                  05f
2021-12-01T08:19:55.370Z SERIAL » [ACK]                                                                   (0x06)
2021-12-01T08:19:55.373Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 1
2021-12-01T08:19:55.385Z CNTRLR « [Node 009] node neighbors received: 1
2021-12-01T08:19:55.386Z CNTRLR   Turning RF on...
2021-12-01T08:19:55.410Z SERIAL » 0x0104001001ea                                                       (6 bytes)
2021-12-01T08:19:55.412Z DRIVER » [REQ] [SetRFReceiveMode]
                                    enabled: true
2021-12-01T08:19:55.420Z SERIAL « [ACK]                                                                   (0x06)
2021-12-01T08:19:55.424Z SERIAL « 0x0104011001eb                                                       (6 bytes)
2021-12-01T08:19:55.426Z SERIAL » [ACK]                                                                   (0x06)
2021-12-01T08:19:55.429Z DRIVER « [RES] [SetRFReceiveMode]
                                    success: true

AlfaBravoX, 2021-12-01:
[image]

AlfaBravoX, 2021-12-01:
Also Z-wave+ is reported as NO, however the controller is PLUS capable

AlCalzone, 2021-12-01:
I'll look at the logs later, thanks!

> however the controller is PLUS capable

That has nothing to do with the devices. AFAIK the LC-13 are not Z-Wave+ certified. We're not showing most of that information for the controller because it cannot be queried.

AlfaBravoX, 2021-12-01:
Adding more logs.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/7632902/zwavejs2mqtt-store.zip)
..

AlfaBravoX, 2021-12-01:
> I'll look at the logs later, thanks!
> 
> > however the controller is PLUS capable
> 
> That has nothing to do with the devices. AFAIK the LC-13 are not Z-Wave+ certified. We're not showing most of that information for the controller because it cannot be queried.

Thanks for clarification. And additional logs attached. I did set factory reset to all devices and paired them again, but problem is the same. After succesfull pairing they show E5 after while again and wake period remains on 1800

jordanrkirby, 2021-12-01:
I am seeing the same problem with my Devolo TRVs (which I think are Danfoss TRVs with different firmware).
These have worked on ZWaveJS for several months up until early this week maybe around 28th/29th Nov.

AlfaBravoX, 2021-12-01:
> I am seeing the same problem with my Devolo TRVs (which I think are Danfoss TRVs with different firmware). These have worked on ZWaveJS for several months up until early this week maybe around 28th/29th Nov.

Yes, I did migration a week about and it was working few days, so something had to happen with the code on 28th/29th Nov.

AlCalzone, 2021-12-01:
I'm suspecting https://github.com/zwave-js/node-zwave-js/pull/3691 has an unintended interaction. I'm going to revert it later for the 8.x branch.

However I'd like to understand what exactly breaks the communication so I can fix it properly.
@jordanrkirby if you could also contribute some driver logs of the thermostats waking up, it might help.

jordanrkirby, 2021-12-01:
Of course. Debug logs attached.
TRVs are device IDs 7, 8, 10, 11, 12, 15, 16, 17, 18

Thanks

[zwavejs_2021-12-01.log](https://github.com/zwave-js/node-zwave-js/files/7634301/zwavejs_2021-12-01.log)

Minims, 2021-12-01:
Hello, 

i have quite the same issue with TRV Danfoss Living Connect (Popp & Devolo brand) displaying E5 error a few minutes after zwave communication. 

i have reset my controller without success. Same issue appears.

I use zwavejs2mqtt as Hassio add-on. 
I have roll back from v0.31.0 to v0.29.0 and everything is working great again.

No more E5 error and I can control my TRV now.

AlCalzone, 2021-12-01:
@jordanrkirby The logs look okay - there are two instances where the queries couldn't be sent, but for the remaining wakeups, the TRVs get queried as expected.

I guess I need a log which captures the moment the error appears. @Minims maybe?

AlfaBravoX, 2021-12-01:
I reverted even to v0.30.0 and it started to work again. no E5 anymore

AlCalzone, 2021-12-01:
Any chance you could provide a log which captures the moment the error appears?

AlfaBravoX, 2021-12-01:
> Any chance you could provide a log which captures the moment the error appears?

i attached 2 logs. the one with zip format, should have errors captured

AlCalzone, 2021-12-01:
Oh right 👍🏻

AlCalzone, 2021-12-01:
@AlfaBravoX can you make another log where it is working for comparison? It looks like the thermostats are not actively sent back to sleep, which might trigger the error.

Minims, 2021-12-02:
Trying to get logs this morning. 
I have restart v0.31.0 and on first zwave connection with a device, it is stuck in `awake` instead of falling back to Asleep.

So I think that in few minutes my TRV will display E5 error. Can it be bad close connection to the device ?

AlfaBravoX, 2021-12-02:
> Trying to get logs this morning. I have restart v0.31.0 and on first zwave connection with a device, it is stuck in `awake` instead of falling back to Asleep.
> 
> So I think that in few minutes my TRV will display E5 error. Can it be bad close connection to the device ?

Please just read this thread carefully, then you will find response to your question. Problem is not at your side.

AlCalzone, 2021-12-02:
> it is stuck in `awake` instead of falling back to Asleep.

That sounds like the "go sleep" message doesn't get sent (like I'm seeing above). The device still goes to sleep but doesn't get the expected message and starts whining about it.

closed 2021-12-06 by AlCalzone

kpine, 2021-12-12:
Was this also fixed with the reverts in 8.8.3 (for those who can't install a beta)?

AlCalzone, 2021-12-12:
It should, since the bug was caused by the send queue changes

PR #3845 (2021-12-06): fix: send nodes to sleep again after successful response
This is the first in a set of fixes for the reworked send queue. With this PR, sleeping nodes are sent back to sleep again.

fixes: #3828