zwave-js/zwave-js #3829: Network Breaks Nightly Until Zwave-JS Restart Since Friday

kars85, 2021-12-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Every evening (sunset - 4AM approx), automations fail to execute and Zwave devices become unavailable

What do you observe?
Unable to control/manage nodes.
What did you expect to happen?
Stable behavior. When restarting the container and initiating a heal, the heal rate is ~50%.

I am attaching log archives going back to Friday when the behavior began (can't remember if Silly was enabled that far back, but I turned Silly on shortly thereafter over the weekend. Todays 2021-12-01.log**.old** is before I restarted the container this morning and began troubleshooting, so it should contain information on the network "crash".

The 2021-12-01.log is a fresh log starting with a container startup, network graph view, followed by a network heal (~50% success on the heal).

I've also attached my Zooz controller's NVM backup along with my homeid.json

### Device information

All devices

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.0.2
zwave-js: 8.8.2
home id: 3745883352
home hex: 0xdf45a4d8

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Prior to Friday all automations and control worked as expected.

### Attach Driver Logfile

[zwavejs_2021-11-30.log.gz](https://github.com/zwave-js/node-zwave-js/files/7634944/zwavejs_2021-11-30.log.gz)
[zwavejs_2021-11-29.log.gz](https://github.com/zwave-js/node-zwave-js/files/7634945/zwavejs_2021-11-29.log.gz)
[zwavejs_2021-11-28.log.gz](https://github.com/zwave-js/node-zwave-js/files/7634946/zwavejs_2021-11-28.log.gz)
[zwavejs_2021-11-27.log.gz](https://github.com/zwave-js/node-zwave-js/files/7634947/zwavejs_2021-11-27.log.gz)
[zwavejs_2021-12-01.log](https://github.com/zwave-js/node-zwave-js/files/7634949/zwavejs_2021-12-01.log)
[zwavejs_2021-12-01.old.log](https://github.com/zwave-js/node-zwave-js/files/7634951/zwavejs_2021-12-01.old.log)
[df45a4d8.json.txt](https://github.com/zwave-js/node-zwave-js/files/7634953/df45a4d8.json.txt)

AlCalzone, 2021-12-01:
Are you able to test with an older version if the issue persists there (before node-zwave-js 8.8.0 and before zwavejs2mqtt 6.0.0)?

kars85, 2021-12-01:
Sure - what image tag would you like me to revert to and I'll exclude zwavejs2mqtt from watchtower? Do you perceive any issues with my existing setup by downgrading temporarily?

kars85, 2021-12-01:
@Nixon506E - that was the zwavejs2mqtt version I was going to drop down to as well...are you having nearly the same time frame of drop outs? Where, shortly after going through network heals, physically power cycling nodes (breakers/air gaps) things come back, only for them to go dead after a handful of hours?

@AlCalzone - It is very hard for me to isolate this down, and I apologize for the inundation of logs/files. If there is anything I can do on my side, just let me know. I will still try dropping down to 5.12.0 and letting it run for the specific period of time I've consistently seen issues.

Nixon506E, 2021-12-01:
They are not being marked as dead for me but rather they just stop responding. I do not need to reset the devices, only the driver but the rollback did not help and after a few hours it does happen every time.

kars85, 2021-12-01:
> They are not being marked as dead for me but rather they just stop responding. I do not need to reset the devices, only the driver but the rollback did not help and after a few hours it does happen every time.

Got it - just so I'm not getting lost in the translation, when you mention "reset the driver", are you restarting the zwavejs2mqtt container, or something else?

I just changed my compose to stay at 5.12.0 and am running it now with a fresh log despite your issues persisting, and will know by the end of the night if issues are able to reproduced. I have lighting and a space heater automation that make it super easy to know if my network when kaboom.

Nixon506E, 2021-12-01:
Yes, restarting zwavejs

kars85, 2021-12-01:
@Eriner - my response to your issue indicating my stick was showing neighbors (yours wasn't, as I understood?) was shortly after doing myriad of things to get random nodes reconnected like you. Not all of my nodes were responsive, but most. I guess I can't be 100% certain my stick shows neighbors when the network goes haywire, but another data point to check if downgrading to 5.12.0 is an issue.

I can say that I have had weeks worth of success at a time with the Zooz stick, so hopefully all is not lost. To be continued...

wigster, 2021-12-02:
I have noticed strange behaviour in the last couple of days also and am beginning to think this is related.

In particular, my Eurotronic spirit TRVs have stopped responding to temperature increases that I send to them in the morning. The display changes from the default 22 to e.g. the new value 28, but the radiator does not come on and touching the manual controls moves the temp around the default 22 to 22.5 or 21.5. From the point of the view of the zwavejs2mqtt interface everything looks fine.

I haven't really been able to troubleshoot more than this. 

```
zwavejs2mqtt: 6.0.2
zwave-js: 8.8.2
zwavejs2mqtt plugin 0.30
HA 2021.11.5
```

AlCalzone, 2021-12-02:
@wigster I need [driver logs](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) please, loglevel `debug` or `silly` and attach it here as a file.
Make sure to do something that lets me see the problem in the log.

kars85, 2021-12-02:
Downgrading to zwavejs2mqtt 5.12.0 has resolved my issues over the past 36 hours. I would be more than happy to be the guinea pig/constant.

AlCalzone, 2021-12-03:
Having logs of the failing behavior would be immensely helpful. 🙏🏻

mostlychris, 2021-12-03:
I'll just throw my name in the hat that I am also experiencing this same issue. I can get messages from zwave devices, but I can't send anything. Restarting the zwavejs2mqtt addon solves the issue. I'm running a Aeon Labs Z‐Stick Gen5 USB Controller | ZW090. Would be happy to supply logs if someone tells me where to find them. I'm running HA supervised on an Odroid N2+ (HA Blue).

jwestyp, 2021-12-03:
I have looked at logs myself and haven't seen anything that stands out showing the error. I have been rebooting every evening after my nightly automations don't happen, but I am still able to see state changes.

AlCalzone, 2021-12-03:
Guys please... No stories, post logs 🙏

kars85, 2021-12-04:
I am running 8.8.3 since last night, and since then, my network has slowly died to the point of being non-functional. Resetting logging to all nodes indicates that the driver has crashed and is a reboot loop. (Not attaching because this is all the logs capture before looping.

I will restart the driver and capture logging for the rest of the day and re-upload later.

`2021-12-04T14:23:34.364Z DRIVER   version 8.8.3
2021-12-04T14:23:34.364Z DRIVER
2021-12-04T14:23:34.364Z DRIVER   starting driver...
2021-12-04T14:23:34.365Z DRIVER   opening serial port /dev/zwave
2021-12-04T14:23:34.366Z DRIVER   serial port opened
2021-12-04T14:23:34.366Z SERIAL » [NAK]                                                                   (0x15)
2021-12-04T14:23:34.367Z DRIVER   loading configuration...
2021-12-04T14:23:34.368Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2021-12-04T14:23:34.368Z CONFIG   version 8.8.3
2021-12-04T14:23:34.571Z CONFIG   Priority device configuration directory /usr/src/app/store/config not found
2021-12-04T14:23:34.580Z DRIVER   beginning interview...
2021-12-04T14:23:34.580Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2021-12-04T14:23:34.580Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2021-12-04T14:23:34.580Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2021-12-04T14:23:34.580Z CNTRLR   querying controller IDs...
2021-12-04T14:23:34.581Z SERIAL » 0x01030020dc                                                         (5 bytes)
2021-12-04T14:23:34.581Z DRIVER » [REQ] [GetControllerId]
2021-12-04T14:23:35.583Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2021-12-04T14:23:35.684Z SERIAL » 0x01030020dc                                                         (5 bytes)
2021-12-04T14:23:35.684Z DRIVER » [REQ] [GetControllerId]
2021-12-04T14:23:36.687Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2021-12-04T14:23:37.790Z SERIAL » 0x01030020dc                                                         (5 bytes)
2021-12-04T14:23:37.790Z DRIVER » [REQ] [GetControllerId]
2021-12-04T14:23:38.794Z DRIVER   Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK
                                  from the controller (ZW0200)
                                      at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/D
                                  river.ts:3261:23)
                                      at ZWaveController.identify (/usr/src/app/node_modules/zwave-js/src/lib/co
                                  ntroller/Controller.ts:668:33)
                                      at Driver.initializeControllerAndNodes (/usr/src/app/node_modules/zwave-js
                                  /src/lib/driver/Driver.ts:938:26)
                                      at Immediate.<anonymous> (/usr/src/app/node_modules/zwave-js/src/lib/drive
                                  r/Driver.ts:818:16)
2021-12-04T14:23:38.795Z DRIVER   destroying driver instance...
`

AlCalzone, 2021-12-04:
> Timeout while waiting for an ACK from the controller

The stick is stuck and doesn't respond.

kars85, 2021-12-04:
> > Timeout while waiting for an ACK from the controller
> 
> The stick is stuck and doesn't respond.

Yep - even rebooting my docker host wasn't sufficient (you'd think that would re-initialize the USB controller/Zooz stick. Had to physically pull the stick and reinsert after the reboot. My Zooz stick came right back up.

Silly logging has commenced and will post later tonight if it still is of help here. 

Although I'm wondering, are we spinning our wheels until we get a firmware update for the 700 series controller? I have no problem waiting for the NVM migration utility and going back to my Aeotec Z-Stick Gen 5 temporarily.

AlCalzone, 2021-12-04:
> are we spinning our wheels until we get a firmware update for the 700 series controller

For some of the issues that have been popping up recently: yes!

For others that were related to the send queue changes, the logs were helpful, but I'm not expecting any new information from the 8.8.3 ones.

AlCalzone, 2021-12-06:
The driver related issues have been fixed on master. We'll track the firmware issue in https://github.com/zwave-js/node-zwave-js/issues/3810

closed 2021-12-06 by AlCalzone