zwave-js/zwave-js #3834: Missing device configuration: <enter manufacturer and device name>

Mystichrome, 2021-12-02:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Ultraloq U-BOLT-PRO-ZWAVE 

### Manufacturer ID

0x0452

### Product Type

0x0004

### Product ID

0x0001

### Firmware Version

1.01 

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4115

### Do you have a manual?

[U-Bolt Pro Z-Wave QuickGuide V1.4.pdf](https://github.com/zwave-js/node-zwave-js/files/7644141/U-Bolt.Pro.Z-Wave.QuickGuide.V1.4.pdf)

AlCalzone, 2021-12-02:
@zwave-js-bot import config 4115 from zwa

AlCalzone, 2021-12-02:
The bot is going to create a starter file but it will need to be cleaned up according to our style guide. If you're willing to help, create a PR onto the bot's branch.

PR #3855 (2021-12-30): feat(config): add many new devices from Z-Wave Alliance DB
Closes #3841
Closes #3834
Closes #3802
Closes #3804
Closes #3621
Closes #3609
closed 2021-12-30 by AlCalzone