zwave-js/zwave-js #3835: Incorrect error message: The controller does not support the 0 feature (ZW0207)

AlCalzone, 2021-12-02:


large, 2021-12-13:
Got the same error on my Z-stick Gen 5 on a home assistant installation:

```
2021-12-13 14:46:50.623 INFO ZWAVE: Calling api provisionSmartStartNode with args: [
'900126133131464156002711829469742162953459018522105800100205404608022000411000030051501024',
[length]: 1]
2021-12-13 14:46:50.651 INFO ZWAVE: The controller does not support the 0 feature (ZW0207) provisionSmartStartNode undefined
```

Product code: ZW090, fw 1.1

AlCalzone, 2021-12-13:
The error message should be 
> The controller does not support the **SmartStart** feature

You'll need to do a firmware upgrade:
https://help.aeotec.com/support/solutions/articles/6000252294-z-stick-gen5-v1-02-firmware-update

large, 2021-12-13:
I believe the stick is Gen5 and not Gen5+ (both have same productcode on Aeotec site).
And the firmware it says it has is v1.1, I'll look into it later on.

```
{
  "id": 1,
  "name": "Z-wave stick",
  "loc": "Loftet",
  "values": [],
  "groups": [],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "hexId": "0x0086-0x0001-0x005a",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0086:0x0001:0x005a:1.1",
  "manufacturerId": 134,
  "productId": 90,
  "productType": 1,
  "deviceConfig": {
    "filename": "/data/db/devices/0x0086/zw090.json",
    "isEmbedded": true,
    "manufacturer": "AEON Labs",
    "manufacturerId": 134,
    "label": "ZW090",
    "description": "Z‐Stick Gen5 USB Controller",
    "devices": [
      {
        "productType": 1,
        "productId": 90
      },
      {
        "productType": 257,
        "productId": 90
      },
      {
        "productType": 513,
        "productId": 90
      }
    ],
    "firmwareVersion": {
      "min": "0.0",
      "max": "255.255"
    },
    "associations": {},
    "paramInformation": {
      "_map": {}
    },
    "metadata": {
      "reset": "Use this procedure only in the event that the primary controller is missing or otherwise inoperable.\n\nPress and hold the Action Button on Z-Stick for 20 seconds and then release",
      "manual": "https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/1345/Z%20Stick%20Gen5%20manual%201.pdf"
    }
  },
  "productLabel": "ZW090",
  "productDescription": "Z‐Stick Gen5 USB Controller",
  "manufacturer": "AEON Labs",
  "firmwareVersion": "1.1",
  "protocolVersion": 3,
  "nodeType": 0,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": false,
  "security": "None",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": true,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": false,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 2,
    "generic": 2,
    "specific": 1
  },
  "deviceId": "134-90-1",
  "status": "Alive",
  "interviewStage": "Complete",
  "lastActive": 1639271841067,
  "isBatteryPowered": false,
  "statistics": {
    "commandsTX": 0,
    "commandsRX": 0,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0
  },
  "_name": "Z-wave stick (Loftet)"
}
```

AlCalzone, 2021-12-14:
Yes and the Gen5 (not +) recently got the aforementioned firmware upgrade to bring it on same SDK as the 5+.

large, 2021-12-14:
Thank you for the support @AlCalzone !
Updated the stick with the link you sent (note to others, the UBZ-drivers needs to be installed before the programming drivers: https://help.aeotec.com/support/solutions/articles/6000034892-z-stick-gen5-drivers-inf-files )

Tested the "Smart Start" and it worked right away 👍 

Result is fw after upgrade v1.2:
```
{
  "id": 1,
  "name": "Z-wave stick",
  "loc": "Loftet",
  "values": [],
  "groups": [],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "hexId": "0x0086-0x0001-0x005a",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0086:0x0001:0x005a:1.2",
  "manufacturerId": 134,
  "productId": 90,
  "productType": 1,
  "deviceConfig": {
    "filename": "/data/db/devices/0x0086/zw090.json",
    "isEmbedded": true,
    "manufacturer": "AEON Labs",
    "manufacturerId": 134,
    "label": "ZW090",
    "description": "Z‐Stick Gen5 USB Controller",
    "devices": [
      {
        "productType": 1,
        "productId": 90
      },
      {
        "productType": 257,
        "productId": 90
      },
      {
        "productType": 513,
        "productId": 90
      }
    ],
    "firmwareVersion": {
      "min": "0.0",
      "max": "255.255"
    },
    "associations": {},
    "paramInformation": {
      "_map": {}
    },
    "metadata": {
      "reset": "Use this procedure only in the event that the primary controller is missing or otherwise inoperable.\n\nPress and hold the Action Button on Z-Stick for 20 seconds and then release",
      "manual": "https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/1345/Z%20Stick%20Gen5%20manual%201.pdf"
    }
  },
  "productLabel": "ZW090",
  "productDescription": "Z‐Stick Gen5 USB Controller",
  "manufacturer": "AEON Labs",
  "firmwareVersion": "1.2",
  "protocolVersion": 3,
  "nodeType": 0,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": false,
  "security": "None",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": true,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": false,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 2,
    "generic": 2,
    "specific": 1
  },
  "deviceId": "134-90-1",
  "status": "Alive",
  "interviewStage": "Complete",
  "lastActive": 1639475316501,
  "isBatteryPowered": false,
  "statistics": {
    "messagesTX": 11,
    "messagesRX": 76,
    "messagesDroppedRX": 0,
    "NAK": 0,
    "CAN": 2,
    "timeoutACK": 0,
    "timeoutResponse": 0,
    "timeoutCallback": 0,
    "messagesDroppedTX": 0
  },
  "_name": "Z-wave stick (Loftet)"
}
```

PR #3907 (2021-12-15): fix: print enum member name in assertFeature error message
fixes: #3835
closed 2021-12-15 by AlCalzone