zwave-js/zwave-js #3836: Zooz Zen76 configuration not correct

themackworth, 2021-12-02:
I haven't checked any of the other zen switches that were modified in this commit https://github.com/zwave-js/node-zwave-js/pull/3291, but the zen76 settings are not correct.

https://www.support.getzooz.com/kb/article/549-zen76-s2-on-off-switch-700-advanced-settings/

https://devices.zwave-js.io/?jumpTo=0x027a:0x7000:0xa006:0.0

AlCalzone, 2021-12-03:
Can you specify what you mean with "not correct"? We don't necessarily use the exact labels and descriptions manufacturers use.
/cc @blhoward2

themackworth, 2021-12-03:
@AlCalzone sorry for not being thorough in my report.

For example, parameter 1 should allow the user to control how they need to use the paddle to control the light.

```On/Off Paddle Orientation
Parameter 1: Choose if you want the upper paddle to turn the light on or turn the light off when tapped.
Values: 0 – Upper paddle turns the light on, lower paddle turns it off (default); 1 – Upper paddle turns the light off, lower paddle turns it on; 2 – Any paddle turns light on/off.
Size: 1 byte dec
```
But in that PR it was changed to:
[image]

I also don't see a parameter 7 in their documentation but its listed in this config.

blhoward2, 2021-12-05:
What is shown for parameter 1 is what you quoted. We don't stick with the manufacturer's wording to keep things consistent. 0 is normal. 1 is inverted. 2 is momentary mode, in which any paddle acts like a button. I agree that "momentary mode" may be confusing there. 

Parameter 7 was added in February, at the time it was confirmed with Zooz support.

themackworth, 2021-12-06:
@blhoward2 This is what I see in the UI though:

[image]

How does one select momentary mode (which is poorly named).

blhoward2, 2021-12-06:
That's not a bug in our files, that's a bug at HA. They're setting up a binary switch from parsing things. Try reinterviewing the device. If that doesn't work you'll have to create an issue on the HA project.

themackworth, 2021-12-06:
@blhoward2 Re-interviewing didn't change anything.  I noticed that another zen76 switch I added before the change still has the old working configuration.  I re-interviewed that one and it broke also.  I don't see any structural differences in the json between the old working one and new master template based on.  Which HA project do you think is appropriate to file this is issue in?

blhoward2, 2021-12-06:
All HA issues get opened on the main project.

kpine, 2021-12-06:
There is a bug in the HA frontend, and the ZEN76 device config file was changed in the last two months. The combination of the two is what is causing the problem. You can submit an issue to Core and the devs will triage it as appropriate. https://github.com/home-assistant/core/issues/new/choose

closed 2021-12-10 by zwave-js-assistant[bot]