zwave-js/zwave-js #3840: [bug] Qubino ZMNHJD1 flush pilot with ZMNHEA1 Sensor - no automatic refresh

heaven988, 2021-12-03:
### Version

**Checklist:**

- [x] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 6.0.2
Z-Wave JS version: 8.8.2

### Describe the bug
Hello !

I had several ZMNHJD1 for controlling flush pilot. In order to detect the sensor (ZMNHEA1), i start two exclusion of the device and then i add the device with the sensor into zwavejs2mqtt. I doesn't have problem with integration of the sensor.
Everything is detect, including the "air temperature sensor" wich has a correct value ([5-49-0-Air temperature] Air temperature)
But this value doesn't change unless i clic the "Refresh button" on zwavejs2mqtt interface. The variation of air temperature was over 5°C during my test.

### To Reproduce

Steps to reproduce the behavior:

1. Include ZMNHJD1 with ZMNHEA1
2. Wait the value "49" to refresh

### Expected behavior

I read that on Jeedomus, you have a parameter ton choose the time of refresh of this module. Maybe we just miss something on the integration ?

### Additional context

I also have a Qubino ZMNHND (flush 1D Relay) with the same sensor (ZMNHEA1) and on the same device / integration. On this module, the Air Temperature refresh correctly (parameter 49 as well, and every 0.2°C of variation, that's my setting "[x-112-0-120] Digital temperature sensor reporting").

Let me know if i can help.

Have a nice day

AlCalzone, 2021-12-03:
@heaven988 Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly`. 
Re-interview the device and when done, attach the log here as a file.

heaven988, 2021-12-08:
Hello @robertsLando @AlCalzone,
In groups association, i have this : 
'2021-12-08 09:16:28.248 INFO ZWAVE: Calling api getAssociations with args: [ 31, [length]: 1 ]
2021-12-08 09:16:28.282 INFO ZWAVE: Success zwave api call getAssociations [
  { endpoint: 0, groupId: 1, nodeId: 1, targetEndpoint: undefined },
  [length]: 1
]'
I'm not familiar with this part. I will read about lifeline association.

Here is the log of a reinterview :

`2021-12-08 09:10:16.648 INFO ZWAVE: Calling api refreshInfo with args: [ 31, [length]: 1 ]
2021-12-08 09:10:16.761 INFO ZWAVE: Node 31: interview started
2021-12-08 09:10:16.805 INFO ZWAVE: Success zwave api call refreshInfo undefined
2021-12-08 09:10:16.861 INFO ZWAVE: Node 31: interview stage PROTOCOLINFO completed
2021-12-08 09:10:17.000 INFO ZWAVE: Node 31: interview stage NODEINFO completed
2021-12-08 09:10:17.216 INFO ZWAVE: Node 31: value added: 114-0-manufacturerId => 345
2021-12-08 09:10:17.224 INFO ZWAVE: Node 31: value added: 114-0-productType => 4
2021-12-08 09:10:17.230 INFO ZWAVE: Node 31: value added: 114-0-productId => 81
2021-12-08 09:10:17.507 INFO ZWAVE: Node 31: value added: 134-0-libraryType => 3
2021-12-08 09:10:17.515 INFO ZWAVE: Node 31: value added: 134-0-protocolVersion => 5.3
2021-12-08 09:10:17.520 INFO ZWAVE: Node 31: value added: 134-0-firmwareVersions => 2.0,1.0
2021-12-08 09:10:17.525 INFO ZWAVE: Node 31: value added: 134-0-hardwareVersion => 1
2021-12-08 09:13:12.576 INFO ZWAVE: Node 31: value added: 38-0-targetValue => undefined
2021-12-08 09:13:12.585 INFO ZWAVE: Node 31: metadata updated: 38-0-Up
2021-12-08 09:13:12.591 INFO ZWAVE: Node 31: metadata updated: 38-0-Down
2021-12-08 09:13:12.755 INFO ZWAVE: Node 31: value added: 38-0-currentValue => 20
2021-12-08 09:13:12.954 INFO ZWAVE: Node 31: value added: 37-0-currentValue => true
2021-12-08 09:13:12.966 INFO ZWAVE: Node 31: value added: 37-0-targetValue => undefined
2021-12-08 09:13:13.099 INFO ZWAVE: Node 31: metadata updated: 49-0-Air temperature
2021-12-08 09:13:13.110 INFO ZWAVE: Node 31: value added: 49-0-Air temperature => 18.8
2021-12-08 09:13:13.231 INFO ZWAVE: Node 31: metadata updated: 112-0-1
2021-12-08 09:13:13.243 INFO ZWAVE: Node 31: metadata updated: 112-0-2
2021-12-08 09:13:13.249 INFO ZWAVE: Node 31: metadata updated: 112-0-3
2021-12-08 09:13:13.259 INFO ZWAVE: Node 31: metadata updated: 112-0-4
2021-12-08 09:13:13.265 INFO ZWAVE: Node 31: metadata updated: 112-0-5
2021-12-08 09:13:13.274 INFO ZWAVE: Node 31: metadata updated: 112-0-6
2021-12-08 09:13:13.281 INFO ZWAVE: Node 31: metadata updated: 112-0-11
2021-12-08 09:13:13.290 INFO ZWAVE: Node 31: metadata updated: 112-0-12
2021-12-08 09:13:13.297 INFO ZWAVE: Node 31: metadata updated: 112-0-13
2021-12-08 09:13:13.310 INFO ZWAVE: Node 31: metadata updated: 112-0-30
2021-12-08 09:13:13.542 INFO ZWAVE: Node 31: value added: 112-0-1 => 1
2021-12-08 09:13:13.748 INFO ZWAVE: Node 31: value added: 112-0-2 => 1
2021-12-08 09:13:13.925 INFO ZWAVE: Node 31: value added: 112-0-3 => 1
2021-12-08 09:13:14.087 INFO ZWAVE: Node 31: value added: 112-0-4 => 1
2021-12-08 09:13:14.231 INFO ZWAVE: Node 31: value added: 112-0-5 => 1
2021-12-08 09:13:14.385 INFO ZWAVE: Node 31: value added: 112-0-6 => 1
2021-12-08 09:13:14.539 INFO ZWAVE: Node 31: value added: 112-0-11 => 1
2021-12-08 09:13:14.733 INFO ZWAVE: Node 31: value added: 112-0-12 => 4
2021-12-08 09:13:14.876 INFO ZWAVE: Node 31: value added: 112-0-13 => 5
2021-12-08 09:13:15.054 INFO ZWAVE: Node 31: value added: 112-0-30 => 0
2021-12-08 09:13:15.191 INFO ZWAVE: Node 31: interview stage COMMANDCLASSES completed
2021-12-08 09:13:15.201 INFO ZWAVE: Node 31: interview stage OVERWRITECONFIG completed
2021-12-08 09:13:15.208 INFO ZWAVE: Node 31: interview stage COMPLETE completed
2021-12-08 09:13:15.263 INFO ZWAVE: Node 31: value added 31-38-0-targetValue => undefined
2021-12-08 09:13:15.268 INFO ZWAVE: Node 31: value added 31-38-0-duration => NaNs
2021-12-08 09:13:15.305 INFO ZWAVE: Node 31: value added 31-38-0-currentValue => 20
2021-12-08 09:13:15.309 INFO ZWAVE: Node 31: value added 31-38-0-Up => undefined
2021-12-08 09:13:15.313 INFO ZWAVE: Node 31: value added 31-38-0-Down => undefined
2021-12-08 09:13:15.318 INFO ZWAVE: Node 31: value added 31-134-0-libraryType => 3
2021-12-08 09:13:15.322 INFO ZWAVE: Node 31: value added 31-134-0-protocolVersion => 5.3
2021-12-08 09:13:15.326 INFO ZWAVE: Node 31: value added 31-134-0-firmwareVersions => 2.0,1.0
2021-12-08 09:13:15.332 INFO ZWAVE: Node 31: value added 31-134-0-hardwareVersion => 1
2021-12-08 09:13:15.335 INFO ZWAVE: Node 31: value added 31-114-0-manufacturerId => 345
2021-12-08 09:13:15.339 INFO ZWAVE: Node 31: value added 31-114-0-productType => 4
2021-12-08 09:13:15.343 INFO ZWAVE: Node 31: value added 31-114-0-productId => 81
2021-12-08 09:13:15.376 INFO ZWAVE: Node 31: value added 31-37-0-currentValue => true
2021-12-08 09:13:15.380 INFO ZWAVE: Node 31: value added 31-37-0-targetValue => undefined
2021-12-08 09:13:15.384 INFO ZWAVE: Node 31: value added 31-49-0-Air temperature => 18.8
2021-12-08 09:13:15.390 INFO ZWAVE: Node 31: value added 31-112-0-1 => 1
2021-12-08 09:13:15.394 INFO ZWAVE: Node 31: value added 31-112-0-2 => 1
2021-12-08 09:13:15.399 INFO ZWAVE: Node 31: value added 31-112-0-3 => 1
2021-12-08 09:13:15.402 INFO ZWAVE: Node 31: value added 31-112-0-4 => 1
2021-12-08 09:13:15.407 INFO ZWAVE: Node 31: value added 31-112-0-5 => 1
2021-12-08 09:13:15.410 INFO ZWAVE: Node 31: value added 31-112-0-6 => 1
[... cut ...]
2021-12-08 09:13:15.442 INFO ZWAVE: Node 31 ready: Qubino - ZMNHJD (Flush Pilot)
2021-12-08 09:13:15.492 INFO ZWAVE: Node 31: interview COMPLETED, all values are updated`

And here is the log when i call a manual refresh on "air-temperature" 
'2021-12-08 09:19:33.294 INFO ZWAVE: Calling api refreshCCValues with args: [ 31, 49, [length]: 2 ]
2021-12-08 09:19:33.420 INFO ZWAVE: Node 31: metadata updated: 49-0-Air temperature
2021-12-08 09:19:33.433 INFO ZWAVE: Node 31: value updated: 49-0-Air temperature 18.8 => 18.3'

AlCalzone, 2021-12-09:
@heaven988 these are not driver logs. Please read the link I posted before.

heaven988, 2021-12-09:
@AlCalzone Sorry for the misunderstanding. Here is the driver log
[zwavejs_2021-12-09.log](https://github.com/zwave-js/node-zwave-js/files/7684103/zwavejs_2021-12-09.log)
s

AlCalzone, 2021-12-10:
Please try changing the association from this node to node 1. It should have a target endpoint of "None", change that to 0.
Afterwards, DO NOT re-interview.

heaven988, 2021-12-10:
In groupe association, i control this with Zwave To MQTT.
I have this : 
<html><body>

0 | Lifeline | NodeID_1 | None |  
-- | -- | -- | -- | --

</body>
</html>
When i remove this association, and try a new one : 
Node endpoint : 0
Group : Lifeline
Target Node : NodeID_1
Target Endpoint : 0

I have  again this : 
<html><body>

0 | Lifeline | NodeID_1 | None |  
-- | -- | -- | -- | --

</body>
</html>

I try also this association just for testing :  
<html><body>

0 | Multilevel | NodeID_1 | 0 |  
-- | -- | -- | -- | --

</body>
</html>
<html><body>

0 | Unnamed group 6 | NodeID_1 | 0
-- | -- | -- | --

</body>
</html>

I do not re-interview !

AlCalzone, 2021-12-10:
> I have again this :

Please share a driver log of where you're changing the association

heaven988, 2021-12-10:
```
2021-12-10T09:07:08.576Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[MultiChannelAssociationCCReport]
                                      group id:           1
                                      maximum # of nodes: 1
                                      node ids:           1
                                      endpoints:
```

AlCalzone, 2021-12-10:
The whole sequence please. Start the log, change the association, stop the log, send everything in between.

heaven988, 2021-12-10:
Here is the whol sequence as you ask :
[zwavejs_2021-12-10.log](https://github.com/zwave-js/node-zwave-js/files/7691774/zwavejs_2021-12-10.log)

Thank you for looking this !

AlCalzone, 2021-12-10:
Hm the device doesn't seem to accept that command. I'm not sure if it maybe has a "normal" association blocking it. Try removing this another way: 
[image]

In the window that opens up, paste the following code:
```ts
		const node = driver.controller.nodes.get(31);
		node.commandClasses.Association.removeNodeIds({
			groupId: 1,
			nodeIds: [1],
		});
```
Then try the manual change again and make another log of the entire process.

heaven988, 2021-12-10:
[zwavejs_2021-12-10b.log](https://github.com/zwave-js/node-zwave-js/files/7692131/zwavejs_2021-12-10b.log)
Is this ok ?

AlCalzone, 2021-12-10:
No, that doesn't look like it changed anything. Can you try adding another association, maybe the device reports the sensor values via that one?
Group 2 (should be called Multilevel), target node 1, endpoint none.

heaven988, 2021-12-11:
Hello.
Thank again for trying to understand this case !
So i try :

    Node endpoint : 0
    Group : Multilevel (group 2)
    Target Node : NodeID_1
    Target Endpoint : Non
[zwavejs_2021-12-11.log](https://github.com/zwave-js/node-zwave-js/files/7696794/zwavejs_2021-12-11.log)

heaven988, 2021-12-11:
Another try :

- Node endpoint : 0
- Group : Multilevel (group 2)
- Target Node : NodeID_1
- Target Endpoint : Non
The complete log : [zwavejs_2021-12-11b.log](https://github.com/zwave-js/node-zwave-js/files/7696809/zwavejs_2021-12-11b.log)

And when i try a manual refresh : 

```
2021-12-11T10:39:10.724Z CNTRLR » [Node 031] querying current sensor reading...
2021-12-11T10:39:10.757Z SERIAL » 0x010900131f023104250fe7                                            (11 bytes)
2021-12-11T10:39:10.760Z DRIVER » [Node 031] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      15
                                  └─[MultilevelSensorCCGet]
2021-12-11T10:39:10.769Z SERIAL « [ACK]                                                                   (0x06)
2021-12-11T10:39:10.776Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-12-11T10:39:10.781Z SERIAL » [ACK]                                                                   (0x06)
2021-12-11T10:39:10.785Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-12-11T10:39:10.816Z SERIAL « 0x011800130f00000400c17f7f7f7f00000300000000030200003c              (26 bytes)
2021-12-11T10:39:10.818Z SERIAL » [ACK]                                                                   (0x06)
2021-12-11T10:39:10.821Z DRIVER « [REQ] [SendData]
                                    callback id:     15
                                    transmit status: OK
2021-12-11T10:39:10.861Z SERIAL « 0x010e0004001f0631050122009cc100a6                                  (16 bytes)
2021-12-11T10:39:10.866Z CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2021-12-11T10:39:10.874Z CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 15.6 => 15.6  [Endpoint 0]
2021-12-11T10:39:10.882Z SERIAL » [ACK]                                                                   (0x06)
2021-12-11T10:39:10.889Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Air temperature
                                      scale: Celsius
                                      value: 15.6
2021-12-11T10:39:10.899Z CNTRLR « [Node 031] received current sensor reading:
                                  sensor type: Air temperature
                                  value:       15.6 °C
2021-12-11T10:39:10.907Z SERIAL « 0x010e0004001f0631050122009cc100a6                                  (16 bytes)
2021-12-11T10:39:10.910Z CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2021-12-11T10:39:10.917Z CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 15.6 => 15.6  [Endpoint 0]
2021-12-11T10:39:10.923Z SERIAL » [ACK]                                                                   (0x06)
2021-12-11T10:39:10.928Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Air temperature
                                      scale: Celsius
                                      value: 15.6
2021-12-11T10:39:10.942Z SERIAL « 0x010e0004001f0631050122009cc400a3                                  (16 bytes)
2021-12-11T10:39:10.944Z CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2021-12-11T10:39:10.952Z CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 15.6 => 15.6  [Endpoint 0]
2021-12-11T10:39:10.958Z SERIAL » [ACK]                                                                   (0x06)
2021-12-11T10:39:10.963Z DRIVER « [Node 031] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Air temperature
                                      scale: Celsius
                                      value: 15.6
```

heaven988, 2021-12-14:
Hello,
how can i help to identify the problem ?

AlCalzone, 2021-12-14:
So in the logs above, setting the associations seems to have worked.

Are you sure that no reports are coming automatically? It usually requires the measurements to change by a certain amount first to trigger another one.

If that's not the case, please contact the manufacturer support. They should be able to help you with device-specific settings and issues.

heaven988, 2021-12-14:
Ok. Wich logs say it's working ? 
zwavejs_2021-12-11.log or zwavejs_2021-12-11b.log ? I try two parameter (Target Endpoint : None or 0 for Multilevel).

AlCalzone, 2021-12-14:
Both. But I'm not sure which one is the correct configuration.

I'd start with Endpoint None and try to trigger the automatic report, e.g. by warming the device in your hands or blowing on it.
If that doesn't work, change it to Endpoint 0 and try again.
If that doesn't work, ask the support.

heaven988, 2021-12-14:
Ok, thank you ! Il will report any news on this subject

closed 2022-01-15 by zwave-js-assistant[bot]

al1138, 2022-03-08:
Hello, same issue here, the temperature value doesn't change.
I used before the Python OpenZwave server and the temperature value was sent with the interval defined CommandClass 49.

log of a reinterview
[zwave_js.log-2.txt](https://github.com/zwave-js/node-zwave-js/files/8206676/zwave_js.log-2.txt)

Thanks !

AlCalzone, 2022-03-08:
@al1138 this is likely an association that was set up incorrectly. If I understand correctly, node 1 must be added to association group 2 of your device to enable the multilevel reports:
[image]

You can do that with zwavejs2mqtt.

AlCalzone, 2022-03-09:
Honestly, I think its best to contact Qubino support about this. They should be able to tell you how to configure the device so the temperature reports work.

al1138, 2022-03-09:
Even if it was working well for years with OZW ?

AlCalzone, 2022-03-09:
Yes. OZW used to regularly poll devices, so you might just not have noticed missing reports.

al1138, 2022-03-10:
Thanks.
As far as I understand, you're right ! I found a zwave_device_config.yaml with a polling var... :)
So I need to create a simple automation to refresh the value...

robertsLando, 2022-03-10:
@al1138 No need to create an automation, just add that value to gateway values table and set the desired poll interval

heaven988, 2022-11-25:
> Thanks for your answer ! I think I succced to add the association but the temp value is still stuck [image]

Hello,
is this the association wich is working for you ?
Can you explain how do you manage to refresh the value then ? @robertsLando propose to add a value to gateway values table, but i have no idea how to do it !

robertsLando, 2022-11-25:
Check settings, general section, there is a table there when you can add configurations

heaven988, 2022-11-25:
> 

Thank you ! I try this, hope it will work ... !
[image]