zwave-js/zwave-js #3841: Missing device configuration: Namron Z-wave plug with thermostat (4512725) aka Sunricher SR-ZV9021A

jkaberg, 2021-12-03:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Namron Z-wave plug with thermostat (4512725) aka Sunricher SR-ZV9021A

### Manufacturer ID

Namron: 0x0438 | Sunricher: 

### Product Type

0x0200

### Product ID

0xd004

### Firmware Version

2.3

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4302

### Do you have a manual?

https://www.elektroimportoren.no/docs/lib/4512725-Brukerveiledning-5.pdf

jkaberg, 2021-12-03:
So after doing some investigating it seems like this plug already exists in the config db? -> https://devices.zwave-js.io/?jumpTo=0x0330:0x0201:0xd004:0.0 (I'm not 100% sure!)

As mentioned in the title from my small research this plug currently goes by the name of:

- Namron Z-wave plug with thermostat (art number 4512725)
- Sunricher SR-ZV9021A (the one linked to at Z-wave alliance, counterpart for Namron doesn't exist)

It didn't get recognized when added (see attached node export from zwavejs2mqtt -> [node_66.json.txt](https://github.com/zwave-js/node-zwave-js/files/7651667/node_66.json.txt))

Unsure how to proceed, let me know what you think @AlCalzone

AlCalzone, 2021-12-03:
Our file has a product type of `0x0201` and the linked one `0x0200`. The config parameters also seem to be different from the ones in our file.

I'm making the bot create a PR with a starter config file for this one. From experience, the imported files never match our [style guide](https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide) and have to be cleaned up. If you're willing to help, this will speed up the process of getting the device fully supported.

@zwave-js-bot import config 4302 from zwa

jkaberg, 2021-12-04:
@AlCalzone thanks, I'll get on it later today. 

Just one question; regarding the "2 brands" (or rebrand if you wish) - how should I tackle this? My Namron one's are identified by the manufacturer ID being 0x0438, however when doing inclusion the config file (in its current state) has manufacturer ID 0x0330 and won't match mine?

Should I make 2 config files (one for each)? Or do you want me to do something more elaborate?

AlCalzone, 2021-12-04:
If both are identical, but with different manufacturer IDs, we can do something like with the Aeotec/Fantem devices where the fantem clone imports parameters from the Aeotec device:
Fantem: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x016a/ft100.json
Aeotec: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0086/zw100.json

jkaberg, 2021-12-08:
I didn't have an chance to get to this yet (trying to wrap my head around how the config file is supposed to be). 

However your PR @blhoward2 doesn't include the mentioned devices in OP (just fyi so this issue doesn't get closed before its actually resolved)

blhoward2, 2021-12-08:
My PR is still a draft. You haven't pushed up all of the files yet. It's in there.

PR #3855 (2021-12-30): feat(config): add many new devices from Z-Wave Alliance DB
Closes #3841
Closes #3834
Closes #3802
Closes #3804
Closes #3621
Closes #3609
closed 2021-12-30 by AlCalzone