zwave-js/zwave-js #3847: OBSERVATION: Not sure if this is zwave-js, zwavejs2mqtt, or HA, or an issue with a Device Config file

GrizzlyAK, 2021-12-05:
I don't know if this is a bug, oversight, deficiency in the Device Config file, HA, zw2m, or HA, but did want to point it out just in case. I'm using zw2m v5.12.0, zwjs v8.7.6, with a locally installed Config file (the one currently in Master) for the ZEN15 (including templates).
FYI, @blhoward2 

I updated the FW on 5 Zooz ZEN15s from 1.4 to 1.6 using the zw2m CP OTA update (which worked perfectly, btw!). This update added the two new Config Params (#33 & #34), shown below (from the Config file - zen15.json:

```	
{
	"#": "33",
	"$if": "firmwareVersion >= 1.6",
	"label": "Auto-On Timer",
	"valueSize": 1,
	"unit": "minutes",
	"minValue": 0,
	"maxValue": 99,
	"defaultValue": 0,
	"unsigned": true,
	"options": [
		{
			"label": "Disable",
			"value": 0
		}
	]
},
{
	"#": "34",
	"$if": "firmwareVersion >= 1.6",
	"label": "Auto-Off Timer",
	"valueSize": 1,
	"unit": "minutes",
	"minValue": 0,
	"maxValue": 99,
	"defaultValue": 0,
	"unsigned": true,
	"options": [
		{
			"label": "Disable",
			"value": 0
		}
	]
},

```
Upon successfully completing the FW update and restarting the device (re-interview) (all automatic), in each case the new Parameters appeared in Configuration with the custom value 255, NOT the Default. When clicked on, Default was the other option in the drop-down, and could be set, but it is curious that the default was not chosen upon restart. All other existing parameters were as they were prior to the update. See image below.
[image]

blhoward2, 2021-12-05:
Eh, it's not much of anything. We don't set the default every as it would clobber things. It's just for the UIs to display. We just set it to what the manufacturer says the default is supposed to be.

GrizzlyAK, 2021-12-05:
I see, that makes sense.

> What's more curious is why they're setting it to a value not in the range defined.
right

GrizzlyAK, 2021-12-05:
let me check.

GrizzlyAK, 2021-12-05:
Yeah, the device reported 255. Down toward the bottom is where I changed it.

[image]

AlCalzone, 2021-12-05:
I wonder if 255 is just the default "empty" memory and they didn't set the value during upgrade on the device

blhoward2, 2021-12-05:
I let Zooz know.

closed 2021-12-05 by blhoward2

blhoward2, 2021-12-09:
Zooz said they will fix this in new fw, likely in January.