zwave-js/zwave-js #3848: Missing device configuration: <Vesternet Z-Wave Switch>

richardob50, 2021-12-05:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Vesternet Z-Wave Switch

### Manufacturer ID

0x0200

### Product Type

0x0200

### Product ID

0x0200

### Do you have a manual?

https://cdn.shopify.com/s/files/1/0066/8149/3559/files/VES-ZW-DIM-001.pdf

AlCalzone, 2021-12-06:
@zwave-js-bot config help wanted

AlexCPU, 2022-01-17:
Based on the device handlers that Vesternet have [published for Smartthings](https://github.com/vesternet/smartthings-z-wave-device-handlers), it looks like the answers in the initial request should be:

### Manufacturer ID
0x0330 (Which is actually registered to "ShenZhen Sunricher Technology, Ltd")
### Product Type
0x0200
### Product ID
0xD00C

I'll try and add the whole range.  Ironically, I haven't bought any yet as zwave-js doesn't support them...

richardob50, 2022-01-17:
Thanks. Would have through the the devices would be on database already given the z wave certification process, but I am obviously assuming too much.

They can be configured fine by temporarily switching to zwave2mqtt and then switching back, but not ideal obviously. From a European 2 wire perfective they offer the best comp ability.

PR #4105 (2022-01-18): feat(config): add Vesternet devices
Added dimmer and switch from Vesternet range.

Fixes #3848 
Fixes #3389

I've also proposed to add dimmer type (leading or trailing edge) to master template, if approved, I can update all the existing instances in other config files.  A quick search shows one or other of these properties in 9x other devices and 3x manufacturer templates (used in a further 6 devices).  Which is admittedly less than I expected when I was writing it
closed 2022-01-18 by AlCalzone