zwave-js/zwave-js #3852: Devolo - Smoke Detectors: After switching to node-zwave-js, sometimes they sound their alert

brotkastn, 2021-12-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Differences between openzwave and node-zwave-js in handling the "Devolo - Home Control Smoke Detector"

What do you observe?
I have three of said devices. After switching from openzwave to node-zwave-js, two of the smoke detectors have been signaling a "warning" - short beep, like if the battery is dead. After getting new batteries, and still having them sound their alert, i looked at the voltages of the batteries, and they are fine - so it is something different.
After manually waking up the smoke detectors (1 second press on button), they will wake up, connect to the controller, and everything will be fine. Until, after some time, they start beeping again.

The two smoke detectors that do beep use other nodes as their hops, the one that does not is near the controller. Looking at the network-graph in the web-ui, there are only two connections to the controller, the others all have only neighbours without a route to the controller - but they work.

My guess is that there is something different in the wakeup communication, and the smoke detector thinks it lost its connection. The alarms have been are around the times of the scheduled wakeup.

What did you expect to happen?
Being able to sleep :(

Steps to reproduce the behavior:
  1. Buy several of those Devolo Smoke Detectors.
  2. Have some far away from the Controller
  3. Sleep
  4. Hear alarm

### Device information

Manufacturer: Devolo
Model name: Home Control Smoke Detector | 09813
Node ID in your network: 18, 32

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwavejs2mqtt: 6.0.3
zwave-js: 8.8.3
home id: 3814029858
home hex: 0xe3557a22

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Same devices with openzwave did not show the same behaviour

### Attach Driver Logfile

[zwavejs_2021-12-05.log](https://github.com/zwave-js/node-zwave-js/files/7656852/zwavejs_2021-12-05.log)

Node 18 started beeping around 16:50. Node 25 is the smoke detector that works fine.

brotkastn, 2021-12-06:
This might be a better [zwavejs_2021-12-06.log](https://github.com/zwave-js/node-zwave-js/files/7658940/zwavejs_2021-12-06.log)

Today at 6:46 local time, which is 5:46 in the logfile, the smoke detector decided to beep again. I took out the batteries, and reinserted them at ~8:40. Also, the reinsertion was followed by route updates, so thats in there as well.

Thank you for your work, by the way. For my other devices, zwavejs is far more reliable and quite a bit faster than the two other implementations :)

AlCalzone, 2021-12-06:
> 5:46 in the logfile, the smoke detector decided to beep again
> I took out the batteries, and reinserted them at ~8:40.

All the device sends then are wakeup notifications, but the driver sends it back to sleep since it has no queued messages.
I see a couple of those commands failing with "NoAck", but that is relatively normal for battery-powered devices, which often go back to sleep before the ACK has reached the controller.

At `2021-12-05T23:03:02.652Z` (and some other timestamps), the driver fails to refresh the wakeup interval with the same error though, which might indicate that you do have a range issue here. You can try doing a network heal to let the controller figure out new routes.

brotkastn, 2021-12-25:
So, i did the heal on December 6 and it finished after 6 hours. Everything looked fine, until there were those Warning-Beeps on Node 18 again. I tried to log the timestamps, but since some of those were handled by my wife, and sometimes it was in the middle of the night, i do not have a complete list. There were, at least:

December 11, 23:07 ; December 18, 05:45 ; December 20, 07:27, December 21, 07:10 (after that, node 18 got disabled).

I do know that this looks like a Problem with my Z-Wave Setup, but i never had those problems with a year of running https://github.com/OpenZWave/Zwave2Mqtt . Also, the maximal Distance between the nearest active router or the controller is maybe 5 meters, and after the migration to node-zwave-js i did not change the z-wave network - neither adding new nodes nor moving the current ones around.

Since zwave2mqtt is dead, and i would really like the smoke detectors to stop waking me up, i would really like to help to get the support for those devices better. I however, failed to find reproducable way to get them to beep. Also, i have not found a technical document that describes the behaviour of the smoke alarms that would mention the "lost connection to zwave" beep. What i did try is to remove one of the routers, and then the kitchen smoke alarm had the beeping twice (which did work prevously).

Do you know if ozwd handles failed messages differently than node-zwave-js does? Maybe this might help.

Thank you for your reply, and the whole team of node-zwave-js for the work. My other devices work quite a bit "faster", for example those "active routers" are switchable power outlets. node-zwave-js is quite a lot faster, and works for every "switch". (ozwd needed a second "switch toggle" in 1 of 20 cases)

AlCalzone, 2021-12-27:
Since the Danfoss LC-13 thermostats and the Devolo variant require a certain set of commands after waking up, I tried finding a reference to something similar for the smoke detectors in the OZW repo but there seems to be none. Also if there were, I'd expect more users to report this issue - you're the only one so far.

When the nodes wake up, they send a command to the controller. Afterwards this should happen:
1. The controller itself acknowledges the command.
2. zwave-js checks if there are pending messages and sends them
3. zwave-js sends the node back to sleep

There isn't really a lot that zwave-js can be doing wrong here. 
1\. is handled on the protocol level, initiated by the stick, and we have no control over that.
2\. didn't seem necessary in your logs, unless you queued a config change or something
3\. also happens as can be seen in your logs above

Honestly, this is a case for the manufacturer support. They should be able to tell you why the sensors start beeping and what's necessary to prevent them from doing so.

brotkastn, 2021-12-28:
First of all, thank you for looking up if there were device specific quirks for the "Devolo Smoke Detector" in OZW.

I reread the Manual, and the only beep they mentioned was the "battery low" beep. So, surely enough, in the homeassistant some of the 3 smoke detectors had "battery low" states in their recent history. This was weird because: 
1. i had several looks inside the zwave2mqtt ui if the battery was empty the first times the smoke detectors started beeping after the migration
2. the migration to node-zwave-js happened because one of the smoke detectors was empty, and i just changed all batteries because they were all quite old.

I did look at the voltages of the batteries (as stated in the first post). However, according to recent tests i suspect the batteries. if i measure the voltages they are fine (3,0-2,9V). However, once there is the slightest load applied they drop quite a bit. I changed to a new set of batteries in two of the detectors (which is extra annoying, because the CR123 batteries are not really that common and arent cheap either). Also, this would explain why the garage detector was the first to be weird, because the lower temperatures should cause the batteries to drop sooner.

After reading your comments, i am optimistic that this might have been the problem all along. I will close this in a few days if this is indeed the case.

Thank you for your support!

brotkastn, 2022-01-03:
I am happy to report back that i have yet to encounter any issue. Thank you for your work, and have a nice 2022!

closed 2022-01-03 by brotkastn

AlCalzone, 2022-01-03:
You too, glad you figured it out.