zwave-js/zwave-js #3854: Driver interviews USB controller during first run

kpine, 2021-12-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Driver interviews the USB controller, starting with an attempt to query for `VersionCCCommandClassGet`. I don't think it's valid to do so?

What do you observe?

When starting node-zwave-js for the first time, it attempts to interview the controller, starting with an attempt to send a `VersionCCCommandClassGet`. That fails after 3 tries. The controller is marked dead, then the interview completes and it is marked alive again.

What did you expect to happen?

I don't expect a `VersionCCCommandClassGet` to work on a USB serial controller. It also doesn't make much sense for the controller to go dead and then alive?

There don't seem to be any problems caused by these events, it just seems like a weird thing to do.

Steps to reproduce the behavior:
  1. Start node-zwave-js. Wipe the cache files if necessary.
  2. Driver queries controller information.
  3. Driver tries to send `VersionCCCommandClassGet` command to controller.
  4. Restart node-zwave-js, the interview is not attempted again

### Device information

Manufacturer: Zooz
Model name: ZST10 700
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.8.3
`zwavejs2mqtt` branch: 6.0.2

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2021-12-05.log](https://github.com/zwave-js/node-zwave-js/files/7657400/zwavejs_2021-12-05.log)

AlCalzone, 2021-12-07:
Can you easily test if the same happens on 8.7.x?

AlCalzone, 2021-12-07:
I double checked what happens with my stick:
```
14:59:36.036 CNTRLR « [Node 001] received response for protocol info:
                      basic device class:    Static Controller
                      generic device class:  Static Controller
                      specific device class: PC Controller
                      node type:             Controller
```

and here's yours:
```
2021-12-06T01:50:38.478Z CNTRLR « [Node 001] received response for protocol info:
                                  basic device class:    Static Controller
                                  generic device class:  Static Controller
                                  specific device class: Gateway
                                  node type:             Controller
```

The difference is the specific device class `Gateway`, which must support `Manufacturer Specific` and `Version` CCs while the `PC Controller` has no such requirement.

In any case, no CCs should be queried on the controller.

PR #3863 (2021-12-07): fix: skip CC interview step for controller node
Apparently there are some device classes for controllers that caused an incorrect and unnecessary interview step.

fixes: #3854
closed 2021-12-07 by AlCalzone