zwave-js/zwave-js #3865: BE469ZP locks going dead sporadically

cdonovan, 2021-12-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Issuing lock commands

What do you observe?
Lock nodes become dead, but are immediately revived with a manual ping

What did you expect to happen?
Locks to remain alive

Not sure if this is the same/similar issue as #3842. From looking at my own logs, it does appear a bit different, but I don't really understand the zwave protocal well enough to know for sure.  Sorry in advance if it is the same (could be closed or appended to that issue).

### Device information

Manufacturer: Allegion
Model name: BE469ZP
Node ID in your network: 62, 63

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I started noticing these issues since switching from s0 to s2.

### Attach Driver Logfile

[zwavejs_2021-12-06.log](https://github.com/zwave-js/node-zwave-js/files/7670892/zwavejs_2021-12-06.log)

AlCalzone, 2021-12-07:
```
                                    transmit status: NoAck
2021-12-07T00:40:11.807Z CNTRLR   [Node 062] Node 62 did not respond after 3 attempts, it is presumed dead
```
You have sporadic connectivity problems (in the order of several seconds). NoACK means that the node did not acknowledge the command or the ACK got lost.

At the same time multiple other nodes fail, so I'm guessing your controller is affected.

Z-Wave sticks in particular are prone to interference by USB ports, especially by USB3 ports. We recommend putting the stick in a suitable location:

-   on an **USB extension cord** (this works wonders!)
-   away from other USB ports
-   away from metallic surfaces
-   and especially not in the back of a server rack

cdonovan, 2021-12-07:
My server is in a closet, but when I built out the system, I figured there could be interference, and have had it attached to a 5ft extension cord from the beginning.  It's still in the closet but attached to a piece of drywall quite far away from any conduit, wires or metal.  I also never saw issues like these when I was running w/ s0 security (which I thought was actually also less performant).  I'll see if I can move my zwavejs2mqtt instance to a new device that is more in the open to see if the situation improves.  RIght now I just have an automation set up to ping the node whenever its status turns to dead, which has been working fine, but is definitely a hack.

If it were a controller issue, wouldn't I see more failures of other nodes?  The only nodes that actually fail are my two BE469ZP locks (which are both probably the farthest devices in my mesh). There are many other mains-powered devices in between the locks and the controller however, so I guess coverage itself is likely not the issue. I would just expect to see a lot more dropped messages from other nodes if it were interference at the controller.

AlCalzone, 2021-12-08:
> I also never saw issues like these when I was running w/ s0 security

That's most likely just a coincidence. Security encapsulation happens at the "application" level, this issue is more of a physical nature. If anything, using S0 should make it more apparent because there are more messages exchanged which can fail.

> If it were a controller issue, wouldn't I see more failures of other nodes?

The automatic retries might be mostly masking the issue for you. Here's an example from Node 18 which only succeeds on the 2nd attempt:
```
2021-12-07T00:40:27.109Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      91
                                  └─[MultilevelSwitchCCSet]
                                      target value: 0
...
2021-12-07T00:40:32.041Z DRIVER « [REQ] [SendData]
                                    callback id:     91
                                    transmit status: NoAck
2021-12-07T00:40:32.043Z CNTRLR   [Node 018] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2021-12-07T00:40:32.545Z SERIAL » 0x010a00131203260100255bae                                          (12 bytes)
2021-12-07T00:40:32.546Z DRIVER » [Node 018] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      91
                                  └─[MultilevelSwitchCCSet]
                                      target value: 0
...
2021-12-07T00:40:33.437Z DRIVER « [REQ] [SendData]
                                    callback id:     91
                                    transmit status: OK
```

Try taking a look at the statistics in zwavejs2mqtt (they are hidden a bit). I bet you'll see a couple of nodes where this entry is significantly greater than 0
[image]

cdonovan, 2021-12-08:
@AlCalzone, thanks very much for all the info you've given.  The most troublesome lock (node 62) has a fair number of dropped commands, the other lock actually seems relatively good (1% failures). Most of the other nodes I checked have no TX failures but between 1-10% RX failures.  I think getting this info very clearly indicates that there are some issues w/ my mesh, so please feel free to close this out.  I'm going to try moving my instance to an RPI and position it closer to the troublesome nodes. Thanks again.  For info, here's the stats I saw for nodes towards the problematic area of the house.

Node 62 (~5ft from nodes 54 & 63 w/ one wall - drywall - in between:
Commands Dropped TX: 3 (~6-7%)
Commands TX: 45
Commands dropped RX: 27 (~23%)
Commands RX: 45

Node 63:
Commands Dropped TX: 1 (~1%)
Commands TX: 93
Commands dropped RX: 12 (~1%)
Commands RX: 129

Node 54 (right next to node 63):
Commands Dropped TX: 0 (0%)
Commands TX: 62
Commands dropped RX: 20 (~7%)
Commands RX: 284

closed 2021-12-08 by cdonovan