zwave-js/zwave-js #3867: Double tap not supported on GE/Jasco 45609 switches

vitiock, 2021-12-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Missing configuration values in the GE/Jason 45609 config file
What do you observe?
double taps are not available for this product, but are for similar jasco/ge switches
What did you expect to happen?
double tap should trigger an event

Steps to reproduce the behavior:
  1. Go to 'a 45609 light switch and double tap on or off'
  2. Click on 'the silly logs for zwave js'
  3. Scroll down to 'the bottom'
  4. See there is nothing representing that the double tap happened

### Device information

Manufacturer: Jasco/GE
Model name: 45609
Node ID in your network: 12

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [X] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js@8.8.3
zwave-js/server@1.13.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Did not work with zwavejs but did previously work with smartthings

### Attach Driver Logfile

N/A  Empty file attached

[log.txt](https://github.com/zwave-js/node-zwave-js/files/7671984/log.txt)

vitiock, 2021-12-07:
Likely related to or the same as https://github.com/zwave-js/node-zwave-js/issues/1277 but for another device

AlCalzone, 2021-12-07:
We don't have the corresponding compat flag set for this one.
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0063/45609.json

/cc @blhoward2

AlCalzone, 2021-12-07:
@zwave-js-bot add compat flag to 0x0063:0x5257:0x3533:0.0
```json
// Enable double-tap support
"treatBasicSetAsEvent": true
```

AlCalzone, 2021-12-07:
@vitiock can you test [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/8c30f52cb75a2dccd599f5a57ca135a449fabc55/packages/config/config/devices/0x0063/45609.json) as described [here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files)?

You might have to re-interview the device for the changes to have an effect.

vitiock, 2021-12-07:
I got the file to load, however I still am not seeing anything in the logs when I double tap the switch

blhoward2, 2021-12-07:
Did you reinterview the device?

vitiock, 2021-12-07:
I did I even see the compat change in the dump of the network

```
{
            "nodeId": 2,
            "index": 0,
            "status": 4,
            "ready": true,
            "isListening": true,
            "isRouting": true,
            "isSecure": false,
            "manufacturerId": 99,
            "productId": 13619,
            "productType": 21079,
            "firmwareVersion": "3.25",
            "name": "",
            "location": "",
            "deviceConfig": {
              "filename": "/cache/config/45609.json",
              "isEmbedded": false,
              "manufacturer": "GE/Jasco",
              "manufacturerId": 99,
              "label": "45609",
              "description": "On/Off Relay Switch",
              "devices": [
                {
                  "productType": 21079,
                  "productId": 13619
                },
                {
                  "productType": 25443,
                  "productId": 13619
                }
              ],
              "firmwareVersion": {
                "min": "0.0",
                "max": "255.255"
              },
              "associations": {},
              "paramInformation": {
                "_map": {}
              },
              "compat": {
                "valueIdRegex": {},
                "treatBasicSetAsEvent": true
              }
            },
            "label": "45609",
            "interviewAttempts": 0,
            "endpoints": [
              {
                "nodeId": 2,
                "index": 0,
                "deviceClass": {
                  "basic": {
                    "key": 4,
                    "label": "Routing Slave"
                  },
                  "generic": {
                    "key": 16,
                    "label": "Binary Switch"
                  },
                  "specific": {
                    "key": 1,
                    "label": "Binary Power Switch"
                  },
                  "mandatorySupportedCCs": [
                    32,
                    37,
                    39
[... 315 lines cut ...]
              {
                "id": 119,
                "name": "Node Naming and Location",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 134,
                "name": "Version",
                "version": 1,
                "isSecure": false
              }
            ],
            "interviewStage": "Complete",
            "deviceDatabaseUrl": "https://devices.zwave-js.io/?jumpTo=0x0063:0x5257:0x3533:3.25",
            "statistics": {
              "commandsTX": 1,
              "commandsRX": 0,
              "commandsDroppedRX": 0,
              "commandsDroppedTX": 0,
              "timeoutResponse": 0
            },
            "highestSecurityClass": -1
          }
          ```

AlCalzone, 2021-12-09:
@vitiock It might be necessary to add the controller to an extra association group.
Please try again with [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/add-compat-flag-1638916658/packages/config/config/devices/0x0063/45609.json).

blhoward2, 2021-12-10:
Are we sure this is necessary? Does this device not send central scene events?

blhoward2, 2021-12-10:
Ah, yeah. The controller needs to be added to group 3.

vitiock, 2021-12-11:
Hey tried with the file, not sure I can test this though as when I try to add an association I get back an error "Node * does not support associations! (ZW0302) driverFunction undefined" so might be something on my side

AlCalzone, 2021-12-11:
> driverFunction undefined

Are you not using the associations UI for this?

vitiock, 2021-12-11:
When I use that UI from zwavejs2mqtt the drop down is empty and there are no existing associations

AlCalzone, 2021-12-12:
To be sure you're not missing any steps:
1. replace the file with the one I linked 
2. restart zwavejs2mqtt
3. re-interview the node
4. try adding the association to group 3

vitiock, 2021-12-12:
Number 4 is where I get stuck, zwavejs2mqtt has no associations/groups for anything for me.

AlCalzone, 2021-12-12:
Can you make a node dump? If the config file is loaded, there should be 3 associations

blhoward2, 2021-12-12:
We don't have any association groups in this file, right? I bet the device isn't reporting them.

blhoward2, 2021-12-12:
Oh, nm you've done that. So clearly the custom file isn't getting loaded.

PR #3868 (2021-12-15): fix(config): add double tap support to GE/Jasco 45609
fixes: #3867

**TODO:**
- [ ]	Change PR title to be more specific
closed 2021-12-15 by AlCalzone