zwave-js/zwave-js #3872: ARZ Solar (productType 0x0003 - productId 0x0011) Configuration - "COPY" ARZ Z-wave (productType 0x0003 - productId 0x0012)

gitd8400, 2021-12-08:
My ARZ Solar does not allow configuration.
However the ARZ Solar is pretty much the same as the ARZ Z-wave but with an added solar power source.
One of the problems I seem to have is that my shutter/cover does not fully open when the command is issued. At 100% open I still see a part of the shutter covering the window. 

**Describe the solution you'd like**
Could the configuration of the ARZ Z-wave be copied in the configuration of the ARZ Solar. By having a calibration option I think I might be able to solve the "100% open"-problem.

AlCalzone, 2021-12-08:
@blhoward2 are you touching these in your open PR or should we do this separately?
There's also a duplicate file for FW 1.1+ which doesn't make sense to have at all.

blhoward2, 2021-12-08:
I processed all zwave alliance db entries >4,075. If it's in there and above that, I'll push it up to my PR as I finish it.

AlCalzone, 2021-12-08:
Ok its not in there

AlCalzone, 2021-12-08:
@gitd8400 are you sure about the product ID? We have these files currently:
- `0x0003:0x0111` ARZ 1.1 (whatever this is)
- `0x0003:0x0011` ARZ Z-Wave
- `0x0003:0x0112` ARZ Solar

gitd8400, 2021-12-08:
> @gitd8400 are you sure about the product ID? We have these files currently:
> 
>     * `0x0003:0x0111` ARZ 1.1 (whatever this is)
> 
>     * `0x0003:0x0011` ARZ Z-Wave
> 
>     * `0x0003:0x0112` ARZ Solar

You are right. I messed up the id's. Sorry. But the ARZ Solar is a ARZ Z-Wave with solar power component.

blhoward2, 2021-12-08:
I'm confused. Isn't @AlCalzone saying we already have a file for that fingerprint?

PR #3874 (2021-12-10): fix(config): cleanup and consolidate ARZ Roller Shutters
fixes: #3872
closed 2021-12-10 by AlCalzone