zwave-js/zwave-js #3877: CNTRLR   Turning RF off... Log Entry - Network is Unresponsive

kars85, 2021-12-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown. Logs are attached, but only two nodes appear to be responsive (Node 9 & 30)

What do you observe?
Network was stable for the past 72+ hours, then last night/early this morning, I noticed automations didn't fire.

What did you expect to happen?
Network to be stable. Heals are initiated, but do not proceed after 5-10 minutes of initiating. Manual invocations via zwavejs2mqtt of commands or configuration changes to devices do not appear to transmit (i.e. no entry in log every being sent via the controller)

Steps to reproduce the behavior:

### Device information

Manufacturer: 
Model name: 
Node ID in your network: All but Nodes 9 & 30 are experiencing issues. The only unique thing about these two functioning sensor nodes is that they are battery capable, but powered via USB, and that they're all Aeotec sensors. Switches and outlets are unresponsive and not reporting any status (healthy/failed/ping response acknowledgements).

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.1.0
zwave-js: 8.9.0-beta.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Watchtower pulled the :latest image down at 4AM on 12/8. Despite the 8.9.0-beta.0 being tagged as prerelease in github, somehow the image pulled was recognized as the :latest release instead of 8.8.3.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I will attempt to reboot the container and see if the network resumes, and will await further instruction if it would be beneficial to drop down to 8.8.3.

### Attach Driver Logfile

[zwavejs_2021-12-08.log](https://github.com/zwave-js/node-zwave-js/files/7686416/zwavejs_2021-12-08.log)
[zwavejs_2021-12-09.log](https://github.com/zwave-js/node-zwave-js/files/7686417/zwavejs_2021-12-09.log)

kars85, 2021-12-09:
Restarting the container appears to have the network functional again. Here are current logs after the container restart
[post_container_restart_zwavejs_2021-12-09.log](https://github.com/zwave-js/node-zwave-js/files/7686643/post_container_restart_zwavejs_2021-12-09.log)
.

AlCalzone, 2021-12-09:
I'm seeing several transmit attempts that take exceedingly long (20s+), even in the "functional" log. In fact after the startup is over, there isn't much of a difference. So far this looks to be exactly what I described here: https://github.com/zwave-js/node-zwave-js/issues/3842#issuecomment-985849934

Side note: The "CNTRLR Turning RF off..." log entry is for the 200 milliseconds or so where you opened the network map and it requested all neighbors. This does a lot of memory access on the stick and requires the radio to be turned of meanwhile.

kars85, 2021-12-10:
Thanks - wasn't sure if this was an unrelated issue, or still tied to firmware. What is the most efficient way to parse the zwave-js logs if I want to take a deeper dive on my end before opening an issue? I use Notepad++ heavily - is there a certain language or plugin I can use to help identify where problems lie?

AlCalzone, 2021-12-10:
I'm opening the log files in VSCode (mainly because I use it for everything else) and then use the search bar to look for messages from and to specific nodes. Nothing fancy, but you'll need to get an eye for things.

kars85, 2021-12-10:
One other thought - is SiLabs aware of there 700 series problems? From what I can tell, this is widespread and potentially hampering forward development of the platform for not only this project, but any other Z-wave vendor.

It appears like they may be on a pseudo-quarterly release schedule, but likely more of an "as needed" basis. I'm registered on their community, and can't see anything definitive mentioned about what we're seeing here, so thought I would ask. I thought about submitting a case, but not sure if that is futile or worthwhile. Thoughts?

AlCalzone, 2021-12-10:
I have an open support ticket with them. The HomeSeer devs do too and we've asked Zooz and Aeotec to try and reproduce so they can give some input too.

AlCalzone, 2021-12-15:
Duplicate of #3906

closed 2021-12-15 by AlCalzone