zwave-js/zwave-js #3880: Zooz Titan Water Valve Actuator (ZAC36) reporting false information

shred86, 2021-12-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I apologize in advance if this is not an issue with node-wave-js but rather the device firmware. I've gone back and forth with Zooz for over a month sending them multiple troubleshooting logs and they claim the issue is on the node-wave-js side.

What do you observe?
This device is reporting battery related values but the device itself does not have a battery. The specific items as seen from the zwavejs2mqtt front end are:
- [21-128-0-level] Battery level
- [21-128-0-isLow] Low battery level
- [21-113-0-Power Management-Backup battery level status] Backup battery level status

What did you expect to happen?
Zooz claims their early firmware versions reported some battery related information as the device originally was going to have a battery back up. They've sent me updated firmware which supposedly removed all battery related reporting. Even after applying this firmware, resetting the device, unpair/repairing, it still reports battery related statuses/values.

Steps to reproduce the behavior:
  1. Pair the Zooz Titan Water Valve Actuator (ZAC36) with node-zwave-js.
  2. Check reported information/statuses in the zwavejs2mqtt front end.

### Device information

Manufacturer: Zooz
Model name: ZAC36
Node ID in your network: 21

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.8.2
`zwavejs2mqtt` branch: 6.0.2

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2021-12-11.log](https://github.com/zwave-js/node-zwave-js/files/7696445/zwavejs_2021-12-11.log)

AlCalzone, 2021-12-11:
It still reports Battery CC in the secure command list:
```
2021-12-11T04:44:23.790Z CNTRLR « [Node 021] received secure commands (S2_Authenticated)
                                  supported CCs:
                                  · Binary Switch
                                  · Association
                                  · Multi Channel Association
                                  · Association Group Information
                                  · Notification
                                  · Version
                                  · Manufacturer Specific
                                  · Device Reset Locally
                                  · Indicator
                                  · Powerlevel
                                  · Firmware Update Meta Data
                                  · Multilevel Sensor
                                  · Configuration
                                  · Battery
```

and it responds to the version check for Battery CC:
```
2021-12-11T04:44:24.654Z DRIVER « [Node 021] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -72 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 110
                                    └─[VersionCCCommandClassReport]
                                        CC:      Battery
                                        version: 1
```

and it responds to battery queries:
```
2021-12-11T04:44:24.968Z DRIVER » [Node 021] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      50
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 178
                                    └─[BatteryCCGet]
...
2021-12-11T04:44:25.002Z DRIVER « [Node 021] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -72 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 115
                                    └─[BatteryCCReport]
                                        level:  0
                                        is low: false
```

So whatever Zooz sent you, it didn't remove the battery functionality.

shred86, 2021-12-12:
Thanks @AlCalzone, that's what I suspected. I've emailed Zooz support back with this information and they are looking into it. I'll post any updates I get here.

blhoward2, 2021-12-17:
I've also informed Zooz support. They've emailed me a few times asking about this.

closed 2022-01-03 by zwave-js-assistant[bot]

mikedm139, 2022-01-18:
I spoke to Zooz support about this as well and they provided me a firmware update to test. It removes the `battery level` and `low battery level` reporting but leaves the backup battery reporting in place. I asked about the backup battery and support said that they are working on a backup battery for the unit which will be available for sale separately.