zwave-js/zwave-js #3881: Node interview stoped working - Unknown manufacturer 0xXXXX - Unknown product 0xXXXX

luzik, 2021-12-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I am unable to interview my two Aeotec Multisensor 7 with zwave2mqtt 6.1.0, node-zwave-js 8.9.0-beta.0

Inclusion was successful (S2) but after that I got:

```
2021-12-11 13:13:17.007 DEBUG SOCKET: Event ZWAVE_API emitted to s3x2TeYGd5jVd3f3AAAH
2021-12-11 13:13:17.008 INFO ZWAVE: Calling api refreshInfo with args: [ 25, [length]: 1 ]
2021-12-11 13:13:17.014 INFO ZWAVE: Node 25: interview started
2021-12-11 13:13:17.016 INFO ZWAVE: Success zwave api call refreshInfo undefined
2021-12-11 13:13:19.274 INFO ZWAVE: Controller status: Scan completed
2021-12-11 13:13:19.275 INFO ZWAVE: Network scan complete. Found: 9 nodes
2021-12-11 13:13:19.276 INFO ZWAVE: Node 25 is now dead
2021-12-11 13:13:19.295 INFO ZWAVE: Node 25: interview stage PROTOCOLINFO completed
2021-12-11 13:13:19.377 INFO ZWAVE: Node 25: interview stage NODEINFO completed
2021-12-11 13:13:19.445 INFO ZWAVE: Node 25: has returned from the dead
2021-12-11 13:13:19.477 INFO ZWAVE: Node 25 is now awake
2021-12-11 13:13:19.595 INFO ZWAVE: Node 25: interview stage COMMANDCLASSES completed
2021-12-11 13:13:19.596 INFO ZWAVE: Node 25: interview stage OVERWRITECONFIG completed
2021-12-11 13:13:19.598 INFO ZWAVE: Node 25: interview stage COMPLETE completed
2021-12-11 13:13:19.603 DEBUG ZWAVE: Node 25: mapping absence of CC 128 (Battery) to node property 'isBatteryPowered
2021-12-11 13:13:19.604 INFO ZWAVE: Node 25 ready: Unknown manufacturer 0xXXXX - Unknown product 0xXXXX (0xXXXX)
2021-12-11 13:13:19.608 INFO ZWAVE: Node 25: interview COMPLETED, all values are updated
```

I have two other same piece of hardware that was added to network about month ago without any problems, so something changed

### Device information

Manufacturer: Aeotec
Model name: ZST10-700 + Unknown product 0x0018

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It worked about month ago. I have 4 x Aeotec multisensor 7. Two included about month ago, and two that don't want to work. They ware bought all together. So I believe they use same firmware.

### Attach Driver Logfile

{
  "id": 25,
  "name": "",
  "loc": "",
  "values": [],
  "groups": [],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "hexId": "0xXXXX-0xXXXX-0xXXXX",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0xXXXX:0xXXXX:0xXXXX:0.0",
  "productLabel": "Unknown product 0xXXXX",
  "productDescription": "0xXXXX",
  "manufacturer": "Unknown manufacturer 0xXXXX",
  "protocolVersion": 3,
  "zwavePlusVersion": 2,
  "zwavePlusNodeType": 0,
  "zwavePlusRoleType": 6,
  "nodeType": 1,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": true,
  "security": "S0_Legacy",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": false,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 7,
    "specific": 1
  },
  "deviceId": "undefined-undefined-undefined",
  "status": "Awake",
  "interviewStage": "Complete",
  "statistics": {
    "commandsTX": 84,
    "commandsRX": 38,
    "commandsDroppedRX": 42,
    "commandsDroppedTX": 0,
    "timeoutResponse": 5
  },
  "lastActive": 1639224799602,
  "isBatteryPowered": false,
  "_name": "NodeID_25"
}

AlCalzone, 2021-12-12:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and **attach** it here as a file.
While you're making the log, re-interview the problematic devices.

luzik, 2021-12-12:
Node 26

[zwavejs2mqtt_2021-12-12.log](https://github.com/zwave-js/node-zwave-js/files/7698904/zwavejs2mqtt_2021-12-12.log)

luzik, 2021-12-13:
I have a clue. After inclusion security was shown as S2 Auth, but as you can see now it is S0_Legacy. Second time I checked S2 Access with success, but after save it is S2 UnAuth!

AlCalzone, 2021-12-14:
> Second time I checked S2 Access with success, but after save it is S2 UnAuth!

What do you mean by "after save"?

luzik, 2021-12-14:
While in inclusion web dialog of zwave2mqtt popup there is check list with security classes. First time I left if default to S2 Auth, entered PIN close dialog, but new node was shown on list with security S0_Legacy. Then I removed node, include it again but with extra S2 Access option checked. Inclusion was successful, but on a list security is set to S2 UnAuth.

AlCalzone, 2021-12-14:
Ok, I'll need to see a driver log of that. The log above isn't a driver log, please check out the link I posted before.

robertsLando, 2021-12-14:
> @robertsLando are the security classes that are not requested by the device disabled or checkable?

Them are all checkable

AlCalzone, 2021-12-14:
You should change that. It doesn't make sense to "add" security classes to the list requested by the device. Only removing them.

robertsLando, 2021-12-14:
> You should change that. It doesn't make sense to "add" security classes to the list requested by the device. Only removing them.

Ok!

luzik, 2021-12-14:
Here it is full inclusion log + interview ..after interview it changing security class from S2 Auth to S2 Unauth
[zwavejs_2021-12-14.log](https://github.com/zwave-js/node-zwave-js/files/7714160/zwavejs_2021-12-14.log)

PR #3900 (2021-12-15): fix: don't query all security classes if the highest one is known
After the node inclusion, we definitely know the granted security classes. However that information was only used for endpoints and the root was still queried for all security classes.

fixes: #3881