zwave-js/zwave-js #3885: Missing device configuration: Aeotec MultiSensor 7

luzik, 2021-12-11:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Aeotec MultiSensor 7

### Manufacturer ID

0x0371

### Product Type

0x0002

### Product ID

0x0018

### Firmware Version

1.1

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4061

jmgiaever, 2021-12-11:
Duplicate: https://github.com/zwave-js/node-zwave-js/issues/3853

AlCalzone, 2021-12-11:
Duplicate of #3802

closed 2021-12-11 by AlCalzone