zwave-js/zwave-js #3887: Missing device configuration: <enter manufacturer and device name>

PETEIOT, 2021-12-12:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Yale yrd256

### Manufacturer ID

0x0129

### Product Type

0x8002

### Product ID

0xA570

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3882?selectedFrequencyId=2

### Do you have a manual?

[image]
[image]
[image]

marcus-j-davies, 2021-12-12:
The Identifies you have provided are already supported.
https://devices.zwave-js.io/?jumpTo=0x0129:0x8002:0xa570:0.0 - (YRD256-ZW3 )

However, the one in your screenshot: 0x0131:0x003:0x1111 is not found.

What ID's are we looking at here in your report - My head hurts!

blhoward2, 2021-12-12:
Please post the actual dump, not a screenshot. We need to be able to scroll through it. What node is at issue?

PETEIOT, 2021-12-14:
When I try to upload the zwake_jsdump.json I get an error saying the file type is not supported. So how do I include it?

blhoward2, 2021-12-14:
Just zip it and upload the zip file.

PETEIOT, 2021-12-14:
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/7708699/zwave_js_dump.zip)

AlCalzone, 2021-12-14:
Node 14 in your dump has almost no information and the bit that's there seems like zwave-js wasn't able to determine which security classes it uses. It is also missing the identification part, which is probably why its not being identified properly. To better diagnose this, please do the following:

1. Make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly`
2. Exclude the node, and re-include it. Leave the log active
3. Wait until it is interviewed completely
4. Send me the logfile to the e-mail in my profile, referencing this issue

PETEIOT, 2021-12-16:
So I did it and during inclusion this time it pulled the correct attribute's not sure why I tried this 3 times before but it's working great not thanks anyway.

closed 2021-12-16 by PETEIOT