zwave-js/zwave-js #3892: Distinguish between fast and slow device classes when doing optimistic value updates

AlCalzone, 2021-12-14:
We should distinguish between devices that do fast and device that do slow transitions and only do optimistic updates for fast ones.
This way, UIs could give immediate feedback about the state of lights but it would still be possible to wait for the slower ones like roller shutters.

Fast would be:
- Lights without a specified transition time
- Lights with a short transition time (< 5s for example)
- Binary switches, etc.

Slow would be:
- Motor control
- Window Covering

MaelDieudonne, 2024-04-14:
Sorry do I need to tag you @AlCalzone? Still hoping that one day this change could be implemented. Would be a game changer for me. Thanks

jankornblum, 2025-07-17:
In my case, I am using disabled optimistic updates for devices which are meant to be „fast“, bit often respond slow or not at all because of the 700 series issues (Shelly Wave 1 PM, Shelly Wave Plug etc.). So for my usecase, this feature wouldn’t make sense. Or do I misunderstand anything?

fmauNeko, 2025-07-18:
From what I understand, optimistic updates are triggered when [`shouldUpdateOptimistically`](https://github.com/zwave-js/zwave-js/blob/master/packages/zwave-js/src/lib/node/Node.ts#L637) is `true`.
Looking more into the code, I think implementing a new `SlowDeviceClasses` registry containing a list of generic/specific slow classes, then using it in the implementation of [`isSetValueOptimistic`](https://github.com/zwave-js/zwave-js/blob/master/packages/cc/src/lib/API.ts#L343) to return `false` if the device class of the endpoint is marked as slow would be a good way to go.
What do you think @AlCalzone ?

@jankornblum From what I understand that indeed would not help in your case.