zwave-js/zwave-js #3902: Missing device configuration: EVA LOGIK ZW39 Z-Wave Smart Plug (Dimmer)

vgehts, 2021-12-15:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

EVA LOGIK ZW39 Z-Wave Smart Plug (Dimmer)

### Manufacturer ID

0x0312

### Product Type

0xc000

### Product ID

0xc002

### Firmware Version

2.2 

### Is the device listed on the Z-Wave Alliance website?

Older version: https://products.z-wavealliance.org/products/2358

### Do you have a manual?

[ZW39001.pdf](https://github.com/zwave-js/node-zwave-js/files/7716711/ZW39001.pdf)

AlCalzone, 2021-12-15:
@zwave-js-bot config help wanted

blhoward2, 2021-12-29:
@zwave-js-bot import config 2358 from zwa

blhoward2, 2021-12-29:
We weren't actually missing this file, though it was misdescribed. I've corrected that in the PR.

PR #3980 (2021-12-29): fix(config): correct EVA LOGIK ZW39 device files
fixes: #3902

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-12-29 by AlCalzone