zwave-js/zwave-js #3904: [question]  NVM backup error - Error while calling api backupNVMRaw: Could not turn off the Z-Wave radio before creating NVM backup!

Swallowtail23, 2021-12-15:
Aeotec Z-Stick Gen5+, pass-through USB (KVM/virtio) to Home Assistant - works fine for control. 

When I try and run an NVM backup in the ZWave JS to MQTT Control Panel, I get the following error:

```Error while calling api backupNVMRaw: Could not turn off the Z-Wave radio before creating NVM backup!```

How do I work around this to be able to backup the NVM?

AlCalzone, 2021-12-15:
Start by making a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) of the process, loglevel `debug` or `silly` and attach it here as a file.

Swallowtail23, 2021-12-15:
[zwavejs2mqtt_2021-12-15.log](https://github.com/zwave-js/node-zwave-js/files/7717955/zwavejs2mqtt_2021-12-15.log)

AlCalzone, 2021-12-15:
That's not a driver log

AlCalzone, 2021-12-15:
That's why I always link to the docs for the second 😉

AlCalzone, 2021-12-15:
Ok the stick just returns that it didn't disable the radio... Maybe it is in some weird state.
Try re-plugging it while zwavejs2mqtt is shut down.

Swallowtail23, 2021-12-15:
I stopped ZWJS2MQTT, and pulled the USB. After replugging it I had no USB stick visible in HA; although lsusb showed 0658:0200 visible in both KVM/virtio host and in HA (terminal), it was not at /dev/serial/by-id or TTYUSB. Supervisor restart... still nothing. Reboot HA from terminal and no stick in lsusb at all, so obviously nothing available in ZWJS2MQTT. 

Guest shutdown from KVM/virtio host command line, full guest poweroff and restart, and the stick reappeared finally. 

After doing all of that, finally it allowed me to do an NVM backup. 

Why the need for mucking around with reboots?

AlCalzone, 2021-12-16:
> Why the need for mucking around with reboots?

I'm not sure, but these sticks tend to get stuck in a weird state when the application shuts down. We started using the soft reset command a while back to force a stick restart on driver startup, but that causes a whole heap of issues. The 500 series sticks all disconnect and re-connect from USB, which can also lead to the "missing" stick you mentioned. Some sticks even require to be physically replugged after the soft reset, which is why they are blacklisted from that functionality.

HA disables the soft reset by default on VMs, because it would require users to set up the correct (host specific) udev rules and sometimes even special VM configuration.
So... yeah, you can set it all up in a way that the stick is soft reset automatically and comes back, but its not feasibly to document every possible scenario.

Swallowtail23, 2021-12-18:
After I do *one* NVM backup, this error reoccurs. I need to then do a hardware shutdown/restart of the VM and then it comes good.

closed 2022-01-05 by zwave-js-assistant[bot]