zwave-js/zwave-js #3906: META-Issue: Problems with 700 series (healing, delays, neighbors, ...)

AlCalzone, 2021-12-15:
It seems that **700-series sticks** (firmware <= 7.16.3) have some problems which mostly seems to appear on networks that:

* are relatively busy (lots of unsolicited reports like motion sensor, power meters)
* are large and/or
* contain some battery-powered devices

It seems that when lots of reports reach the controller in a short time, the 700-series sticks are not able to send any message.
To me, it looks like the stick is somehow blocked and simply doesn’t send anything, not even the protocol level acknowledgements for receiving the messages, causing end nodes to repeat their messages over and over, making the situation even worse.

We're working with Silabs to get to the bottom of this. Please don't open any further issues.

---

We believe the following symptoms are all caused by this:
- Huge delays (up to 60 seconds) when sending messages
- Floods of incoming reports that are transmitted over and over
- Incorrect neighbor information (e.g. controller doesn't list some/any nodes as neighbors, etc.)
- Failure to heal the network or individual nodes, especially in busy situations

darkbasic, 2021-12-15:
> Failure to heal the network or individual nodes, especially in busy situations

If **any** message gets sent in the network while healing it will fail.
If the healed node is not in direct reach from the controller, it will **always** fail.
Right now I gave up coding anything zwave because of this, it makes it completely useless and unusable.I hope they can manage to find a solution soon, VERY soon.

andras-tim, 2021-12-16:
And what version do you recommend? As I see in the [release notes](https://docs.silabs.com/search#t=Software%20Docs&sort=date%20descending&f:@common_allowed_content_type=[Release%20Notes]&f:@common_allowed_language=[English]&f:@common_product_hierarchy=[Wireless,Z-Wave,Z-Wave%20700%20Modules]), the 7.16 line ended with **7.16.3**, but here's coming the **7.17.0**. Is it good?

AlCalzone, 2021-12-16:
I don't have any experience with 7.17 but judging by the release notes it doesn't change anything relevant.

If you're having these problems I fear the only "solution" right now is to use a 500 series stick.

sstarcher, 2021-12-16:
Does any correlation exist with the AeoTec firmware versions? Or are they entirely separate?

AlCalzone, 2021-12-16:
All 700 series sticks (even from different manufacturers) use the same official Silabs firmware with the same versions.

darkbasic, 2021-12-17:
> If you're having these problems I fear the only "solution" right now is to use a 500 series stick.

I would be glad to do so, but it's not even really a solution because the NVM format changed between 500 and 700 series and we can't convert it back because it's undocumented.

AlCalzone, 2021-12-17:
I'm working on a solution.

kars85, 2021-12-17:
@andras-tim - Simplicity Studio had updates for me last night, but only pushed 7.17.0.0 for the Z-Wave Z-IP Gateway SDK, not the Z-Wave SDK under the Gecko Platform. That is still at 7.16.3.0.  

[image]

darkbasic, 2021-12-17:
AFAIK Silabs still has to find the root cause of the issue, thus 7.17 won't fix it anyway.

Danny2100, 2021-12-17:
If I just received a zooz 7 stick what are the chances it has this firmware? How do you even check what firmware the stick has before I use it just to be sure and can you downgrade to a firmware that doesn't have this issue so I can get setup using the stick? Also are there major enhancements done through firmware updates as I have a nortek and never updated it.

sstarcher, 2021-12-17:
@Danny2100 my understanding is this firmware does not fix the problem.  Honestly unless you want to fight through tons of issues I would recommend returning the stick.  If not you will be waiting until a  non-buggy firmware gets released.

darkbasic, 2021-12-17:
Every single series 700 firmware suffers from this issue. If you can, return it and get a 500 series one.

justindthomas, 2021-12-17:
> Every single 700 series firmware suffers from this issue. If you can, return it and get a 500 series one.

Fwiw, I've been running a Z-Stick 7 for a few months on a Z-Wave network with 69 nodes. I've definitely experienced the issues noted (most visibly, the healing). Recently, I've found that some of my nodes will not respond to their automations and I see that they've been marked unavailable. They come back online as soon as I manually activate them. I suspect that that's related.

But I'll stick with it even though I also have a 5+ version sitting around. At least now it's a known issue that's getting attention. Up until now, I just figured ZwaveJS was still a bit unstable.

Danny2100, 2021-12-17:
I thought it was just a specific firmware version that had this issue? It's
all of them?

I am still using the nortek and was about to move to the zooz stick I have
29 devices the majority are mains powered and I was about to add 10 zooz
switches too.

sstarcher, 2021-12-17:
It's all of them.  I have a Aeotec Gen7 that I have had for 6 months and it has always had these issues.  20ish devices on main power and another dozen on battery and I have tons of issues.  Never a issue with manual touches, but always a issue with automation.

Danny2100, 2021-12-17:
Wow 6 months is a long time for a firmware bug to still exist.

andras-tim, 2021-12-18:
> @andras-tim - Simplicity Studio had updates for me last night, but only pushed 7.17.0.0 for the Z-Wave Z-IP Gateway SDK, not the Z-Wave SDK under the Gecko Platform. That is still at 7.16.3.0.
> 
> [image]

Yap, under Gecko 3.2, I had 7.16.3 only... and I couldn't find any firmware file in the Gecko 4 too... :frowning_face:

justindthomas, 2021-12-19:
I may walk back my comment on not reverting to the Z-Stick 5+; it seems like the problem with nodes being marked unavailable is much worse than I had thought.

I don't know if it's due to recent changes in the software or if I'm just now really noticing what is going on, but a few of my nodes just will not stay online for more than a few hours. Interestingly, it's a 700 series switch (Zooz Scene Controller) and 2 500 series Inovellis that are beyond that switch (probably routing through it) that are the most affected.

Pinging them from zwavejs2mqtt brings them right back online, but I have to be constantly on top of it to catch them go offline.

Danny2100, 2021-12-19:
> I may walk back my comment on not reverting to the Z-Stick 5+; it seems like the problem with nodes being marked unavailable is much worse than I had thought.
>
> I don't know if it's due to recent [...]

That's scary. I am about to setup 10 zooz scene controllers. Although I am using nortek. I am wondering if I should test it out on a few first now vs an a complete roll out before testing.

AlCalzone, 2021-12-19:
> Although I am using nortek

That's a 500-series controller though, right? Those are not affected to my knowledge.

AlCalzone, 2021-12-19:
If anyone wants to take a little risk and try the migration back to the 500 series (requires Node.js and npm to be installed):
1. make an NVM backup of the current 700 series stick
2. make an NVM backup of the target 500 series stick
3. execute the `convert` command here: https://github.com/zwave-js/node-zwave-js/tree/nvmedit/packages/nvmedit#convert-one-nvm-to-be-compatible-with-another-one, but replace `@zwave-js/nvmedit` with `@zwave-js/nvmedit@8.9.0-beta.4-pr-3789-c208add`
4. Restore the resulting NVM file on the target stick

❗❗❗ **Disclaimer:** I have quite a few unit tests ensuring the correct format but I haven't actually tested restoring the resulting files on a stick, so I'm not 100% certain if it will work (both the backup and the stick 😬). If it doesn't work, you should be able to hard-reset the target stick to get it working again, but I'm not making guarantees here. **Try at your own risk!**

Danny2100, 2021-12-20:
> > Although I am using nortek
> 
> That's a 500-series controller though, right? Those are not affected to my knowledge.

Yes it is. I was just making sure as he said his aeotec 5 stick was having issues too

justindthomas, 2021-12-20:
> Yes it is. I was just making sure as he said his aeotec 5 stick was having issues too

No I didn't - I said that it's 700 devices (the Zooz Scene Controllers) and 500 devices that are connected through 700 devices that seem to be having the most issues. But that may be a red herring.

Either way, they're all managed by a 700 controller (Aeotec Z-Stick 7) and I am now considering switching to a 500 (Aeotec Z-Stick 5+) that I've had sitting around for a while but haven't used in months. I'll probably give the conversion script a whirl in a bit.

Danny2100, 2021-12-20:
> No I didn't - I said that it's 700 devices (the Zooz Scene Controllers) and 500 devices that are connected through 700 devices that seem to be having the most issues. But that may be a red herring. [...]

I completely missed the "not reverting" and read it as "reverting to Z-Stick 5+" I misread that whoops.

Danny2100, 2021-12-20:
> If anyone wants to take a little risk and try the migration back to the 500 series (requires Node.js and npm to be installed):
>
> 1. make an NVM backup of the current 700 series stick
> 2. make an NVM [...]

I see it can also be used for migrating from 500 to 700 whenever this bug is fixed. Can this tool run in the home assistant container similar to the ZigBee migration tool? I remember you talking about this in the state of the home. However I think you said it was going to be released in the future and I assume this may have been rushed out as a quick fix but requires further development to become stable? Is there a way to backup the 500 series stick in case the migration to the 700 series fails?

justindthomas, 2021-12-20:
@AlCalzone I gave that a try, but I can't write the resulting output back to the 500 stick. I get an error about it being the wrong size.

~~~~~
$ npx @zwave-js/nvmedit@8.9.0-beta.4-pr-3789-c208add convert --source ./NVM_2021-12-20_7.bin --target ./NVM_2021-12-20_5.bin --out ./NVM_converted.bin

$ ls -la NVM*
-rw-r--r-- 1 justin justin 262144 Dec 19 17:15 NVM_2021-12-20_5.bin
-rw-r--r-- 1 justin justin  49152 Dec 19 17:15 NVM_2021-12-20_7.bin
-rw-rw-r-- 1 justin justin  14375 Dec 19 17:16 NVM_converted.bin
~~~~~

I'm not sure why the NVM backup for the 5 is large - there are no nodes listed when I plug it in.

AlCalzone, 2021-12-20:
> Can this tool run in the home assistant container similar to the ZigBee migration tool? I remember you talking about this in the state of the home.

It is built in a way that the underlying functionality can also be used by the driver directly. So down the line, applications will just be able to call the method with the old NVM backup buffer.

> However I think you said it was going to be released in the future and I assume this may have been rushed out as a quick fix but requires further development to become stable?

I've actually been working on it for some weeks and it only now got far enough to be released. 

> Is there a way to backup the 500 series stick in case the migration to the 700 series fails?

zwavejs2mqtt can do that. For now you need backups of both sticks anyways.

@justindthomas 
> I get an error about it being the wrong size.

Ahh, damn I forgot to change that. The conversion utility only outputs the part of the 500-series NVM that was changed. They are 256 kB in size, but only the first ~14 kB are what interests us.

Danny2100, 2021-12-20:
> > Can this tool run in the home assistant container similar to the ZigBee migration tool? I remember you talking about this in the state of the home.
> 
> It is built in a way that the underlying functionality can also be used by the driver directly. So down the line, applications will just be able to call the method with the old NVM backup buffer.
> 
> > However I think you said it was going to be released in the future and I assume this may have been rushed out as a quick fix but requires further development to become stable?
> 
> I've actually been working on it for some weeks and it only now got far enough to be released.
> 
> > Is there a way to backup the 500 series stick in case the migration to the 700 series fails?
> 
> zwavejs2mqtt can do that. For now you need backups of both sticks anyways.
> 
> @justindthomas
> 
> > I get an error about it being the wrong size.
> 
> Ahh, damn I forgot to change that. The conversion utility only outputs the part of the 500-series NVM that is relevant. They are 256 kB in size, but only the first ~14 kB are what interests us.

Would the migration tool work on a Nortek HUSBZB-1 that has Z-Wave 6.51.06 sdk?

AlCalzone, 2021-12-20:
No, I haven't reverse-engineered the NVM format of pre-6.61 SDKs yet.

Danny2100, 2021-12-20:
> No, I haven't reverse-engineered the NVM format of pre-6.61 SDKs yet.

How would we get notified of improvements to the tool? I will definitely be interested in trying it once this bug with 700 series sticks is fixed and you're able to reverse engineer pre-6.61 sdks.

AlCalzone, 2021-12-20:
The changelog or release notes

dearekaelle, 2021-12-20:
@AlCalzone I think I was too quick in jumping into 7.17.0 firmware on my z-stick 7.. cannot convert the NVM

Error: Could not parse source NVM - invalid format!
    at migrateNVM (/root/.npm/_npx/0e291f38aba7805a/node_modules/@zwave-js/nvmedit/build/convert.js:746:19)
    at Object.handler (/root/.npm/_npx/0e291f38aba7805a/node_modules/@zwave-js/nvmedit/build/cli.js:168:45)

AlCalzone, 2021-12-20:
Can you send me the backup to the email in my profile? Might just be a small detail.

justindthomas, 2021-12-20:
@AlCalzone looks like it's still just writing out the first 15k.

~~~~~
╭─justin@pop-os ~/node-zwave-js ‹nvmedit*› 
╰─$ npx @zwave-js/nvmedit@8.9.0-beta.4-pr-3789-f60a13a convert --source ./NVM_2021-12-20_7.bin --target ./NVM_2021-12-20_5.bin --out ./NVM_converted.bin
npx: installed 69 in 2.039s
Converted NVM written to ./NVM_converted.bin
╭─justin@pop-os ~/node-zwave-js ‹nvmedit*› 
╰─$ ls -la NVM*
-rw-r--r-- 1 justin justin 262144 Dec 19 17:15 NVM_2021-12-20_5.bin
-rw-r--r-- 1 justin justin  49152 Dec 19 17:15 NVM_2021-12-20_7.bin
-rw-rw-r-- 1 justin justin  14375 Dec 20 09:06 NVM_converted.bin
~~~~~

AlCalzone, 2021-12-20:
Yeah but you can use that to restore the backup now.

justindthomas, 2021-12-20:
I get the same error with the new file.

[image]

AlCalzone, 2021-12-20:
Ok can you send me that too?

justindthomas, 2021-12-20:
Sure thing; do you want all 3 or just the output?

AlCalzone, 2021-12-20:
Just the output

AlCalzone, 2021-12-20:
@justindthomas next try please, manual was updated

justindthomas, 2021-12-20:
Same result - "The given data does not match the NVM size".

~~~~~
╭─justin@pop-os ~/node-zwave-js ‹nvmedit*› 
╰─$ npx @zwave-js/nvmedit@8.9.0-beta.4-pr-3789-97b47a4 convert --vv --source ./NVM_2021-12-20_7.bin --target ./NVM_2021-12-20_5.bin --out ./NVM_converted.bin
npx: installed 69 in 2.808s
Converted NVM written to ./NVM_converted.bin
╭─justin@pop-os ~/node-zwave-js ‹nvmedit*› 
╰─$ ls -la NVM*
-rw-r--r-- 1 justin justin 262144 Dec 20 11:26 NVM_2021-12-20_5.bin
-rw-r--r-- 1 justin justin  49152 Dec 19 17:15 NVM_2021-12-20_7.bin
-rw-rw-r-- 1 justin justin  14375 Dec 20 11:27 NVM_converted.bin
~~~~~

I also tried taking another backup of the 5+ (in case the first was corrupted or something) and deleting the `NVM_converted.bin` file before running the conversion, but no change in result.

justindthomas, 2021-12-20:
Here are the log messages leading up to that, in case they're at all helpful.

~~~~~
2021-12-20 11:26:35.457 INFO ZWAVE: Success zwave api call backupNVMRaw {
  data: <Buffer 38 5f 2f d0 46 52 45 45 db c4 7b e8 ff ff ff ff 00 00 00 00 00 54 a5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 262094 more bytes>,
  fileName: 'NVM_2021-12-20'
}
2021-12-20 11:28:16.655 INFO ZWAVE: Calling api restoreNVMRaw with args: [
  <Buffer 38 26 2f d0 00 00 00 00 c9 5f d3 c4 ff ff ff ff 00 00 00 00 01 54 a5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 14325 more bytes>,
  [length]: 1
]
2021-12-20 11:28:16.713 INFO ZWAVE: The given data does not match the NVM size - cannot restore! (ZW0322) restoreNVMRaw undefined
~~~~~

AlCalzone, 2021-12-20:
Ohh - your zwavejs2mqtt instance is still running the official version which doesn't include the fix. Not sure how you're running z2m. Since you obviously have Node installed, you can do this:
1. clone https://github.com/zwave-js/zwavejs2mqtt
2. in the cloned repo, run `yarn up zwave-js@8.9.0-beta.4-pr-3789-97b47a4`
3. start with `yarn dev:server`
4. open z2m in the browser (http://localhost:8091)
5. restore there.

AlCalzone, 2021-12-20:
@dearekaelle The latest test release should work for 7.17.0 too.

justindthomas, 2021-12-20:
Got it. Yeah, I'm running the conversion on my laptop and attempting the restore on my Home Assistant server.

I'll try doing it all locally on my laptop per your instructions.

johanschelin, 2021-12-20:
Following this to evaluate a possible switch for me too from Aeotec 7 to UZB-stick as I have the exact same issues.

justindthomas, 2021-12-20:
Sorry - I know this is remedial, but it doesn't look like there's a `dev:server` script:

 ~~~~~
yarn dev:server 
Usage Error: Couldn't find a script named "dev:server".

$ yarn run [--inspect] [--inspect-brk] [-T,--top-level] [-B,--binaries-only] <scriptName> ...
~~~~~

The `yarn up` command ran fine though.

AlCalzone, 2021-12-20:
Uhh, in which directory are you doing that? In the cloned zwavejs2mqtt repo?

justindthomas, 2021-12-20:
Oh, sorry - sheesh. I was using this repo  `node-zwave-js`.

AlCalzone, 2021-12-20:
> I was using this repo node-zwave-js.

Make sure to revert the changes and run `yarn` there again, if you plan to work with that repo. Otherwise it doesn't matter.

dearekaelle, 2021-12-20:
> @dearekaelle The latest test release should work for 7.17.0 too.

Many thanks, it did work now to create output file.
-rw-r--r--  1 darkell  staff  14375 Dec 20 21:53 NVM_out.bin

However restoring it, fails the same way as experienced by @justindthomas. 

I am currently running zwavejs2mqtt: 6.1.1; zwave-js: 8.9.0-beta.3

AlCalzone, 2021-12-20:
Yeah, you need to make sure to run the test version of the driver to be able to update for now --> https://github.com/zwave-js/node-zwave-js/issues/3906#issuecomment-998231525.

justindthomas, 2021-12-20:
It seems like it may have worked. The restore seemed to stick at 5%, but I think that might just be a miscalculation. The log messages seemed to indicate it worked and was complete.

When I plugged the 5+ in to HA in place of my Z-Stick 7, I see the 69 entities and they link up to the names and locations already in place. But the device data is all missing and they're spinning on "ProtocolInfo". As time passes, they're all reporting in as "dead". Do i just need to wait for that process to complete and then do manual interviews on all of them?

AlCalzone, 2021-12-20:
> But the device data is all missing and they're spinning on "ProtocolInfo". As time passes, they're all reporting in as "dead"

I'd be interested in seeing a driver log of that startup. Since the devices are in memory, so should the home ID, but maybe that wasn't transferred correctly.

justindthomas, 2021-12-20:
Those should both be in your inbox now.

AlCalzone, 2021-12-20:
Ok, so there is a difference in RF config which might explain the connectivity problem.

AlCalzone, 2021-12-20:
Last round for tonight, this time use `8.9.0-beta.4-pr-3789-6b3eaa1` for the version of the migration tool. You should be able to restore using what you already have.

justindthomas, 2021-12-20:
Awesome, thanks. I'm out and about but will give it a go when I get back to my office. Have a good night!

justindthomas, 2021-12-21:
No luck. The behavior looks unchanged.
[image]

~~~~~
2021-12-20 16:13:55.345 INFO ZWAVE: Connecting to /dev/ttyACM0
2021-12-20 16:13:55.350 INFO ZWAVE: Zwavejs usage statistics ENABLED
2021-12-20 16:13:55.351 INFO APP: POST /api/settings 200 38.685 ms - 1002
2021-12-20 16:13:55.499 INFO APP: GET /api/auth-enabled 304 0.779 ms - -
2021-12-20 16:13:58.736 INFO ZWAVE: Zwave driver is ready
2021-12-20 16:13:58.736 INFO ZWAVE: Controller status: Driver ready
2021-12-20 16:13:58.742 DEBUG ZWAVE: Binding to node 1 events
2021-12-20 16:13:58.742 DEBUG ZWAVE: Node 1 has been added to nodes array
2021-12-20 16:13:58.743 DEBUG ZWAVE: Binding to node 2 events
2021-12-20 16:13:58.743 DEBUG ZWAVE: Node 2 has been added to nodes array
<snip>
2021-12-20 16:13:58.761 INFO ZWAVE: Scanning network with homeid: 0x0
2021-12-20 16:13:58.762 INFO ZWAVE: Node 1: interview started
2021-12-20 16:13:58.778 INFO ZWAVE: Node 1: interview stage PROTOCOLINFO completed
2021-12-20 16:13:58.788 INFO ZWAVE: Node 1: interview stage OVERWRITECONFIG completed
2021-12-20 16:13:58.788 INFO ZWAVE: Node 1: interview stage COMPLETE completed
2021-12-20 16:13:58.790 INFO ZWAVE: Node 1 ready: AEON Labs - ZW090 (Z‐Stick Gen5 USB Controller)
2021-12-20 16:13:58.790 INFO ZWAVE: Node 1: interview COMPLETED, all values are updated
2021-12-20 16:13:58.791 INFO ZWAVE: Node 1 is alive
2021-12-20 16:13:58.791 INFO ZWAVE: Node 2: interview started
2021-12-20 16:13:58.793 INFO ZWAVE: Node 3: interview started
2021-12-20 16:13:58.794 INFO ZWAVE: Node 4: interview started
<snip>
2021-12-20 16:13:58.891 INFO ZWAVE: Node 2: interview stage PROTOCOLINFO completed
2021-12-20 16:13:58.903 INFO ZWAVE: Node 3: interview stage PROTOCOLINFO completed
2021-12-20 16:13:58.912 INFO ZWAVE: Node 4: interview stage PROTOCOLINFO completed
2021-12-20 16:13:58.921 INFO ZWAVE: Node 5: interview stage PROTOCOLINFO completed
<snip>
2021-12-20 16:14:05.434 INFO ZWAVE: Node 2 is dead
2021-12-20 16:14:11.986 INFO ZWAVE: Node 3 is dead
2021-12-20 16:14:18.865 INFO ZWAVE: Node 4 is dead
2021-12-20 16:14:24.672 INFO ZWAVE: Node 5 is dead
2021-12-20 16:14:31.997 INFO ZWAVE: Node 6 is dead
~~~~~

AlCalzone, 2021-12-21:
Ok can you send me another copy of the original 500 series backup and the newly restored one? I think I might have mixed up the home IDs. The 500 series one has two in the NVM and it wasn't clear which is which.

[465 more replies]

closed 2022-04-28 by AlCalzone