zwave-js/zwave-js #3916: [bug] Illuminance sensor found at pairing, never updated and lost after re-interview

Moffus, 2021-12-16:
### Version

**Checklist:**

- [X] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
Problem not depends on HA, every interface, including conrol panel and MQTT are affrcted
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 6.1.0
Z-Wave JS version: 8.9.0-beta.0

### Describe the bug
At pairing: Illuminance sensor detected well, value transferred normally. One time
At normal flow: Value never updated

[image]
Actually, 0 lux now, window covers are closed.

Debug out quote:
```
"values": [
    {
      "id": "50-49-0-Illuminance",
      "nodeId": 50,
      "commandClass": 49,
      "commandClassName": "Multilevel Sensor",
      "endpoint": 0,
      "property": "Illuminance",
      "propertyName": "Illuminance",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Illuminance",
      "ccSpecific": {
        "sensorType": 3,
        "scale": 1
      },
      "stateless": false,
      "unit": "Lux",
      "list": false,
      "value": 91,
      "lastUpdate": 1639585094555,
      "newValue": 91
    },
```
Sensor type 3 - it Ok.
Scale 1 = Lux. It Ok too.

Quote from zwavejs.log:
```
021-12-16T09:26:36.745Z CNTRLR » [Node 050] querying Illuminance sensor reading...
2021-12-16T09:26:36.745Z CNTRLR   [Node 050] Preferred scale 1 not supported for sensor type 3, using the first
                                  supported scale
2021-12-16T09:26:36.750Z SERIAL » 0x010b0013320431040300258341                                        (13 bytes)
2021-12-16T09:26:36.751Z DRIVER » [Node 050] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      131
                                  └─[MultilevelSensorCCGet]
                                      sensor type: Illuminance
                                      scale:       Percentage value
```
WTF? Scale 1 (Lux) only a normal scale fo sensor type 3!
```
2021-12-16T09:26:36.753Z SERIAL « [ACK]                                                                   (0x06)
2021-12-16T09:26:36.759Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-12-16T09:26:36.760Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T09:26:36.760Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-12-16T09:26:37.027Z SERIAL « 0x011800138300001b01c9b07f7f7f010103300000000101000058              (26 bytes)
2021-12-16T09:26:37.028Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T09:26:37.028Z DRIVER « [REQ] [SendData]
                                    callback id:            131
                                    transmit status:        OK, took 270 ms
                                    repeater node IDs:      48
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 9.6 kbit/s
                                    ACK RSSI:               -55 dBm
                                    ACK RSSI on repeaters:  -80 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2021-12-16T09:26:37.160Z SERIAL « 0x010e00040032063105030a0000c90035                                  (16 bytes)
2021-12-16T09:26:37.160Z DRIVER   Dropping message with invalid payload
2021-12-16T09:26:37.161Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport] [INVALID]
                                      error: Unsupported sensor type Lux or corrupted data
2021-12-16T09:26:37.161Z SERIAL » [ACK]                                                                   (0x06)
```
Yes, Lux valye sent, but logic awaiting a percentage data.

Possible sources of problem:
1. Variable from Humidity sensor analyzed, when sensor switched from lux to percentage in logic (array index shifted). Only humidity sensor have percentage scale.
2. Something worong with illuminance sensor

Node configuration attached.
[zwavenodedebug.txt](https://github.com/zwave-js/zwavejs2mqtt/files/7726080/zwavenodedebug.txt)

### To Reproduce
Go to control panel, multilevel section and click "refresh".

AlCalzone, 2021-12-16:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly`. Re-interview the device, then attach the log here as a file.

Moffus, 2021-12-16:
[zwavejs_2021-12-16.log](https://github.com/zwave-js/node-zwave-js/files/7728433/zwavejs_2021-12-16.log)

Moffus, 2021-12-16:
After re-interview illuminance sensor not found in multilevel sensors list
[debuginfo.txt](https://github.com/zwave-js/node-zwave-js/files/7728453/debuginfo.txt)
.

AlCalzone, 2021-12-16:
That's a device bug. The temperature scales seem to be encoded correctly, but for the Illuminance sensor, the supported scale report has the marked byte set to `1`:
[image]

The manufacturer probably meant to encode that scale `1` (Lux) is supported, but since this is a bitmask, the byte should have been set to `2` instead. The first scale is `0` (%).

---

If you scroll down [here](https://www.heltun.com/support), you'll find an upgrade for your thermostat to FW 1.7
Not sure if they fixed this, but it's worth a try.

closed 2021-12-20 by zwave-js-assistant[bot]