zwave-js/zwave-js #3917: Security S2: Investigate better recovery from message collision

AlCalzone, 2021-12-16:
```
2021-12-16T12:07:54.812Z DRIVER » [Node 027] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      130
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 142
                                    └─[ConfigurationCCNameGet]
                                        parameter #: 4
2021-12-16T12:07:54.813Z SERIAL « 0x012000a800011b179f032900bb213f7322d99a9acb20ecd6a315dbdbaaeb5400c (34 bytes)
                                  b5b
2021-12-16T12:07:54.813Z DRIVER   Dropping message with invalid payload
2021-12-16T12:07:54.814Z DRIVER « [Node 027] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -53 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command
2021-12-16T12:07:54.814Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.822Z SERIAL « [ACK]                                                                   (0x06)
2021-12-16T12:07:54.823Z SERIAL « 0x010401a90152                                                       (6 bytes)
2021-12-16T12:07:54.823Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.823Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2021-12-16T12:07:54.856Z SERIAL « 0x011800a98200000300cb7f7f7f7f000003000000000302000006              (26 bytes)
2021-12-16T12:07:54.857Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.858Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            130
                                    transmit status:        OK, took 30 ms
                                    routing attempts:       2
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -53 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2021-12-16T12:07:54.872Z SERIAL « 0x012200a800011b199f032a004562745fa429bb0c801e8dccfeedac3395519d2ed (36 bytes)
                                  100cbc8
2021-12-16T12:07:54.875Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.876Z DRIVER « [Node 027] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -53 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2021-12-16T12:07:54.877Z CNTRLR » [Node 027] Message authentication failed, cannot decode command. Requesting a 
                                  nonce...
2021-12-16T12:07:54.883Z SERIAL » 0x012000a9011b149f028f010e101a63b17a2f7f472340e38331e0bc05000000008 (34 bytes)
                                  338
2021-12-16T12:07:54.884Z DRIVER » [Node 027] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x05
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      131
                                  └─[Security2CCNonceReport]
                                      sequence number:  143
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x0e101a63b17a2f7f472340e38331e0bc
2021-12-16T12:07:54.886Z SERIAL « 0x011d00a800011b149f022b016ce9ccda2699e46f80d5660be23c930000cbea    (31 bytes)
2021-12-16T12:07:54.887Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.887Z DRIVER « [Node 027] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -53 dBm
                                  └─[Security2CCNonceReport]
                                      sequence number:  43
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x6ce9ccda2699e46f80d5660be23c9300
2021-12-16T12:07:54.888Z CNTRLR   [Node 027] failed to decode the message, retrying with SPAN extension...
2021-12-16T12:07:54.895Z SERIAL « [ACK]                                                                   (0x06)
2021-12-16T12:07:54.896Z SERIAL « 0x010401a90152                                                       (6 bytes)
2021-12-16T12:07:54.896Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.897Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2021-12-16T12:07:54.908Z SERIAL « 0x011800a98300000100cb7f7f7f7f000003000000000301000006              (26 bytes)
[... 46 lines cut ...]
                                    TX channel no.:         0
2021-12-16T12:07:54.947Z SERIAL » 0x012000a9011b149f0291014454f9fba1709b9548252c960e143e5005000000008 (34 bytes)
                                  59b
2021-12-16T12:07:54.948Z DRIVER » [Node 027] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x05
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      133
                                  └─[Security2CCNonceReport]
                                      sequence number:  145
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x4454f9fba1709b9548252c960e143e50
2021-12-16T12:07:54.952Z SERIAL « 0x011d00a800011b149f022d018b6c83c73a3d843a46477389d77a014300cb30    (31 bytes)
2021-12-16T12:07:54.953Z SERIAL » [ACK]                                                                   (0x06)
2021-12-16T12:07:54.954Z DRIVER « [Node 027] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -53 dBm
                                  └─[Security2CCNonceReport]
                                      sequence number:  45
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x8b6c83c73a3d843a46477389d77a0143
2021-12-16T12:07:54.955Z CNTRLR   [Node 027] failed to decode the message after re-transmission with SPAN extens
                                  ion, dropping the message.
```

LordMike, 2021-12-25:
Hey - I just re-included a device to get S2 security (yay), and noticed and error with this log snippet. I've included the full log as well. Let me know if I should test anything - it seems reproducible (toggle some switches and the error happens)... It doesn't seem hindering for operations though, but I'll stop upgrading to S2 for now :).

```
2021-12-25T13:10:14.931Z SERIAL » [ACK]                                                                   (0x06)
2021-12-25T13:10:14.932Z DRIVER « [REQ] [SendData]
                                    callback id:     147
                                    transmit status: OK
2021-12-25T13:10:14.939Z SERIAL » 0x013000134b299f031c0112417b02bc12f472a5bdc0be589cb672b268f60c097cc (50 bytes)
                                  9d51ff7ff08782c22b89414f07025259457
2021-12-25T13:10:14.940Z DRIVER » [Node 075] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      148
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 28
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x7b02bc12f472a5bdc0be589cb672b268
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 5
                                      └─[SupervisionCCGet]
                                        │ session id:      1
                                        │ request updates: true
                                        └─[MultilevelSwitchCCSet]
                                            target value: 99
2021-12-25T13:10:14.945Z SERIAL « [ACK]                                                                   (0x06)
2021-12-25T13:10:14.952Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-12-25T13:10:14.952Z SERIAL » [ACK]                                                                   (0x06)
2021-12-25T13:10:14.953Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-12-25T13:10:15.134Z SERIAL « 0x01070013940000136c                                                 (9 bytes)
2021-12-25T13:10:15.134Z SERIAL » [ACK]                                                                   (0x06)
2021-12-25T13:10:15.135Z DRIVER « [REQ] [SendData]
                                    callback id:     148
                                    transmit status: OK
2021-12-25T13:10:15.227Z SERIAL « 0x011a0004004b149f0246012c884d94d1ebea9757d632abfdb3b977c6          (28 bytes)
2021-12-25T13:10:15.228Z SERIAL » [ACK]                                                                   (0x06)
2021-12-25T13:10:15.229Z DRIVER « [Node 075] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceReport]
                                      sequence number:  70
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x2c884d94d1ebea9757d632abfdb3b977
2021-12-25T13:10:15.231Z CNTRLR   [Node 075] failed to decode the message after re-transmission with SPAN extens
                                  ion, dropping the message.
2021-12-25T13:10:15.235Z SERIAL » 0x01140013200d600d0003330503022803000400259513                      (22 bytes)
2021-12-25T13:10:15.236Z DRIVER » [Node 032] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      149
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 3
                                    └─[ColorSwitchCCSet]
                                        Red:   40
                                        Green: 0
                                        Blue:  0
2021-12-25T13:10:15.238Z SERIAL « [ACK]                                                                   (0x06)
2021-12-25T13:10:15.245Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-12-25T13:10:15.245Z SERIAL » [ACK]                                                                   (0x06)
```

Full log: [zwavejs_2021-12-25.zip](https://github.com/zwave-js/node-zwave-js/files/7775858/zwavejs_2021-12-25.zip)

AlCalzone, 2021-12-25:
> 8.9.0-beta.0

Whatever you do, upgrade to 8.9.1 first.

LordMike, 2021-12-25:
Nvm me, 8.9.1 does not show the same issue.

AlCalzone, 2022-08-04:
Another S2 collision example:
```
2022-08-04T10:50:45.103Z DRIVER » [Node 023] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      17
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 30
                                    └─[IndicatorCCSet]
                                        values:
                                        · indicatorId: 17
                                          propertyId:  7
                                          value:       30
2022-08-04T10:50:45.108Z SERIAL « 0x010c0004000d043003ff0a7e0046                                      (14 bytes)
2022-08-04T10:50:45.109Z CNTRLR   [Node 013] [Binary Sensor] Door/Window: metadata updated          [Endpoint 0]
2022-08-04T10:50:45.110Z CNTRLR   [Node 013] [~] [Binary Sensor] Door/Window: false => true         [Endpoint 0]
2022-08-04T10:50:45.111Z SERIAL » [ACK]                                                                   (0x06)
2022-08-04T10:50:45.111Z DRIVER « [Node 013] [REQ] [ApplicationCommand]
                                  └─[BinarySensorCCReport]
                                      type:  Door/Window
                                      value: true
2022-08-04T10:50:45.114Z SERIAL « [CAN]                                                                   (0x18)
2022-08-04T10:50:45.116Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2022-08-04T10:50:45.124Z SERIAL « 0x012100040017199f03ae003cb795de909a2d392ef814ce5c6ac54bd9549d17d5c (35 bytes)
                                  f0091
2022-08-04T10:50:45.127Z SERIAL » [ACK]                                                                   (0x06)
2022-08-04T10:50:45.129Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2022-08-04T10:50:45.130Z CNTRLR » [Node 023] Message authentication failed, cannot decode command. Requesting a 
                                  nonce...
2022-08-04T10:50:45.159Z SERIAL « 0x012100040017199f03ae003cb795de909a2d392ef814ce5c6ac54bd9549d17d5c (35 bytes)
                                  d0093
2022-08-04T10:50:45.160Z DRIVER   Dropping message with invalid payload
2022-08-04T10:50:45.160Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command
2022-08-04T10:50:45.160Z SERIAL » [ACK]                                                                   (0x06)
2022-08-04T10:50:45.218Z SERIAL » 0x011a001317139f031e00d7b7a5d6f09833b77f7b4fa32bc7782511c7          (28 bytes)
2022-08-04T10:50:45.219Z DRIVER » [Node 023] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      17
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 30
                                    └─[IndicatorCCSet]
                                        values:
                                        · indicatorId: 17
                                          propertyId:  7
                                          value:       30
2022-08-04T10:50:45.222Z SERIAL « [ACK]                                                                   (0x06)
2022-08-04T10:50:45.228Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-08-04T10:50:45.229Z SERIAL » [ACK]                                                                   (0x06)
2022-08-04T10:50:45.230Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-08-04T10:50:45.268Z SERIAL « 0x011800131100000500cc7f7f7f7f01010300000000020100002c              (26 bytes)
2022-08-04T10:50:45.269Z SERIAL » [ACK]                                                                   (0x06)
2022-08-04T10:50:45.269Z DRIVER « [REQ] [SendData]
                                    callback id:            17
                                    transmit status:        OK, took 50 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -52 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2022-08-04T10:50:45.271Z SERIAL » 0x011b001317149f021f01cfb0d911d977f28f2655175218c507e505120d        (29 bytes)
2022-08-04T10:50:45.272Z DRIVER » [Node 023] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      18
                                  └─[Security2CCNonceReport]
                                      sequence number:  31
[... 101 lines cut ...]
                                    │ sequence number: 177
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0xd3cff25b7366a32a6e0e2d70c78a3c15
                                    └─[SupervisionCCGet]
                                      │ session id:      49
[... cut ...]
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 33
                                    └─[SupervisionCCReport]
                                        session id:          49
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
```

AlCalzone, 2022-08-08:
And another example:
```
2022-08-06T10:03:14.755Z DRIVER » [Node 105] [REQ] [SendData]
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 191
                                    └─[SupervisionCCReport]
                                        session id:          1
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
2022-08-06T10:03:14.757Z SERIAL « [ACK]                                                                   (0x06)
2022-08-06T10:03:14.764Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-08-06T10:03:14.765Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.766Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-08-06T10:03:14.793Z SERIAL » 0x011a001369139f03c0005ff7aa313b79296255996c691798f8259bea          (28 bytes)
2022-08-06T10:03:14.794Z DRIVER » [Node 105] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      155
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 192
                                    └─[IndicatorCCSet]
                                        values:
                                        · indicatorId: 9
                                          propertyId:  1
                                          value:       100
2022-08-06T10:03:14.800Z SERIAL « [ACK]                                                                   (0x06)
2022-08-06T10:03:14.803Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-08-06T10:03:14.804Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.804Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-08-06T10:03:14.825Z SERIAL « 0x0122000400691a9f03d900b4f7a83a437d4d5046e7f1805554b887496cc74d723 (36 bytes)
                                  2bb00a7
2022-08-06T10:03:14.831Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.832Z DRIVER « [Node 105] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2022-08-06T10:03:14.833Z CNTRLR » [Node 105] Message authentication failed, cannot decode command. Requesting a 
                                  nonce...
2022-08-06T10:03:14.849Z SERIAL « 0x011800139b00000500b97f7f7f7f0101030000000002010000d3              (26 bytes)
2022-08-06T10:03:14.849Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.850Z DRIVER « [REQ] [SendData]
                                    callback id:            155
                                    transmit status:        OK, took 50 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -71 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2022-08-06T10:03:14.854Z SERIAL » 0x011b001369149f02c101ba87332f27bb79ac0f843ff782c60b6e059c44        (29 bytes)
2022-08-06T10:03:14.855Z DRIVER » [Node 105] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      156
                                  └─[Security2CCNonceReport]
                                      sequence number:  193
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0xba87332f27bb79ac0f843ff782c60b6e
2022-08-06T10:03:14.862Z SERIAL « 0x011c00040069149f02da015a2588bebdf183d98d2ee368e3520522b6008b      (30 bytes)
2022-08-06T10:03:14.864Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.865Z DRIVER « [Node 105] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceReport]
                                      sequence number:  218
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x5a2588bebdf183d98d2ee368e3520522
2022-08-06T10:03:14.868Z CNTRLR « [Node 105] received S2 nonce, not sure what to do with it
2022-08-06T10:03:14.870Z SERIAL « [CAN]                                                                   (0x18)
[... 36 lines cut ...]
                                  │ transmit options: 0x05
                                  │ callback id:      157
                                  └─[Security2CCNonceReport]
                                      sequence number:  194
                                      SOS:              true
                                      MOS:              false
[... cut ...]
                                    transmit status:        OK, took 20 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -69 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2022-08-06T10:03:19.317Z SERIAL « 0x010e0004000a0631050422054ec90069                                  (16 bytes)

```

stewartadam, 2023-01-12:
I'm seeing a lot of similar messages in my logs as well, it looks like sometimes a device and the controller get their comms out of sync, there's a bunch of network chatter as they try to recover. I'm doubtful it's related to data corruption or signal strength since these notes are very close to each other, and if I can manage to get a health check without these comm de-sync's I get 10/10 scores.

I can reproduce it fairly easily by attempting a health check between two neighbour nodes in the graph UI:
```
2023-01-12T22:45:41.575Z CNTRLR   [Node 024] At Normal Power, 10/10 pings were acknowledged by node 41...
2023-01-12T22:45:41.575Z CNTRLR   [Node 024] Sending 10 pings to node 41 at -1 dBm...
2023-01-12T22:45:41.587Z SERIAL « 0x011800133e00000600b57f7f7f7f00000300000000030200007b              (26 bytes)
2023-01-12T22:45:41.588Z SERIAL » [ACK]                                                                   (0x06)
2023-01-12T22:45:41.588Z DRIVER « [REQ] [SendData]
                                    callback id:            62
                                    transmit status:        OK, took 60 ms
                                    routing attempts:       2
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -75 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2023-01-12T22:45:41.595Z SERIAL » 0x011d001318169f035800f229236d79b112984075e843d9a04a419206253f8e    (31 bytes)
2023-01-12T22:45:41.596Z DRIVER » [Node 024] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      63
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 88
                                    └─[SupervisionCCGet]
                                      │ session id:      3
                                      │ request updates: true
                                      └─[PowerlevelCCTestNodeSet]
                                          test node id:     41
                                          power level:      -1 dBm
                                          test frame count: 10
2023-01-12T22:45:41.599Z SERIAL « [ACK]                                                                   (0x06)
2023-01-12T22:45:41.603Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-01-12T22:45:41.604Z SERIAL » [ACK]                                                                   (0x06)
2023-01-12T22:45:41.605Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-01-12T22:45:41.621Z SERIAL « 0x011800133f00000200b57f7f7f7f00000300000000030100007d              (26 bytes)
2023-01-12T22:45:41.622Z SERIAL » [ACK]                                                                   (0x06)
2023-01-12T22:45:41.622Z DRIVER « [REQ] [SendData]
                                    callback id:            63
                                    transmit status:        OK, took 20 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -75 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2023-01-12T22:45:41.636Z SERIAL « 0x011900040018119f031c00e5cec925f77333fde30d642b80b60071            (27 bytes)
2023-01-12T22:45:41.637Z SERIAL » [ACK]                                                                   (0x06)
2023-01-12T22:45:41.638Z DRIVER « [Node 024] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 28
                                    └─[SupervisionCCReport]
                                        session id:          3
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
2023-01-12T22:45:42.647Z SERIAL » 0x01150013180e9f035900d594045d099b15f71bcc2540f0                    (23 bytes)
2023-01-12T22:45:42.647Z DRIVER » [Node 024] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      64
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 89
                                    └─[PowerlevelCCTestNodeGet]
2023-01-12T22:45:42.650Z SERIAL « [ACK]                                                                   (0x06)
2023-01-12T22:45:42.654Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-01-12T22:45:42.655Z SERIAL » [ACK]                                                                   (0x06)
2023-01-12T22:45:42.656Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-01-12T22:45:42.671Z SERIAL « 0x011800134000000200b67f7f7f7f000003000000000301000001              (26 bytes)
2023-01-12T22:45:42.672Z SERIAL » [ACK]                                                                   (0x06)
2023-01-12T22:45:42.672Z DRIVER « [REQ] [SendData]
[... cut ...]
                                    │ sequence number: 36
                                    └─[PowerlevelCCTestNodeReport]
                                        test node id:        41
                                        status:              Success
                                        acknowledged frames: 10
2023-01-12T22:45:49.263Z CNTRLR   [Node 024] At -1 dBm, 10/10 pings were acknowledged by node 41...
2023-01-12T22:45:49.264Z CNTRLR   [Node 024] Sending 10 pings to node 41 at -2 dBm...
```

AlCalzone, 2023-03-08:
Nonce handling has been rewritten to be more deterministic as part of https://github.com/zwave-js/node-zwave-js/pull/5475, so I'm going to close this for now.

closed 2023-03-08 by AlCalzone