zwave-js/zwave-js #3929: Missing device configuration: Multisensor 7 - aeotec

Eliza1700, 2021-12-19:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Multisensor 7 - aeotec (ZWA024)

### Manufacturer ID

0x0371

### Product Type

0x0371

### Product ID

0x0371

### Firmware Version

1

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4088

### Do you have a manual?

https://help.aeotec.com/support/solutions/folders/6000237394

AlCalzone, 2021-12-19:
Duplicate of #3802

AlCalzone, 2021-12-19:
side note: the IDs you provided are plain wrong

closed 2021-12-19 by AlCalzone

Eliza1700, 2021-12-19:
dude....come on.....I provided the links.
I have no idea where I can find the id's requested. As you are probably smatter and especially better experienced in these things I think you should be able to find the ID's needed. Or am I wrong? ;-)

AlCalzone, 2021-12-19:
I've answered to your email so you'll know the next time.
Nonetheless, just saying _"I don't know"_ is better than providing false information.

Eliza1700, 2021-12-19:
I changed the input, can we reopen this issue?

AlCalzone, 2021-12-19:
Why? It is a duplicate of another issue requesting the exact same device to be added.