zwave-js/zwave-js #3930: Missing device configuration: Assa Abloy YRD-ZWM-1 Door Lock

arakasi55, 2021-12-20:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Assa Abloy YRD-ZWM-1

### Manufacturer ID

0x129

### Product Type

0xc002

### Product ID

0x0601

### Firmware Version

1.78

arakasi55, 2021-12-20:
This is the document for a similar, but different module by the same manufacturer if it helps
[YaleRealLivingAssureSL_Z-WavePlus_SystemIntegratorsGuide.pdf](https://github.com/zwave-js/node-zwave-js/files/7742537/YaleRealLivingAssureSL_Z-WavePlus_SystemIntegratorsGuide.pdf)

AlCalzone, 2021-12-20:
The Yale locks are pretty similar, but their configs differ in small details, so it would be important to have the correct manual for this one.

AlCalzone, 2021-12-20:
Also...
@zwave-js-bot config help wanted

arakasi55, 2021-12-27:
This appears to be the manual from the actual lock I'm using. Although this lock is branded Yale and mine is Lockwood, it is otherwise identical
[YaleRealLivingAssureSL_Z-WavePlus_SystemIntegratorsGuide (1).pdf](https://github.com/zwave-js/node-zwave-js/files/7779141/YaleRealLivingAssureSL_Z-WavePlus_SystemIntegratorsGuide.1.pdf)

blhoward2, 2022-10-09:
I've taken a look at this as part of a recent merge of all Yale locks, but I can find no information on this device. If OP wants it merged they're going to have to do some investigating/testing.

closed 2022-10-20 by zwave-js-assistant[bot]